bblanchon/ArduinoJson #1210: sketchbook6\SpiffsConfig Fails to Deserialize Configuration

lbussy, 2020-03-10:
I am using the code right out of your book examples.  Whenever I run it a second time (when it should have already created the ` /config.json`) it tells me it "Failed to deserialize configuration."  When the JSON prints it looks fine, and I have run it through a [JSON Validator](https://jsonformatter.curiousconcept.com/) and it returns ok.

The only changes I've made are as follows:
- Renamed `SpiffsConfig.ino` to `main.cpp` as I am using PlatformIO
- Added definitions for the functions at the top of the `main.cpp`
- Added a `SPIFFS.format();` call if the file is not found.  I did this after trying without it as well - no difference

I'm not going to include that code - because I got it from you. :)

So - problem statement:  I am unable to use the example provided.

Why?

It never seems to be able to deserialize the file.  The file prints out fine when writing, but when the ESP restarts the file is unable to deserialize again.  The file exists.  When I print out the file it looks like the first part of it is being stripped.  Here's the output:

```
Failed to deserialize configuration
Even though we failed to deserialize, here's the file:
erver": {
    "username": "admin",
    "password": "s3cr3t",
    "host": "example.com",
    "path": "/resource"
  },
  "access_points": [
    {
      "ssid": "SSID 1",
      "passphrase": "Passphrase 1"
    },
    {
      "ssid": "SSID 2",
      "passphrase": "Passphrase 2"
    }
  ]
}
```

Hypothesis:  `serializeConfig(config, file);` is truncating the output.

That's a bold statement, so here's my code (using the same Config.cpp and Config.h as in your example) which reproduces this output:

```
#include <FS.h>
#include "Config.h"

const char *filename = "/config.json";
Config config;

bool loadFile(const char *filename, Config &config);
void saveFile(const char *filename, const Config &config);
void printFile(const char *filename);
void setup();
void loop();

// Loads the configuration from a file on SPIFFS
bool loadFile(const char *filename, Config &config) {
  // Open file for reading
  File file = SPIFFS.open(filename, "r");

  // This may fail if the file is missing
  if (!file) {
    Serial.println(F("Failed to open config file."));
    SPIFFS.format();
    Serial.println("Formatted SPIFFS.");
    return false;
  }

  // Parse the JSON object in the file
  bool success = deserializeConfig(file, config);

  // This may fail if the JSON is invalid
  if (!success) {
    Serial.println(F("Failed to deserialize configuration"));
    Serial.println(F("Even though we failed to deserialize, here's the file:"));

    while (file.available())
    {
      Serial.print((char)file.read());
    }
    file.close();
    Serial.println();
    Serial.println();

    return false;
  }

  return true;
}

// Saves the configuration to a file on SPIFFS
void saveFile(const char *filename, const Config &config) {
  // Open file for writing
  File file = SPIFFS.open(filename, "w");
  if (!file) {
    Serial.println(F("Failed to create config file"));
    return;
  }

  // Serialize JSON to file
  bool success = serializeConfig(config, file);
  if (!success) {
    Serial.println(F("Failed to serialize configuration"));
  }
  file.close();
}

// Prints the content of a file to the Serial
void printFile(const char *filename) {
  // Open file for reading
  File file = SPIFFS.open(filename, "r");
  if (!file) {
    Serial.println(F("Failed to open config file, formatting SPIFFS."));
    SPIFFS.format();
    return;
  }

  // Extract each by one by one
  while (file.available()) {
    Serial.print((char)file.read());
  }
  Serial.println();
  file.close();
}

void setup() {
  // Initialize serial port
  Serial.begin(BAUD);
  while (!Serial)
    continue;

  // Mount SPIFFS
  delay(1000);
  if (!SPIFFS.begin()) {
    Serial.println(F("Failed to mount SPIFFS"));
    return;
  }

  // Load configuration
  bool loaded = loadFile(filename, config);

  if (!loaded) {
    Serial.println(F("Using default config"));
    strcpy(config.server.host, "example.com");
    strcpy(config.server.path, "/resource");
    strcpy(config.server.username, "admin");
    strcpy(config.server.password, "s3cr3t");
    strcpy(config.accessPoint[0].ssid, "SSID 1");
    strcpy(config.accessPoint[0].passphrase, "Passphrase 1");
    strcpy(config.accessPoint[1].ssid, "SSID 2");
    strcpy(config.accessPoint[1].passphrase, "Passphrase 2");
    config.accessPoints = 2;
  }

  // Save configuration
  saveFile(filename, config);

  // Dump config file
  printFile(filename);

  if (!loaded) {
    Serial.println(F("Restarting to load the config"));
    delay(1000);
    ESP.reset();
  } else{
    Serial.println(F("Done!"));
  }
}

void loop() {
}
```

Using this I can reproduce 100% of the time - on PlatformIO as my dev environment.

lbussy, 2020-03-11:
I thought some system information might be in order here:

- Windows 10
- Visual Studio Code
- PlatformIO
- Platform: Espressif 8266 2.3.3 > WeMos D1 R2 and mini
- Hardware: ESP8266 80MHz, 80KB RAM, 4MB Flash
- Framework: arduinoespressif8266 3.20603.200130 (2.6.3)
- getCoreVersion(): 2_6_3
- system_get_sdk_version():  2.2.2-dev(38a443e)
- system_get_boot_version(): 31
- getFreeHeap(): 48912

Using whatever branch of ArduinoJson that pulls via platformio.ini indicating "ArduinoJson".

bblanchon, 2020-03-12:
Hi @lbussy,

I just tried the code from the book with:

* Arduino IDE: 1.8.10
* Core for ESP8266: 2.6.3
* Board: [Adafruit Feather Huzzah](https://www.amazon.com/dp/B074XMF9W7/?tag=bblanchon0b-20)

I can confirm it works as expected.

I think our best clue is the error code: [IncompleteInput](https://arduinojson.org/v6/api/misc/deserializationerror/).
As the documentation says, it can also come from:

* an empty input
* a timeout

To better see what's happening, I suggest that you wrap the file with a LoggingStream from [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils):

```c++
LoggingStream fileWithLog(file, Serial);
if (!deserializeConfig(fileWithLog, config)) {
```

This will log what ArduinoJson pulls from the file, and it will allow us to see where ArduinoJson stops. In particular, I want to know if it stops at the beginning or in the middle.

I suppose that the problem is due to a timeout.
In that case, you could try to [increase the timeout value](https://arduinojson.org/v6/how-to/change-the-timeout/).
Alternatively, you can decorate the file with a ReadBufferingStream to force ArduinoJson to read large chunks:

```c++
ReadBufferingStream fileWithBuffer(file, 64);
if (!deserializeConfig(fileWithBuffer, config)) {
```

Please keep me informed of your progress.

Best regards,
Benoit

lbussy, 2020-03-12:
Thank you for your reply.  I may be uninformed, but I don't think extending the timeout of Serial or WiFiClient will help read from SPIFFS will it?  I added it just in case there was something I was missing but no change.

I tried to insert StreamUtil into the code and received different input although I have to admit I am not sure how to interpret it.  Here's how I implemented it in your `Config.cpp` example:

```
bool deserializeConfig(Stream &src, Config &config) {
    DynamicJsonDocument doc(1024);

    // Parse the JSON object in the file
    ReadLoggingStream stream(src, Serial);
    DeserializationError err = deserializeJson(doc, stream);
    if (err)
        Serial.print(F("ERROR: deserializeConfig() failed with error: ")); // DEBUG
        Serial.println(err.c_str()); // DEBUG
        return false;

    config.load(doc.as<JsonObject>());
    return true;
}
```

And here is the output:

```
{
  "sERROR: deserializeConfig() failed with error: IncompleteInput
Failed to deserialize configuration, here's the file we attempted:
{
  "server": {
```

So what I see (maybe you see something different) is the initial `{"s` streaming and then the whole thing bombing out (starting with my `ERROR` debug line.).

lbussy, 2020-03-12:
And then I re-read what you said and did what you asked. :)

Changed code block accordingly:

```
    // Parse the JSON object in the file
    LoggingStream fileWithLog(src, Serial);
    if (!deserializeConfig(fileWithLog, config)) {
        Serial.print(F("ERROR: deserializeConfig() failed with error: ")); // DEBUG
        // Serial.println(err.c_str()); // DEBUG
        return false;
    }
```

Here's the output:

```
Used Bytes: 753
Block Size: 8192
Page Size: 256
Max Open Files: 5
Max Path Length: 32
-------------------
-------------------
/config.json - 309B
-------------------

�L␀C␋␁␁��       �␌%��u�@�␌)S
␁�O8�
�␌5�TD��T␑␄␑-␑  )�@␗�   5Y
␓␁
nZP␋JJk�A␌T␋J�
␑T      �RM␅#*␂��Q�
```

... at which point of course the controller restarts.

I throw in a little debug:

```
bool deserializeConfig(Stream &src, Config &config) {
    Serial.println(F("DEBUG: Checkpoint (0)"));
    DynamicJsonDocument doc(1024);
    Serial.println(F("DEBUG: Checkpoint (1)"));

    // Parse the JSON object in the file
    LoggingStream fileWithLog(src, Serial);
    Serial.println(F("DEBUG: Checkpoint (2)"));
    if (!deserializeConfig(fileWithLog, config)) {
        Serial.println(F("DEBUG: Checkpoint (4)"));
        Serial.print(F("ERROR: deserializeConfig() failed with error: ")); // DEBUG
        Serial.println(F("DEBUG: Checkpoint (0)"));
        // Serial.println(err.c_str()); // DEBUG
        return false;
    }

    config.load(doc.as<JsonObject>());
    return true;
}
```

And get the following:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
[... 241 lines cut ...]
DEBUG: Checkpoint (1)
DEBUG: Checkpoint (2)
DEBUG: Checkpoint (0)
DEBUG: Checkpoint (1)
DEBUG: Checkpoint (2)
DEBUG: Checkpoint (0)
DEBUG: Checkpoint (1)
DEBUG: Checkpoint (2)
DEBUG: Checkpoint (0)
DEBUG: Checkpoint (1)
DEBUG: Checkpoint (2)
DEBUG: Checkpoint (0)
DEBUG: Checkpoint (1)
DEBUG: Checkpoint (2)
DEBUG: Checkpoint (0)
␂@�Z-n␀␙␕1�Mt'��u�@�␌)Q
        �-)␂x�P���)�A
␑T␓%␑   )�@␗�   5Y�␕    �LPA�
�k�A␄T␋��
␑T␋�RO␅)"F��S�
```

.. and a crash.  I did erase the flash + SPIFFS and tried again, same issue.

I can read the entire file into a string and deserialize the string, but that seems rather wastefull.

bblanchon, 2020-03-13:
A loop like this one could be due to a stack corruption, but I don't see any large object on the stack.

I see that you call FS.begin() in several places, are you sure it's OK to call it multiple times.

It could be a bug in SPIFFS implementation, did you try other versions of the core?
I'm using the same version, but on different hardware; maybe there are some options to tweak, like the frequency.

Did you try the `ReadBufferingStream`? It would significantly reduce the number of read operations and maybe reduce the probability of failure.

I don't think this problem is due to ArduinoJson.
You should be able to reproduce it with a simple loop like:

```c++
char c;
while (file.readBytes(&c, 1))
   Serial.write(c);
```

lbussy, 2020-03-13:
> A loop like this one could be due to a stack corruption, but I don't see any large object on the stack.

Seems like it's just my luck I hit stupid things like this.  I appreciate you trying to help.  I stripped this to the bare bones and as you say there's just nothing on the stack to account for it.

> I see that you call FS.begin() in several places, are you sure it's OK to call it multiple times.

I don't know that it's not.  I explicitly close it each time.  The instance should remain scoped to the function even so.  Keep in mind I got the issue using the code verbatim from your example as well.  The way I have it now is just to separate what's going on for debug purposes.

> It could be a bug in SPIFFS implementation, did you try other versions of the core?

I have not - I'll try to figure out how to do that. :)

> You should be able to reproduce it with a simple loop

I tried that and I had no errors reading the file.

> Did you try the ReadBufferingStream? 

I did, as you suggested above.  I get that strange looping as I did with `LoggingStream`.

I mean clearly I'm toxic to ArduinoJson. :smile:  I've been working remotely so when I get home I am going to replace the controller with a new one out of the bag since that's the only thing I have not tried.  Assuming I can get out of this airport without getting Corona Virus, I should be able to get back to this in a few hours.

lbussy, 2020-03-14:
Okay - I missed where you said you used Arduino IDE.  I tried with Arduino IDE and it worked.  I tried with PlatformIO and no other changes and I got the error.  So, the difference is in the libs that each uses.  I'm not quite sure how to get the sketch to print out the core version.

If it would help, I'd like to post on [PlatformIO.org](https://community.platformio.org/) as well, but this is the sketch that you include with your book and I don't want to do that without your permission.

lbussy, 2020-03-15:
I noticed that you were using esp8266/Arduino (v 2.6.3) which is typical for Arduino IDE, and I am using platformio/platform-espressif8266 (v2.3.3) which is what PlatformIO uses.

So from there, I tried to switch the core version, and then completely messed up my dev environment.

I went nuclear-mode, uninstalled, hacked and slashed all of the configuration out of my system, then reinstalled and guess what?  It works. It's still using platformio/platform-espressif8266 (v2.3.3) but it works.  No idea why. I don't care anymore :)

Thanks again for your assistance.  Even without you giving me a solution - it helped to know it worked for you (and how).

Cheers!

bblanchon, 2020-03-15:
Thanks for the update Lee.
I'm glad you fix your problem because I was running out of ideas.
Good luck with your project!

lbussy, 2020-03-18:
I re-opened because I did figure out my exact fail case, and I thought you would want to know:

When I compile in `debug`, the sketch fails to deserialize.  When I use `release` it works.  I am not certain however how to replicate this in Arduino IDE.  The PlatformIO docs covering this are [located here](https://docs.platformio.org/en/latest/projectconf/build_configurations.html).

bblanchon, 2020-03-19:
I tried to reproduce the problem; here is what I did:

1. Installed PlatformIO IDE
2. Installed ArduinoJson 6.14.1 from the PlatformIO IDE
3. Clicked "Import Arduino Project" and selected "SpiffsConfig" folder
4. Change "build_type" to "debug"
5. Clicked "Upload and Monitor"

The log shows "Building in debug mode," so I'm pretty sure I did everything right.
Yet, the program works fine!

I ran the test on my Huzzah. I ordered a D1 Mini on Amazon, but I'll only receive it next week.

Again, I don't believe the problem comes from ArduinoJson.

lbussy, 2020-03-19:
I don't necessarily think the problem is yours, I'm just hoping you can lend some sanity to it all. :)  I do have a post on the PlatformIO community forum as well with a much scaled down version of the test.

Did you use VSCode or CLion?  I am using VSCode.  I can try it in CLion ... as soon as I figure my way around in it.

bblanchon, 2020-03-20:
I used VSCode, but I don't think it matters.

bblanchon, 2020-03-20:
Good news! I received the D1 mini early, and I managed to reproduce the bug with 6.14.1.

However, it works fine when I use the tip of branch 6.x.
So, I probably already fixed the bug, but it could also be luck.

I'm currently bisecting to find the exact commit...

lbussy, 2020-03-20:
Wow, crazy!

And I'm *really* happy to hear it was not me. :)

bblanchon, 2020-03-20:
Apparently, 6e52f242 fixed this bug, but I'm not sure it should.
I'll investigate further.

bblanchon, 2020-03-20:
I have a bad feeling about this.
Simply adding a `DUMP(c)` after [reading the next char](https://github.com/bblanchon/ArduinoJson/blob/d2a67f362ffd95bedb52aca1f7114050287e297c/src/ArduinoJson/Json/JsonDeserializer.hpp#L54) is sufficient to fix the bug.
It looks like an undefined behavior.

lbussy, 2020-03-20:
Oh my.  Well, I feel for you, sir.  Sorry I dropped this on you but I am glad it turns out to be a "you" thing rather than "me."  I wish I knew how long I spent chasing myself in circles trying to figure it out. :)

bblanchon, 2020-05-07:
Adam Iredale reported a similar crash that only occurs in debug mode.
He published an MCVE here:
https://bitbucket.org/adamiredale/mvce-arduinojson

I could reduce the MCVE down to just:

```c++
DynamicJsonDocument doc(512);
String json = "{\"ps\":{\"src\":{\"group\":\"pan\",\"unit\":\"esp\"},\"dest\":{\"group\":\"pan\",\"unit\":\"mmcu\"},\"type\":\"req\",\"seq\":19,\"cmd\":{\"code\":\"PM\",\"payload\":{\"power\":[true,false,false,false,true,false,false,false]}}}}";
deserializeJson(doc, json);
```

Replacing `json` with `json.c_str()` fixes the crashes, so there is something wrong with the way `deserializeJson()` handles `String`.

bblanchon, 2020-05-07:
I spoke too quickly: adding `.c_str()` doesn't solve the problem with the original code from Adam.
There is definitely something else...

bblanchon, 2020-05-07:
It seems that the stack grows too quickly in debug mode.

In release mode, each recursion in `deserializeJson()`, adds 192 bytes to the stack.
In debug mode, however, each recursion adds 976 bytes.

In other words, **the stack grows five times faster in debug mode**, so this explains the crash when you have more than four nesting levels in the JSON document.

For now, I have no idea why this happens...

bblanchon, 2020-05-07:
Upgrading the Espressif 8266 platform on PlatformIO improves the situation, but it's still +880 bytes per recursion.

bblanchon, 2020-05-09:
I played with the build flags and saw that `-Og` triggers the huge stack consumption.

The table below summarizes all the tests I made.  
As you can see, only `-Og` increases the stack usage dramatically.

| `debug_build_flags` | stack usage per recursion |
|:-----|:----|
| *none* | 192 |
| `-O0` | 192 |
| `-Os` | 176 |
| `-Og` | 880 🡄 |
| `-g` | 192 |
| `-g3` | 192 |

So, a simple workaround is to change the [`debug_build_flags`](https://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-build-flags).
Instead of the default `-Og -g2 -ggdb2`, use `-O0 -g2 -ggdb2`.

```ini
; platformio.ini
build_type = debug
debug_build_flags = -O0 -g2 -ggdb2
```

lbussy, 2020-05-09:
How crazy ...

ewaldc, 2020-05-12:
Ah, so sorry, saw this issue too late. I hit this problem a while ago, not with D1 though, but nodemcu V1 and came to the same conclusion: stack overflow and/or corruption.  Almost filed an issue myself, but then also discovered that apparently insignificant changes, suddenly made the problem disappear, and thus that AJ could never be the problem...
Ended up changing the code and abandoning compilation in debug mode and use conditional Serial.print instead.   
Now at last I can try debug_build_flags workaround!  Great finding!

bblanchon, 2020-05-18:
I don't think I can do anything in the source to fix this issue, so I described the issue in the documentation.
See: [ESP8266 crashes when compiled in Debug mode](https://arduinojson.org/v6/error/esp8266-crashes-in-debug-mode/)

lbussy, 2020-05-25:
Thanks for chasing this down Benoît.  I empathize with you - bugs are frustrating!

roy-zahor, 2020-05-27:
Hi,

I can confirm the same problem on ESP32 + Visual Studio + Visual Micro. 

What I have tried:
1. Selecting "Release" and "No Optimization".
2. Going to vMicro -> Compiler -> Advanced... -> Micro General, and selecting "True" to disable all optimizations.
--> No success

According to Dr. Google, Build flags are taken from `arduinoide\hardware\arduino\avr\platform.txt` and not from Visual Micro properties window.

Try 1:
Since I didn't want to edit paltform.txt, I have tried to add `#pragma GCC optimize ("O0")` to the source file(s), but it didn't work.
--> No success

Try 2:
Edit platform.txt and replaced "-Og" with "-O0".
--> No success

Try 3:
Edit `__VM\Compile.vmps.xml` and replaced "-Og" with "-O0".
--> No success

bblanchon, 2020-05-27:
Hi @roy-zahor,

I tried ESP32 with PlatformIO, and it worked fine.
Are you sure this is the same issue?
From your message, it's not clear whether your program works in Release mode.

Best regards,
Benoit

roy-zahor, 2020-05-27:
> Hi @roy-zahor,
> 
> I tried ESP32 with PlatformIO, and it worked fine.
> Are you sure this is the same issue?
> From your message, it's not clear whether your program works in Release mode.
> 
> Best regards,
> Benoit

Hi,

I can't be sure, but it looks like the same issue. The stack dump looks like the stack has ~6-8 recursive calls and then it crashes.

I'm building with Visual Micro and not with PlatformIO, so I did the best I can to define release build like you have suggested, but I couldn't get it to work. Visual Micro build mechanism is different and I wasn't sure I have set all the flags correctly.

bblanchon, 2020-05-28:
Hi @roy-zahor,

I compiled a test project with Visual Micro, and I can see that `-Os` is passed to the compiler.
You can find the compiler command line in "Output" panel, with "Micro Output" selected:
[image]
I tried to change the Debug mode, but it did not affect the compile flags.

Is it possible that you get a "regular" stack overflow?
Are you using a `StaticJsonDocument` or `DynamicJsonDocument`?
Do you have large buffers in the stack?
Can you reproduce the bug in a [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

See also:

* [Why does my device crash or reboot?](https://arduinojson.org/v6/faq/why-does-my-device-crash-or-reboot/#stack-overflow)

Best Regards,
Benoit

sparkplug23, 2020-06-11:
Any updates on this? I have the same problem. The nodemcu is crashing immediately and showing arduinojson in the decoded trace.

bblanchon, 2020-06-12:
@sparkplug23, I don't think there is anything we can do on the library side. You must change `debug_build_flags` [as explained](https://arduinojson.org/v6/error/esp8266-crashes-in-debug-mode/).

sparkplug23, 2020-06-12:
@bblanchon Yes I found your webpage on the issue after posting this last night. Unfortunately those build flags are causing me other problems with pgmspace.h not allowing it to compile.

bblanchon, 2020-06-12:
That's very surprising.
What is the error exactly and how can I reproduce it?

sparkplug23, 2020-06-12:
I am using these build flags:

`monitor_filters = esp8266_exception_decoder
build_type = debug
debug_build_flags = -O0 -g2 -ggdb2`

and the build error is:

`In file included from C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/assert.h:10:0,
Compiling .pioenvs\esp_8266_com3_debug\lib965\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
                 from C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/reent.h:459,
                 from C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/errno.h:11,
                 from C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/errno.h:9,
                 from C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp:26:
C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/pgmspace.h: In function 'uint32_t pgm_read_dword_unaligned(const void*)':
C:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/pgmspace.h:107:1: error: a15 cannot be used in asm here
 }
 ^
*** [.pioenvs\esp_8266_com3_debug\lib965\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o] Error 1`

bblanchon, 2020-06-13:
This seems to be a bug in Xtensa SDK:

* [define debug uncovers error in xtensa sdk](https://www.esp8266.com/viewtopic.php?p=86110)
* [new compiler breaks programs: error: a15 cannot be used in asm here](https://www.reddit.com/r/esp8266/comments/ggapj8/new_compiler_breaks_programs_error_a15_cannot_be/)
* [Is pgm_read_word required? ESP8266 does not make happy - but seems correctable.](https://www.reddit.com/r/FastLED/comments/gd4k7m/is_pgm_read_word_required_esp8266_does_not_make/)

I can confirm that I don't have the problem with version 2.5.2.

I don't see any reason why `-O0` would trigger this bug.
Maybe the default value for `debug_build_flags` contains additional flags.
Can you do a verbose build with the old and new `debug_build_flags` and compare the command lines?

sparkplug23, 2020-06-13:
These are the command lines for the normal and debug build types:

```cpp
Processing esp_8266_com3 (build_flags: -mtarget-align, -Wl,-Map,firmware.map, -Wl,-Teagle.flash.1m.ld, -DBEARSSL_SSL_BASIC, -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703, -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++ -w; upload_port: COM3; platform: espressif8266@2.5.2; framework: arduino; board: esp12e; board_build.flash_mode: dout; board_build.f_cpu: 80000000L; build_unflags: -Wall; monitor_speed: 115200; upload_resetmethod: nodemcu; upload_speed: 115200; extra_scripts: pio/strip-floats.py)
```

```cpp
Processing esp_8266_com3_debug (build_flags: -mtarget-align, -Wl,-Map,firmware.map, -Wl,-Teagle.flash.1m.ld, -DBEARSSL_SSL_BASIC, -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703, -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++ -w; upload_port: COM3; monitor_filters: esp8266_exception_decoder; build_type: debug; debug_build_flags: -O0 -g2 -ggdb2; platform: espressif8266@2.5.2; framework: arduino; board: esp12e; board_build.flash_mode: dout; board_build.f_cpu: 80000000L; build_unflags: -Wall; monitor_speed: 115200; upload_resetmethod: nodemcu; upload_speed: 115200; extra_scripts: pio/strip-floats.py)
```

I had seen the links you sent me already (thank you though!). I have tried with and without `-D DEBUG` added with the same outcome.

bblanchon, 2020-06-13:
This compares release vs debug, but I wanted to compare debug builds with default (the one that works) vs custom `debug_build_flags` (the one that doesn't work).
Anyway, I'm not sure this line includes all the build flags, please copy the gcc command line instead.

Also, it seems that you have version 2.5.2 of the platform; according to the comments, this version should not be affected by the bug.
Did you downgrade?
If so, did it fix the issue?

ewaldc, 2020-06-13:
@sparkplug23, I am struggling with the same VSCode debug build issue.  In _release_ mode, the code runs perfectly on nodemcu (and the code stretches the ESP/ArduinoJson to the max).
@bblanchon, I am running 2.5.2 also, which may not be affected by the bug, but I can't test it since I don't manage to compile it with `-DDebug` in VSCode/PlatformIO. 
Managed to get around the `error: a15 cannot be used in asm here` error though by using `-DPGM_READ_UNALIGNED=0`, but then it hits another error with the linker...

I managed to get around it to by changing the toolchain/compiler, but I forgot the write down the exact combination that works :-(
Anyhow, it was an ugly solution, requiring patching the Arduino libraries and using a different ESP runtime library than the default (July 2019 I believe)...

Posted on the PlatformIO forum but no response...

sparkplug23, 2020-06-13:
@ewaldc We must have read the same post, I also tried `DPGM_READ_UNALIGNED` too.
It's frustrating, I have not used the esp_decoder before and my code is reaching the point where any bugs are infrequent/random and probably impossible to trace without the decoder. I am hoping some more experienced users of vscode can prevail here as I have reached the limits of my knowledge on it.

ewaldc, 2020-06-13:
@sparkplug23, the workaround is to use Arduino IDE.  Basically I created symbolic links to my source and include directories, and main.cpp --> main.ino so I don't have to copy anything.
The stack decoder there works perfectly.  BTW the problem is with PlatformIO.   I found the problem and fixed it, but then installed PIO 4.3.5a1 (to try the ESP exception decoder) and lost my changes.
If I have some time, I will look into it again.

Since Arduino IDE does not allow to set cpp defines in an easy way (e.g. with -D), add this to your include files so that they compile on both IDE's.
```
// Arduino IDE
#if defined(ARDUINO) && !defined(PLATFORMIO)
#define OpenHABDebug
#define SPIFFS_OBJ_NAME_LEN 64
#define ARDUINOJSON_DECODE_UNICODE 1
#endif
```

sparkplug23, 2020-06-13:
@bblanchon As @ewaldc also said, I am unable to test the bug as I can't get the project to compile with those extra build flags. I am using 2.5.2, I tried several versions but am unable to build.

Ah my bad, I misunderstood the builds you asked for. Forgive me, but I am unsure how to get the gcc command line you asked for. I tried googling but am still unsure. I would appreciate it if you could point me in the right direction.

sparkplug23, 2020-06-13:
@ewaldc Thanks I will take a look into that. My hope is that esp_decoder will eventually be properly (bug free) be supported in vscode. I looked hard at this at the end of last year and it was not supported natively at all, so I am hoping it's just a few hiccups in the development of it. It makes me miss the days of ICD and stepping line by line on hardware.

ewaldc, 2020-06-13:
I tried it on PIO 4.3.5a1 and it works really well with VSCode 1.45.0, not with 1.45.1 (frequent hangs).
The build error with `-DDebug` predates 4.3.4 (4.3.5a1)....

bblanchon, 2020-06-13:
@ewaldc, thank you very much for your help, as usual 😉.

@sparkplug23, to get the GCC command line, build with "Verbose Build" and copy one of the lines that starts with `xtensa-lx106-elf-g++` and ends with `.cpp`:

[image]

As you can see, there seem to be more flags, maybe one of them can help...

sparkplug23, 2020-06-13:
@bblanchon So I was doing this the first time around, but I had "-w" flag set to suppress warnings (still working my code up, was taking 5+ times longer to build without them suppressed). Anyway, after figuring out this was why no gcc was appearing I got what you asked. 

I have isolated just the flag passing part in the next two codeblocks for brevity, but also included the full versions of those lines at the bottom in case you want to see them.

Normal ( esp8266_exception_decoder and build_type debug, NO manual debug_build_flags)
```cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp_8266_com3_debug\src\ConfigUser\mMapConfig.cpp.o -c -fno-rtti -std=c++11 -mtarget-align -w -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Og -g2 -ggdb2 -DPLATFORMIO=40304 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DBEARSSL_SSL_BASIC -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH -DVTABLES_IN_FLASH -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP12E\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=1460 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -D__PLATFORMIO_BUILD_DEBUG__ -Iinclude  
``` 

With manual `debug_build_flags = -O0 -g2 -ggdb2`
```cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp_8266_com3_debug\src\ConfigUser\mMapConfig.cpp.o -c -fno-rtti -std=c++11 -mtarget-align -w -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -O0 -g2 -ggdb2 -DPLATFORMIO=40304 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DBEARSSL_SSL_BASIC -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH -DVTABLES_IN_FLASH -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP12E\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=1460 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -D__PLATFORMIO_BUILD_DEBUG__ -Iinclude
```
As far as I can see, the only difference between the default and manual `debug_build_flags` is `-O0` and `-Og`. With the original, "g", the debug builds successfully but then has it bootloop issue.

>>Just for completness, the full gcc line, you don't need to read this
```cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp_8266_com3_debug\src\ConfigUser\mMapConfig.cpp.o -c -fno-rtti -std=c++11 -mtarget-align -w -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Og -g2 -ggdb2 -DPLATFORMIO=40304 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DBEARSSL_SSL_BASIC -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH -DVTABLES_IN_FLASH -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP12E\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=1460 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -D__PLATFORMIO_BUILD_DEBUG__ -Iinclude -IHomeControlSystem -Ilib\TasmotaModbus-1.1.0\src -Ilib\TasmotaSerial-2.4.1\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\Wire -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\DNSServer\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\SoftwareSerial\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src "-Ilib\ESP Async WebServer_ID306_Modified_Michael\src" -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\Hash\src -Ilib\ESPAsyncTCP_ID305\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266httpUpdate\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPClient\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ArduinoOTA -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\Ticker\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\EEPROM -Ilib\NeoPixelBus-2.5.3\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\libraries\SPI -Ilib\ArduinoJson\src -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu HomeControlSystem\ConfigUser\mMapConfig.cpp
```
```cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp_8266_com3_debug\src\ConfigUser\mMapConfig.cpp.o -c -fno-rtti -std=c++11 -mtarget-align -w -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -O0 -g2 -ggdb2 -DPLATFORMIO=40304 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DBEARSSL_SSL_BASIC -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH -DVTABLES_IN_FLASH -DF_CPU=800000 [... cut ...] Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\micha\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu HomeControlSystem\ConfigUser\mMapConfig.cpp
```

bblanchon, 2020-07-26:
Good news! I finally found the problem! 🎉 

It turns out `-Og` hates when you return a struct by value: each return statement increases the stack usage.

I replaced about a hundred return-by-value, and the stack usage of `parseVariant()` dropped **from 336 to 64 bytes** (according to [`-fstack-usage`](https://gcc.gnu.org/onlinedocs/gnat_ugn/Static-Stack-Usage-Analysis.html)). 🥳

ewaldc, 2020-07-26:
Wow! Great find.  Will give it try on Monday.

lbussy, 2020-07-26:
That's awesome!  So in a twisted way I helped development?  😇

bblanchon, 2020-07-26:
The fix is available on branch `6.x`.

closed 2020-07-26 by bblanchon

ewaldc, 2020-07-26:
The branch went live on 160 nodes just now. Normally they don't last a whole week without crash. If all is well they are all alive before the weekly reboot on Sunday. All I can do now is wait patiently :sunglasses:

sparkplug23, 2020-07-27:
> The fix is available on branch `6.x`.

Awesome thank you, will test it soon.

bblanchon, 2020-08-01:
This fix is available in [ArduinoJson 6.16.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.16.0).

sparkplug23, 2020-08-01:
Things are working again, finally able to use stack decoder to debug my code. Many thanks for the prompt fix.

commit 144ff3b0 (2020-07-26): Updated the change log. Closes #1210 and closes #1314.