bblanchon/ArduinoJson #1211: Arduino Json library to serialize object (not JsonObejct)

fabiuz7, 2020-03-12:
Hi,
Thank you for implementing Message Pack, IMHO this is a really great addition!

I'm working on a project in which I need to save on flash memory the state of an object, and that object should be restored when Arduino is reset (or it lost power supply).

I was wondering if this library can be used or has some facilities to serialize a generic object, not just a JSON object.

i.e. I was looking for something similar to boost serialization, but in Arduino environment. 
The main problem with boost is that the rely on a typical *stream* of C++, different from Stream class used by Arduino.... also I don't know their performance and memory occupation on a microcontroller.

I would be glad if someone can point me to the right direction to solve this issue.

bblanchon, 2020-03-13:
Hi @fabiuz7,

This is a common question, so I answered in the FAQ:

* [Can I automatically serialize an object?](https://arduinojson.org/v6/faq/automatically-serialize-an-object/)

Let's me know if you have any question or remark.

Best regards,
Benoit

fabiuz7, 2020-03-13:
I didn't see the FAQ page, sorry!
It seems a pretty comprehensive answer, but I remain with a question. I know C#, and I'm not searching for something like automatic reflection, it could be too heavy for a microcontroller.
Instead, I was searching for a "structured" and well-defined approach to serialize objects (maybe nested among them object), and I found [this library](https://github.com/fraillt/bitsery), really lightweight and it compiles with -fno-rtti and c++11. Of course, you have to implement the "serialization" method for each class, but this is fine to me. I was wondering if I can get the "same" result in term of code simplicity with your library (I'm not really concerned about parsing speed since I have few objects to persist).

Hoping this post is clear enough, thank for your time!

bblanchon, 2020-03-13:
> the "same" result in term of code simplicity with your library

Do you think the following code

```c++
template <typename S>
void serialize(S& s, MyStruct& o) {
    s.value4b(o.i);
    s.value2b(o.e);
    s.container4b(o.fs, 10);
}
```

is simpler than

```c++
void serialize(JsonObject s, MyStruct& o) {
    s["i"] = o.i;
    s["e"] = o.e;
    copyArray(os.fs, s.createNestedArray("fs"));
}
```

?

I think they are equally complicated.

fabiuz7, 2020-03-13:
In fact your solution seems even more elegant. The last question that should close my doubts: is your solution a bit more expensive (in terms of memory and computational time) because it internally creates a JSON structure before serialization?

bblanchon, 2020-03-14:
Yes, ArduinoJson is likely to be slower than bitsery, but it's an unfair match: one uses a versatile text format; the other uses a tailored binary format.

Yes, you have to write a deserialize function as well.
In writing this function, you'll realize that there are several decisions that you'll have to make:

* what if a member is missing?
* what if there is an extra member?
* what if a value has the wrong type?
* what if an array is too large?
* what if a value is out of range?

I don't think a library can automatically make these decisions for you.
Even if the library offered options to configure the behavior for each member, you'd end up with something more complex than the hand-written function.

fabiuz7, 2020-03-16:
Thank you for the exhaustive response! In fact, there are many questions and, finally, I've implemented a custom solution for my project.

> Yes, ArduinoJson is likely to be slower than bitsery, but it's an unfair match: one uses a versatile text format; the other uses a tailored binary format.

In the meanwhile, I was wondering if about performance (even if it is not strictly necessary for my project). A JsonDocument internally stores information (integer, float, string) in binary format (my hypothesis, I didn't look inside your library) hence serializationMsgPack should be much faster than serializeJson. Is that true?

I've tested this hypothesis with a simple script and serializeMsmPack seems more than twice faster w.r.t. serializeJson. Is there a performance comparison between MsgPack and JSON on your website? Maybe I miss it...

bblanchon, 2020-03-17:
I never compared the performance of [serializeMsgPack()](https://arduinojson.org/v6/api/msgpack/serializemsgpack/) against  [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/), but a two-fold improvement seems right.

fabiuz7, 2020-03-17:
For the sake of my curiosity, I have just created a simple [gist](https://gist.github.com/fabiuz7/40bf4247f4d762dd483a765c0c2cba41) reporting some results. This will help me to consider Messagepack wherever possible.

bblanchon, 2020-03-18:
Thank you very much for sharing this benchmark.
I added links to you gist from the following pages:

* [deserializeMsgPack](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/)
* [serializeMsgPack](https://arduinojson.org/v6/api/msgpack/serializemsgpack/)

fabiuz7, 2020-03-18:
Hoping that this may help someone else to decide which is the best protocol for their own application. 

I think we can close this issue.

bblanchon, 2020-03-18:
Whatever you do, I think it's worth reading for other ArduinoJson users, so tell me if I need to update the links, or if you want a publication in arduinojson.org/news.

For your information, MessagePack represented only 0.33% of the traffic of arduinojson.org over the last year.

closed 2020-03-18 by bblanchon