bblanchon/ArduinoJson #1212: SerializeJson returns null when using strings

MatejsLab, 2020-03-13:
When I put string variables to "buffer" to make JSON format with SerializeJson, it works for 30 seconds or so and then it breaks, starts returning null as value of the string in the JSON output. 

It works fine with integers and floats, but why that intermittent behavior with strings, as the string doesn't get significantly larger over time. Using ESP12S with plenty of free heap (44 kB).
I have no compiler issues...

Can't find the reason for the problem yet.

There is the code

```
// INITIALIZATION
DynamicJsonDocument  sistem_JSN(512);

// LOOP
  if (Timer1000.Pridobi())
  {
 String Sistem;
      String StevecMqtt = String(Stevec / 3600) + "h" + String((Stevec % 3600) / 60) + "m" + String(Stevec % 60) + "s";
      static int Sis_Lenght,Sis_Size;

      sistem_JSN[String("TimeMQTT_s")] = StevecMqtt;
      sistem_JSN[String("TimeMQTT_len")] = StevecMqtt.length();
      sistem_JSN["TimeMQTT"] = Stevec;
      sistem_JSN["Signal"] = WiFi.RSSI();
      sistem_JSN["Free_DRAM(kB)"] = ESP.getFreeHeap()/1024;
      sistem_JSN[String("Data_len")] = Sis_Lenght;
      sistem_JSN[String("Data_size")] = Sis_Size;
      
      serializeJson(sistem_JSN, Sistem);
      
      Sis_Lenght = Sistem.length();
      Sis_Size = sizeof(Sistem);
      
      client.publish(DeviceID + "/Sistem", Sistem);
}

```
And here is the output data

`{"TimeMQTT_s":null,"TimeMQTT_len":7,"TimeMQTT":961,"Signal":-55,"Free_DRAM(kB)":44,"Data_len":113,"Data_size":12}`

One more thing, how do I properly pass String value through the serializeJson to remedy this problem as the string example presents a lot of options with many related issues. Not sure why my code works fine and only that TimeMQTT_s breaks, but TimeMQTT keeps going.

bblanchon, 2020-03-14:
Hi Matej,

Please see:

* [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)

Best regards,
Benoit

MatejsLab, 2020-03-14:
Thank you!

You have a great website.  >> [Arduino Json](https://arduinojson.org/)

My solution was:

`sistem_JSN.clear(); ` <<--  placed before first entry for new data

Placed before this code in my case....
```
 sistem_JSN[String("TimeMQTT_s")] = StevecMqtt;
 sistem_JSN[String("TimeMQTT_len")] = StevecMqtt.length();
. . . . . .
. . . . . .
```

Cheers,
Matej

closed 2020-03-14 by MatejsLab

bblanchon, 2020-03-14:
You're welcome Matej!
Good luck with your project.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)