bblanchon/ArduinoJson #1213: [V5]: Can't parse more than one nested levels

effelle, 2020-03-15:
Benoît,

I need to expand my json parser to catch a new format.
My code works fine with 1 level nested but fails when there is a second level.
Actually I can grab and parse the following formats:
`{"POWER":{"Voltage":[4.494,4.353]"Current":[0.020,0.021]"Factor":[0.81,0.90]}}`
or
`{"AM2301-04":{"Temperature":28.8,"Humidity":71.2}`
But I'm unable to parse this type of json:
```
 {"TX23":{"Speed":{"Act":8.6,"Avg":8.2,"Min":0.0,"Max":15.8},"Dir":{"Card":"SSO","Deg":157.5,"Avg":145.5,"AvgCard":"SO","Min":112.5,"Max":292.5,"Range":180.0}}}
```
In short, I need a way to know when an object has another object inside.
My actual code:
```
void HAssAnnounceSensors(void)
{
  uint8_t hass_xsns_index = 0;
  bool is_sensor = true;
  uint8_t subqty = 0;
  do
  {
    mqtt_data[0] = '\0';
    int tele_period_save = tele_period;
    tele_period = 2;                                 // Do not allow HA updates during next function call
    XsnsNextCall(FUNC_JSON_APPEND, hass_xsns_index); // ,"INA219":{"Voltage":4.494,"Current":0.020,"Power":0.089}
    tele_period = tele_period_save;

    char sensordata[512]; // Copy because we need to write to mqtt_data
    strlcpy(sensordata, mqtt_data, sizeof(sensordata));

    if (strlen(sensordata))
    {
      sensordata[0] = '{';
      //snprintf_P(sensordata, sizeof(sensordata), PSTR("%s}"), sensordata); // {"INA219":{"Voltage":4.494,"Current":0.020,"Power":0.089}}
      // USE THE FOLLOWING LINE TO TEST JSON
      snprintf_P(sensordata, sizeof(sensordata), PSTR("{\"TX23\":{\"Speed\":{\"Act\":8.6,\"Avg\":8.2,\"Min\":0.0,\"Max\":15.8},\"Dir\":{\"Card\":\"SSO\",\"Deg\":157.5,\"Avg\":145.5,\"AvgCard\":\"SO\",\"Min\":112.5,\"Max\":292.5,\"Range\":180.0}}}"));
      //snprintf_P(sensordata, sizeof(sensordata), PSTR("{\"HX711\":{\"Weight\":[22,34,1023.4], \"Battery\":25}}"));

      StaticJsonBuffer<500> jsonBuffer;
      JsonObject &root = jsonBuffer.parseObject(sensordata);
      if (!root.success())
      {
        AddLog_P2(LOG_LEVEL_ERROR, PSTR("HASS: jsonBuffer failed to parse '%s'"), sensordata);
        continue;
      }
      for (auto sensor : root)
      {
        const char *sensorname = sensor.key;
        JsonObject &sensors = sensor.value.as<JsonObject>();
        if (!sensors.success())
        {
          AddLog_P2(LOG_LEVEL_ERROR, PSTR("HASS: JsonObject failed to parse '%s'"), sensordata);
          continue;
        }
        for (auto subsensor : sensors)
        {
          // If there is more than a value on sensor data, 'n' entitites will be created
          if (subsensor.value.is<JsonArray&>()) {
            JsonArray& subsensors = subsensor.value.as<JsonArray&>();
            subqty = subsensors.size();
            char MultiSubName[20];
            for (int i = 1; i <= subqty; i++) {
              snprintf_P(MultiSubName, sizeof(MultiSubName), PSTR("%s_%d"), subsensor.key, i);
              HAssAnnounceSensor(sensorname, subsensor.key, MultiSubName, i, 1);

            }
          } else { HAssAnnounceSensor(sensorname, subsensor.key, subsensor.key, 0, 0);}
        }
      }
    }
    yield();
  } while (hass_xsns_index != 0);
}
```

Using just `value.is` I can't know if there is a nested object inside the first object and I'm out of ideas. Do you have a clue for me? 
Thanks and have a good Sunday.

Federico

bblanchon, 2020-03-16:
Hi Federico,

You can use [JsonVariant::is<JsonObject>()](https://arduinojson.org/v5/api/jsonvariant/is/) to test if the value is an object.
See demo: https://wandbox.org/permlink/rJLYt1QCxc6RDsU7

Best regards,

effelle, 2020-03-16:
Benoît, 
thanks for your reply.
Partial success but still, a bit is missing.
Using an hardcoded json
```
snprintf_P(sensordata, sizeof(sensordata), PSTR("{\"TX23\":{\"Speed\":{\"Act\":8.6,\"Avg\":8.2,\"Min\":0,\"Max\":15.8},\"Dir\":{\"Card\":\"SSO\",\"Deg\":157.5,\"Avg\":145.5,\"AvgCard\":\"SO\",\"Min\":112.5,\"Max\":292.5,\"Range\":180}}}"));
```

I've added a new IF statement under `for (auto subsensor : sensors)` :
```
          if (subsensor.value.is<JsonObject&>()) {

            JsonObject& subsensors = subsensor.value.as<JsonObject>();
            char test2[20];
            uint8_t subqty = subsensors.size();
            for (int i = 1; i <= subqty; i++) {
              snprintf_P(test2, sizeof(test2), PSTR("%s"), subsensor.key);
              AddLog_P2(LOG_LEVEL_INFO, PSTR("DATA: JsonObject found %d Subsensors: %d = '%s'"),subqty, i, test2 );
            }
          }
```
And I can see how many key I have on level 2:
```
00:00:07 DATA: JsonObject found 4 Subsensors: 1 = 'Speed'
00:00:07 DATA: JsonObject found 4 Subsensors: 2 = 'Speed'
00:00:07 DATA: JsonObject found 4 Subsensors: 3 = 'Speed'
00:00:07 DATA: JsonObject found 4 Subsensors: 4 = 'Speed'

```
and 
```
00:00:08 DATA: JsonObject found 7 Subsensors: 1 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 2 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 3 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 4 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 5 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 6 = 'Dir'
00:00:08 DATA: JsonObject found 7 Subsensors: 7 = 'Dir'

```
Now I need to understand how to translate the index to the correct key name as it always show the level 1 key and I can't use the index directly (`subsensor.key[i]`).

bblanchon, 2020-03-17:
If I understand correctly, you just want to iterate over the nested object:

```c++
JsonObject& subsensors = subsensor.value.as<JsonObject>();
for (auto subsensor : subsensors) {
   AddLog_P2(LOG_LEVEL_INFO, PSTR("Subsensors: '%s'"), subsensor.key);
}
```

effelle, 2020-03-17:
I swear I tried it before and failed miserably. 
Yes. Is what I was looking for, sorry for wasting your time Benoît....
Problem solved.

closed 2020-03-17 by effelle

effelle, 2020-03-17:
Almost solved! 
How to skip the first level if there is a nested level?
 `TX23.Speed` and `TX23.Dir` should not be considered if there are nested object like `TX23.Speed.Act` or  `TX23.Dir.Card` to be parsed.

effelle, 2020-03-17:
Never mind, Benoît, there was a missing IF in my code. Solved!