bblanchon/ArduinoJson #1214: Proper Way to Check for a Null in a Numeric Field?

lbussy, 2020-03-20:
Referencing #1083 where you explained a good portion of how to check for null, I'm curious what is the best way to check for null on a numeric?  Specifically an int or float?

If I were to guess, I'd try this:

```
    float x = obj["number"];
    if (x)
        number = x;l
    else
         number = 0.0; // Default
```

I just don't want to miss a potentially better solution.  Maybe this?

```
number = obj["number"] | 0.0;
```

bblanchon, 2020-03-20:
Hi Lee,

The default value for `float` is `0.0f`, so the three following snippets lead to the same result:

```c++
// 1
float x = obj["number"];
number = x ? x : 0.0;

// 2
number = obj["number"] | 0.0;

// 3
number = obj["number"];
```

However, if your program must differentiate `null` from `0.0`, you have to call [JsonVariant::isNull()](https://arduinojson.org/v6/api/jsonvariant/isnull/):

```c++
JsonVariant x = obj["number"];
if (x.isNull()) {
  Serial.println("Error: number can't be null");
  return;
}
number = x;
```

Best regards,
Benoit

lbussy, 2020-03-20:
Gotcha, thank you for the consult, sir.

closed 2020-03-20 by lbussy