bblanchon/ArduinoJson #1215: Parse config file from SPIFFS on a ESP32

DavidTruyens, 2020-03-20:
Hi Blachon,

I've been using V5 for quite a while. Now decided to move to V6. I've read a couple of time that I just should remove the "readBytes" (as mentioned [here]).(https://github.com/bblanchon/ArduinoJson/issues/1159) .

I've tried a lot of different options, but non is working.

Here's where it goes wrong I guess:
```
/*
    std::unique_ptr<char[]> buf(new char[size]);

    //credFile.readBytes(buf.get(), size);
    credFile.read(buf,size);
    Serial.println(buf.get());
*/
    StaticJsonDocument<1024> jsonBuf;

    DeserializationError error = deserializeJson(jsonBuf, credFile);

```
So I'm not sure if I still need the buffer or if I can just read the file directly..

Here's the full code:

```
bool loadConfig()
{

    if (!SPIFFS.begin())
    {
        Serial.println("Failed to mount file system");
        return false;
    }

    File credFile = SPIFFS.open("/cred.json", "r");
    if (!credFile)
    {
        Serial.println("Failed to open cred file");
        return false;
    }
    size_t size = credFile.size();
    Serial.print("Size cred file: ");
    Serial.println(size);
    if (size > 1024)
    {
        Serial.println("Cred file is bigger than 1024");
        return false;
    }
/*
    std::unique_ptr<char[]> buf(new char[size]);

    //credFile.readBytes(buf.get(), size);
    credFile.read(buf,size);
    Serial.println(buf.get());
*/
    StaticJsonDocument<1024> jsonBuf;

    DeserializationError error = deserializeJson(jsonBuf, credFile);

    if (error)
    {
        Serial.println("Failed to parse cred file");
        return false;
    }
    /*
    const char *J_ssid = json["ssid"];
    const char *J_pass = json["pass"];
    const char *J_deviceId = json["ATTID"];
    const char *J_token = json["ATTToken"];
    const char *J_CONFIG_FIRMWARE_UPGRADE_URL = json["upgrade_URL"];
    const char *J_tennant_URL = json["tennant_URL"];
*/
    strcpy(credentials.ssid, jsonBuf["ssid"]);
    strcpy(credentials.pass, jsonBuf["pass"]);
    strcpy(deviceId, jsonBuf["ATTID"]);
    strcpy(token, jsonBuf["ATTToken"]);
    strcpy(CONFIG_FIRMWARE_UPGRADE_URL, jsonBuf["upgrade_URL"]);
    strcpy(certURL, jsonBuf["cert_URL"]);

    Serial.print("SSID: ");
    Serial.println(ssid);
    Serial.print("pass: ");
    Serial.println(pass);
    Serial.print("ATT settings: ");
    Serial.println(deviceId);
    Serial.println(token);

    return true;
}
```
I'm using an Adafruit ESP32 feather and Platformio on VS code.

Regards,
David

DavidTruyens, 2020-03-20:
I started over all again with a fresh project and this seems to be working just fine.. Now looking for the issues in the bigger project.

Anyway, here's the code as a reference:

```
void readcredentials()
{
  if (!SPIFFS.begin())
  {
    Serial.println("Failed to mount SPIFFS");
  }
  else
  {
    File file = SPIFFS.open("/cred.json", FILE_READ);

    if (!file)
    {
      Serial.println("There was an error opening the file");
      return;
    }

    else
    {
      Serial.println("File opened!");

      StaticJsonDocument<512> doc;

      DeserializationError error = deserializeJson(doc, file);
      if (error)
      {
        Serial.println("error...");
      }
      else
      {
        Serial.println("Another winner!");
        strcpy(credits.ssid, doc["ssid"]);
        strcpy(credits.pass, doc["pass"]);
        strcpy(credits.MQTTid, doc["MQTTID"]);
        strcpy(credits.MQTTpass, doc["MQTTPass"]);
      }
      Serial.println("");
    }

    file.close();
  }
}
```

closed 2020-03-20 by DavidTruyens

bblanchon, 2020-03-20:
Hi @DavidTruyens,

Thanks for the update.

For future reference, you can find an example of a configuration file stored on SD in  [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/).
In addition, [Mastering ArduinoJson](https://arduinojson.org/book/) contains a case study that shows how to store a complex configuration in SPIFFS.

Best regards,
Benoit

DavidTruyens, 2020-03-20:
Hi Benoit,

I already have the book, but that's the old one. So no v6.

But thanks, I'm fine for now 😊

bblanchon, 2020-03-20:
If you purchase the ebook, you should automatically get the new one for free.
If you have a paperback version, please drop me an email.

DavidTruyens, 2020-03-20:
I have the paperback! I really liked a physical copy, it has been very
helpful.

Regards,
David