bblanchon/ArduinoJson #1216: Check for Null in JsonObjectConst

lbussy, 2020-03-21:
I seem to be burning up your issue board lately.

Using `sketchbook6\SpiffsConfig` I have added checks for null as we have discussed elsewhere.  The following changes have been made to your book example to add a bool type:

Config.h
``` cpp
// Stores the server configuration
struct ServerConfig {
  char host[32];
  char path[32];
  char username[32];
  char password[32];
  bool choice;

  void load(JsonObjectConst);
  void save(JsonObject) const;
};
```

Config.cpp:
``` cpp
void ServerConfig::load(JsonObjectConst obj) {
  strlcpy(username, obj["username"] | "", sizeof(username));
  strlcpy(password, obj["password"] | "", sizeof(password));
  strlcpy(host, obj["host"] | "", sizeof(host));
  strlcpy(path, obj["path"] | "", sizeof(path));
  JsonVariant tf = obj["choice"];
  if (!tf.isNull())
      choice = tf;
  else
      choice = true; // Default
}
```

When I do so I receive the following error:

```
.pio\libdeps\d1_mini\ArduinoJson_ID64\src/ArduinoJson/Operators/VariantCasts.hpp: In member function 'ArduinoJson6140_0000010::VariantCasts<TImpl>::operator T() const [with T = ArduinoJson6140_0000010::VariantRef; TImpl = ArduinoJson6140_0000010::VariantConstRef]':
.pio\libdeps\d1_mini\ArduinoJson_ID64\src/ArduinoJson/Operators/VariantCasts.hpp:15:16: error: inlining failed in call to always_inline 'ArduinoJson6140_0000010::VariantCasts<TImpl>::operator T() const [with T = ArduinoJson6140_0000010::VariantRef; TImpl = ArduinoJson6140_0000010::VariantConstRef]': recursive inlining
   FORCE_INLINE operator T() const {
                ^
.pio\libdeps\d1_mini\ArduinoJson_ID64\src/ArduinoJson/Operators/VariantCasts.hpp:16:35: error: called from here
     return impl()->template as<T>();
                                   ^
```

I suspect because I am attempting the cast of a `JsonObjectConst` into a `JsonVariant` instead of it being from a `JsonArray`.  I also suspect there's an easy answer to the problem but I'm not quite following what it might be.  I'm sort of thinking I might serialize the `JsonObjectConst` and then deserialize into a `JsonArray` but that seems unglamorous.

bblanchon, 2020-03-21:
Hi Lee,

Indeed you cannot convert a [JsonVariantConst](https://arduinojson.org/v6/api/jsonvariantconst/) (read-only) to a [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/) (read-write), because:

1. it would violate the const-ness
2. `JsonXxxConst` classes lack the pointer to the memory pool

To fix your program, you must replace `JsonVariant` with `JsonVariantConst`.

Best regards,
Benoit

lbussy, 2020-03-21:
I owe you again sir.   Had I not been so tired yesterday I might even have guessed at that!  Well, maybe not. :)

closed 2020-03-21 by lbussy