bblanchon/ArduinoJson #1217: Latest version gives compiler errors

vortex314, 2020-03-23:
Latest version 6.x
Compiling for ESP32 with Xtensa
I can manually avoid this issues by initializing 'value' and 'key' to "".

/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson6150_1100000::VariantData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<std::__cxx11::basic_string<char>, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:339:27: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     variant.setOwnedString(make_not_null(value));
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:335:17: note: 'value' was declared here
     const char *value;
                 ^~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseObject(ArduinoJson6150_1100000::CollectionData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<std::__cxx11::basic_string<char>, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:241:28: error: 'key' may be used uninitialized in this function [-Werror=maybe-uninitialized]
           slot->setOwnedKey(make_not_null(key));
           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson6150_1100000::VariantData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:339:27: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     variant.setOwnedString(make_not_null(value));
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:335:17: note: 'value' was declared here
     const char *value;
                 ^~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseObject(ArduinoJson6150_1100000::CollectionData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:241:28: error: 'key' may be used uninitialized in this function [-Werror=maybe-uninitialized]
           slot->setOwnedKey(make_not_null(key));
           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
/home/lieven/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'Config.o' failed

bblanchon, 2020-03-23:
Hi @vortex314,

I'm not able to reproduce this warning. I've tried:

- GCC 4.4.7
- GCC 4.6.4
- GCC 4.7.4
- GCC 4.8.5
- GCC 4.9.4
- GCC 5.5.0
- GCC 6.5.0
- GCC 7.4.0
- GCC 8.3.0
- GCC 9.2.1
- Clang 3.5.2
- Clang 3.6.2
- Clang 3.7.1
- Clang 3.8.0
- Clang 3.9.1
- Clang 4.0.0
- Clang 6.0.0
- Clang 7.0.0
- Clang 7.1.0
- Clang 8.0.1
- Visual Studio 2010
- Visual Studio 2012
- Visual Studio 2013
- Visual Studio 2015
- Visual Studio 2017
- Visual Studio 2019
- PlatormIO with the following packages:
   - framework-arduinoespressif32 3.10004.200129 (1.0.4)
   - tool-esptoolpy 1.20600.0 (2.6.0)
   - toolchain-xtensa32 2.50200.80 (5.2.0)
- Arduino with ESP32 Core 1.0.4 (toolchain is `xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0`)
- Xtensa toolchain and ESP-IDF 4.1 (I spend a lot of time on this, compilation seems OK, but I couldn't link for some reason)

All builds were made with `-Wall` which includes `-Wmaybe-uninitialize`, according to GCC's documentation

Can you help me reproduce the warning?
Is there anything special about your setup?
What is the version of the compiler?
Do you pass any flag that could trigger this warning?

I don't want to add initializers for `value` and `key` because it increases the code size (+12 bytes on AVR).
Can you help me find another solution?

Best regards,
Benoit

vortex314, 2020-03-23:
Hi Benoit,

thanks for your swift response. Amazing how you manage this, I see that you do a lot of regression testing, like your professionalism.
The project you can find here : https://github.com/vortex314/nanoAkka
version : 
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.

$ which xtensa-esp32-elf-gcc
/home/lieven/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc

I notice I also use : 
#define ARDUINOJSON_USE_LONG_LONG 1
#define ARDUINOJSON_ENABLE_STD_STRING 1

Single command line to give this error : https://github.com/vortex314/Common/blob/master/issue.sh

xtensa-esp32-elf-c++ -std=gnu++11 -Og -ggdb -Wno-frame-address  \
-ffunction-sections -fdata-sections -fstrict-volatile-bitfields \
-mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function \
-Wno-error=unused-but-set-variable -Wno-error=unused-variable \
-Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter \
-Wno-sign-compare  \
-DESP32_IDF=1  \
-I../Common -I/home/lieven/workspace/ArduinoJson/src \
-I /home/lieven/esp/esp-idf/components/freertos/include/freertos \
-fno-rtti -ffunction-sections -fdata-sections -fno-exceptions \
-fno-rtti -DESP_PLATFORM -D IDF_VER=\"v4.2-dev-792-g6330b3345-dirty\" \
 -MMD -MP -DESP32_IDF=1 -DAAAAA=BBBBBB  \
   -I/home/lieven/esp/esp-idf/components/spi_flash/sim/stubs/esp32/include/ \
    -I/home/lieven/esp/esp-idf/components/nvs_flash/include \
    -I/home/lieven/esp/esp-idf/components/xtensa/include/ \
    -I/home/lieven/esp/esp-idf/components/nvs_flash/test_nvs_host/ \
    -I/home/lieven/esp/esp-idf/components/esp_common/include/ \
    -I/home/lieven/esp/esp-idf/components/spi_flash/include/ \
    -I/home/lieven/esp/esp-idf/components/soc/include/ \
    -I/home/lieven/workspace/ArduinoJson/src   -I . -c Config.cpp -o Config.o

Hope this helps
Lieven

vortex314, 2020-03-24:
Apparently it also occurs with linux c++ 
**Shortest way to reproduce** 
1/ **git clone https://github.com/vortex314/Common**
2/ **cd Common**
3/ execute compile
**c++ -std=gnu++11 -Og -ggdb -Wno-frame-address  \
-ffunction-sections -fdata-sections -fstrict-volatile-bitfields \
-nostdlib -Wall -Werror=all -Wno-error=unused-function \
-Wno-error=unused-but-set-variable -Wno-error=unused-variable \
-Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter \
-Wno-sign-compare  \
 -I/home/lieven/workspace/ArduinoJson/src \
-fno-rtti -ffunction-sections -fdata-sections -fno-exceptions \
-fno-rtti  \
 -MMD -MP  \
 -I . \
    -I/home/lieven/workspace/ArduinoJson/src  \
      -c Config.cpp -o Config.o**

==> gives error messages 
In file included from /home/lieven/workspace/ArduinoJson/src/ArduinoJson.hpp:33:0,
                 from /home/lieven/workspace/ArduinoJson/src/ArduinoJson.h:9,
                 from Config.h:12,
                 from Config.cpp:8:
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function ‘ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson6150_1100000::VariantData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]’:
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:339:27: error: ‘value’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
     variant.setOwnedString(make_not_null(value));
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:335:17: note: ‘value’ was declared here
     const char *value;
                 ^~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In member function ‘ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseObject(ArduinoJson6150_1100000::CollectionData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]’:
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:241:28: error: ‘key’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
           slot->setOwnedKey(make_not_null(key));
           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors

bblanchon, 2020-03-24:
From what I see, the `-Og` flag triggers the warning.
I'll try to find a solution.

vortex314, 2020-03-24:
I've tested your fix and this issue is gone, now I notice another one , see below.
I'll see if I can shorten the test case.  
CXX build/main/main.o
In file included from /home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from /home/lieven/workspace/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from /home/lieven/workspace/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /home/lieven/workspace/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /home/lieven/workspace/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/lieven/workspace/nanoAkka/components/Wifi/MqttWifi.h:24,
                 from /home/lieven/workspace/nanoAkka/main/main.cpp:90:
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp: In static member function 'static _Res std::_Function_handler<_Res(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes&& ...) [with _Res = int; _Functor = ToMqtt<T>::ToMqtt(std::__cxx11::string) [with T = long long unsigned int]::<lambda(MqttMessage&, const long long unsigned int&)>; _ArgTypes = {MqttMessage&, const long long unsigned int&}]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:380:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp: In static member function 'static _Res std::_Function_handler<_Res(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes&& ...) [with _Res = int; _Functor = ToMqtt<T>::ToMqtt(std::__cxx11::string) [with T = unsigned int]::<lambda(MqttMessage&, const unsigned int&)>; _ArgTypes = {MqttMessage&, const unsigned int&}]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:380:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp: In static member function 'static _Res std::_Function_handler<_Res(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes&& ...) [with _Res = int; _Functor = ToMqtt<T>::ToMqtt(std::__cxx11::string) [with T = std::__cxx11::basic_string<char>]::<lambda(MqttMessage&, const std::__cxx11::basic_string<char>&)>; _ArgTypes = {MqttMessage&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:380:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp: In static member function 'static _Res std::_Function_handler<_Res(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes&& ...) [with _Res = int; _Functor = ToMqtt<T>::ToMqtt(std::__cxx11::string) [with T = bool]::<lambda(MqttMessage&, const bool&)>; _ArgTypes = {MqttMessage&, const bool&}]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:380:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp: In static member function 'static _Res std::_Function_handler<_Res(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes&& ...) [with _Res = int; _Functor = ToMqtt<T>::ToMqtt(std::__cxx11::string) [with T = int]::<lambda(MqttMessage&, const int&)>; _ArgTypes = {MqttMessage&, const int&}]':
/home/lieven/workspace/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:380:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
/home/lieven/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'main.o' failed
make[2]: *** [main.o] Error 1
make[2]: Leaving directory '/home/lieven/workspace/nanoAkka/build/main'
/home/lieven/esp/esp-idf/make/project.mk:635: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
make[1]: Leaving directory '/home/lieven/workspace/nanoAkka'
Makefile:35: recipe for target 'GPS' failed
make: *** [GPS] Error 2

vortex314, 2020-03-24:
For the last error, it's sufficient to change this line 
			uint8_t _flags=0;
Hope this helps.

bblanchon, 2020-03-24:
Working on it...

closed 2020-03-25 by bblanchon

vortex314, 2020-03-25:
Great work, thanks !

bblanchon, 2020-03-25:
You're welcome.
Thank you very much for opening this issue.

bblanchon, 2020-04-08:
Shipped in [ArduinoJson 6.15.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.1).

commit fc9e609a (2020-03-25): Fixed "maybe-uninitialized" warning (fixes #1217)