bblanchon/ArduinoJson #1218: Determining the JsonDocument that a JsonVariant belongs to

nobodyinperson, 2020-03-23:
Hi!

I just came across a situation where given a `JsonVariant` (or `JsonObject` or `JsonArray`) it would be handy to know which `JsonDocument` it belongs to: When adding content to a JsonObject (e.g. with `createNestedObject()`) doesn't work, I'd like to clear the whole `JsonDocument` and try again.

I searched the online documentation, but [the `JsonVariant` page](https://arduinojson.org/v6/api/jsonvariant/) doesn't reference something alike. I also tried browsing the code but got a little lost...

> I know that [and why](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/) you're not supposed to use ArduinoJson to store application state and it's better to use custom `struct`s for this. However, in my use case I have a nested structure of unknown depth and I want to avoid dynamic memory allocation, so custom `struct`s seem complicated here. Also, flash size and RAM is not (yet) a problem in my ESP8266 setup.

Thank you very much for your awesome work on this library!

Yann

bblanchon, 2020-03-23:
Hi Yann,

Indeed, there is currently no way to test that a `JsonVariant` (or `JsonArray`, or `JsonObject`) belongs to a particular `JsonDocument`.
It should be quite easy to implement, but I wonder if it really makes sense.
Moreover, I don't want to bloat the library if there is an alternative solution.

Can you explain your use case with greater details?

Best regards,
Benoit

nobodyinperson, 2020-03-23:
Hey Benoit,

Thank you for your answer.

My use case is the following: I have a JSON-file in SPIFFS (which I load into a `DynamicJsonDocument`, let's call it _layout_, the only place I'd like to allow dynamic allocation, thank you very much for `shrinkToFit()`!) where I define what sensors are connected in what manner. With bus multiplexers (e.g. I²C), this leads to nested structures (e.g. multiplexer behind multiplexer, etc...). This structure is then traversed periodically and the connected components are handled along the way. To keep track of the last measurement data and access times etc., I fill up a rather flat `StaticJsonDocument`  (let's call it _cache_) in parallel as a key-value storage. I never remove elements from this document, I only add or update values and never change any strings in the _cache_. After successfully traversing the full _layout_, the _cache_'s `memoryUsage()` shouldn't grow anymore. However, as its capacity is fixed, if the _layout_ provides too much cachable data for it to hold (or if I oversee a memory leak), it may occur that the _cache_ gets filled up. In this case, I'd like to clear it and start over. I have a function to get (or create) a `JsonObject` in the _cache_ for a specific component: 

```cpp
JsonObject getCache(const char* id, JsonDocument&doc) { 
  // now get or add the cache object or clear the doc if it's full
  ...
}
```

Furthermore, I have a function to get (or create) a nested `JsonObject` inside such a _cache_ entry (e.g. to store measurement `"data"`):

```cpp
JsonObject getCacheData(const char* id, JsonDocument&doc) { 
  JsonObject cacheElement = getCache(id, doc);
  // now get or add the "data" object or clear the doc if it's full
  ...
}
```

I also have an overload of this function where the `cacheElement` can be passed if it is already known. And this is the place where it would be handy to determine the `JsonDocument` that a `JsonObject` belongs to: It that were possible, these two functions could just take a `JsonObject` as second argument and it wouldn't be necessary to always pass the `JsonDocument` reference around.

Of course there exist alternative solutions in my case: always passing references to the _cache_ or hard-coding it in the functions. But, you know... convenience :-)

This is most likely a corner case and will not benefit many others. I can totally understand if you won't implement it.

Thanks again for this awesome library!

Yann

bblanchon, 2020-03-24:
Hi Yann,

Thank you very much for this detailed explanation; it's really important that I understand how people use the library to provide relevant features.

In this case, however, I think this is too far away from the purpose of the library (ie JSON serialization).
Moreover, it seems that you not only want to test if a `JsonObject` belongs to a `JsonDocument`, you also need to get the `JsonDocument` from the `JsonObject`, which is not possible.

Suppose that I agree to implement the "belong-to" feature.
Even if your program only uses "`JsonObject` belongs-to `DynamicJsonDocument`", I have to implement, test, document, and maintain all the other combinations: `JsonArray`, `JsonObject`, `JsonVariant`, `JsonArrayConst`, `JsonObjectConst`, `JsonVariantConst`, `JsonPair`, `JsonPairConst`, `StaticJsonDocument`, `BasicJsonDocument`.

As you see, the effort on my side is significant, and I'm pretty sure there will be one user at most.
Then, there is a very high probability that you refactor your code and simply forget about this feature, when I would have to continue to support this feature for years to come.

I hope I made my point clear. This feature is too edgy, too costly, and doesn't help enough users. That's why I cannot reasonably implement it.

Thank you for your understanding.

Best regards,
Benoit

nobodyinperson, 2020-03-24:
I totally understand, thank you for your explanations. I'll leave closing this issue to you.

Just one thing:

> However, in my use case I have a nested structure of unknown depth and I want to avoid dynamic memory allocation, so **custom structs seem complicated here**. 

Am I wrong on this one? I know I could implement `struct`s for all structures in the JSON configuration, but the arbitrarily deep nesting makes this really annoying without dynamic allocation in my eyes. The only thing I could think of is to implement an own static memory pool (something like your `StaticJsonDocument` uses) and ”dynamically” allocate memory from there. But that seems like so much effort for effectively the same result (using ArduinoJson directly), at the cost of a little efficiency and RAM (which is both not so imporant in my case). I would just be interested in a quick opinion from a seasoned C++ developer on this, if it's okay :-)

Merci beaucoup!

bblanchon, 2020-03-24:
Can you show what the "layout" JSON looks like?

nobodyinperson, 2020-03-24:
Sure:

```json
{
  "i2c": {
    "bus": {
      "rtc": {
        "type": "ds3231",
        "interval": 60000
      },
      "multiplexer-1": {
        "type": "tca9548A",
        "address": 112,
        "ports": [
          {
            "multiplexer-2": {
              "type": "tca9548A",
              "address": 113,
              "ports": [
                {
                  "bme-3": {
                    "type": "bme280",
                    "address": 118,
                    "interval": 5000
                  }
                }
              ]
            }
          },
          {
            "bme-1": {
              "type": "bme280",
              "address": 119,
              "interval": 5000
            },
            "scd-1": {
              "type": "scd30",
              "pressure": "bme-1",
              "interval": 5000
            },
            "theben-1": {
              "type": "theben",
              "interval": 5000
            }
          },
          {
            "bme-2": {
              "type": "bme280",
              "address": 119,
              "interval": 5000
            },
            "scd-2": {
              "type": "scd30",
              "pressure": "bme-2",
              "interval": 5000
            },
            "theben-2": {
              "type": "theben",
              "interval": 5000
            }
          }
        ]
      }
    }
  },
  "esp": {
    "wifi": {
      "type": "wifi",
      "interval": 5000
    }
  },
  "spi": {
    "bus": {
      "sd-card": {
        "type": "sd",
        "chipselect": 15,
        "interval": 60000,
        "store": [
          "met-2",
          "co2-2",
          "wifi"
        ]
      }
    }
  }
}
```
Looks a little nicer as YAML:

```yaml
i2c:
  bus:
    rtc:
      type: ds3231
      interval: 60000
    multiplexer-1:
      type: tca9548A
      address: 112
      ports:
      - multiplexer-2:
          type: tca9548A
          address: 113
          ports:
          - bme-3:
              type: bme280
              address: 118
              interval: 5000
      - bme-1:
          type: bme280
          address: 119
          interval: 5000
        scd-1:
          type: scd30
          pressure: bme-1
          interval: 5000
        theben-1:
          type: theben
          interval: 5000
      - bme-2:
          type: bme280
          address: 119
          interval: 5000
        scd-2:
          type: scd30
          pressure: bme-2
          interval: 5000
        theben-2:
          type: theben
          interval: 5000
esp:
  wifi:
    type: wifi
    interval: 5000
spi:
  bus:
    sd-card:
      type: sd
      chipselect: 15
      interval: 60000
      store:
      - met-2
      - co2-2
      - wifi
```

bblanchon, 2020-03-25:
I think I would take the full object-oriented approach here.
I would create an abstract class for devices and concrete classes for particular devices.
Finally, I would implement the "Composite" design pattern if it makes sense.

Assuming that the STL is available (it should be on ESP), I would use `std:unique_ptr` (or `std::shared_ptr`, if it's relevant), `std::vector` an `std::map`.

So, yes, I would use heap memory, but I won't be afraid of fragmentation because the allocations are only made during the setup, and not in the main loop.

nobodyinperson, 2020-03-25:
Thank you very much for your input.

I didn't know that the STL can be used on the ESP. However, the code should also run on Arduinos, so probably an STL-less implementation would be nice. I guess it's time to learn some real C++, then...

You may close this issue now.

Cheers,

Yann

bblanchon, 2020-03-25:
Good luck Yann.

closed 2020-03-25 by bblanchon

nobodyinperson, 2020-04-08:
Hi Benoit,

BTW - took me long enough - but I now have implemented an object-oriented approach with one base-class for hardware components from which all other components derive. I learned about `virtual` methods (and that it's annoying that they can neither be `static` nor templated, meh...) and used `std::vector`s and `std::array`s to realize nesting of components. I also managed to implement something superficially similar to `JsonVariant::is<TYPE>()` to check whether a Base-class pointer really is of a specific derived class under the hood.

I can now translate nested `JsonObject`s to my own C++ object structure and can discard the `JsonDocument` afterwards. Traversing this structure and remembering state is indeed more efficient and easier on RAM than using the `JsonObject` directly.

It's a little annoying that there is no `std::function` (for easier callback function handling) in [uClibc++](https://cxx.uclibc.org/index.html) (and thus all the Arduino STL ports), but now my code works natively, on Arduinos and on ESP8266s ([PlatformIO Unit Testing](https://docs.platformio.org/en/latest/plus/unit-testing.html) FTW :grinning:!).

Thank you very much for pointing me in the right direction! I have a feeling, that among programming languages you can learn, C++ has one of the steepest and most painful learning curves... But maybe that's just me coming from Perl/Python/R/etc...

Keep up the good work with `ArduinoJson`! It is an awesome library and a kind of ”beacon” for me, really. The documentation, the website in general and your book is exemplary, I love it! I also provide and maintain several open source software projects (mostly [Python packages](https://pypi.org/user/nobodyinperson/) or [SailfishOS apps](https://openrepos.net/user/7926/programs)) and try to document them as good as possible, I know how time-consuming and hard it can be.

Best regards,

Yann

bblanchon, 2020-04-08:
Thank you very much for this message Yann.
I wish more people reacted this way.

Good luck with your projects!