bblanchon/ArduinoJson #1219: Problem with root.success when parsing the json

anhoang241998, 2020-03-23:
Hi sir, 
I really love your ArduinoJson library. But I am a newbie to this part. I tried to had the serial communication between the **nodemcu ESP8266** and the **Arduino Mega**. I used your library and the SoftwareSerial library. By that, I tried to send a simple json from the **Arduino Mega** to the **nodemcu ESP8266**:
```javascript
{
  "data1": 100,
  "data2": 200
}
```
But I got the error ***parseObject() failed*** in my Serial Monitor:

```
$`{l⸮⸮⸮⸮@⸮⸮⸮parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
parseObject() failed
```
And when I tried to find out what happened, I deleted these line: 
``` C
 if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
```
It worked but some time, It got an empty json like this: `{ }`
My Arduino Mega code:
```C
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  Serial2.begin(9600);
}

 void loop() {
 StaticJsonBuffer<1000> jsonBuffer;
 JsonObject& root = jsonBuffer.createObject();

 root["data1"] = 100;
 root["data2"] = 200;

 root.prettyPrintTo(Serial); 
 root.printTo(Serial2);
 delay(1000);
}
```

bblanchon, 2020-03-24:
Hi @anhoang241998,

It could be a timeout or an error in the transmission.

I really recommend that you use [ArduinoJson 6](https://arduinojson.org/v6/doc/upgrade/) because, unlike version 5, it returns [an error code](https://arduinojson.org/v6/api/misc/deserializationerror/) when something goes wrong.

Best regards,
Benoit

anhoang241998, 2020-03-24:
Hi, @bblanchon,
By following your suggestion, I updated my library to **ArduinoJson 6**, but it still got error:
```
{"data1":100,"data2":200}{"data1":100,"data2":200}deserializeJson() failed: NoMemory
deserializeJson() failed: InvalidInput
{"data1":100,"data2":200}{"data1":100,"data2":200}deserializeJson() failed: NoMemory
deserializeJson() failed: InvalidInput
{"data1":100,"data2":200}{"data1":100,"data2":200}deserializeJson() failed: NoMemory
deserializeJson() failed: InvalidInput
{"data1":100,"data2":200}{"data1":100,"data2":200}deserializeJson() failed: NoMemory
deserializeJson() failed: InvalidInput
```
My **arduino mega** code:
```C
#include <ArduinoJson.h>

void setup() {
Serial1.begin(9600);
}
 
void loop() {
  
const size_t capacity = JSON_OBJECT_SIZE(2);
DynamicJsonDocument doc(capacity);

doc["data1"] = 100;
doc["data2"] = 200;

serializeJson(doc, Serial1);
}
```
My **nodemcu ESP8266** code:
``` C  
#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(4,5);  //D2, D1 = SRX, STX

void setup() {
// Initialize Serial port
Serial.begin(9600);
mySerial.begin(9600);
while (!Serial) continue;
}
 
void loop() {
const size_t capacity = JSON_OBJECT_SIZE(2) + 20;
DynamicJsonDocument doc(capacity);
deserializeJson(doc, mySerial);
DeserializationError error = deserializeJson(doc, mySerial);
if (error) {
Serial.print(F("deserializeJson() failed: "));
Serial.println(error.c_str());
return;
}
int data1 = doc["data1"]; // 100
int data2 = doc["data2"]; // 200
serializeJson(doc, Serial);
}
```

**NOTE:** It's very ridiculous when I added the `Serial.printn();` to the **nodemcu ESP8266** code (in the **ArduinoJson v5**, It worked, but in the **ArduinoJson v6**, It not worked.
This is my code in  **nodemcu ESP8266**:
```C
#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(4,5);  //D2, D1 = SRX, STX
SoftwareSerial s(13,15);
void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  mySerial.begin(9600);
  s.begin(9600);
  while (!Serial) continue;
 
}
 
void loop() {
s.println();  
const size_t capacity = JSON_OBJECT_SIZE(2) + 30;
DynamicJsonBuffer jsonBuffer(capacity);

JsonObject& root = jsonBuffer.parseObject(mySerial);

 if(root == JsonObject::invalid())
  return;

int data1 = root["data1"]; // 100
int data2 = root["data2"]; // 200
if (data1==100){
  Serial.println("good");
}
else if(data1!=100){
  Serial.println("bad");
  delay(1000);
}
}
```

bblanchon, 2020-03-25:
Hi @anhoang241998,

I just tried these programs with two Arduino UNO and they worked fine.
However, I modified the following:

1. I remove the first call to `deserializeJson()`, which seems to be a mistake
2. I increased the speed of `Serial` (not `mySerial`) because it was slowing down the loop.

You must have a much higher baud rate for the Serial Monitor than for the serial communication between the boards. If you don't, then the receiver runs too slowly, then the `SoftwareSerial` buffer overflows, and then you loose bytes.

The speed issue explains the `InvalidInput` error. Regarding the `NoMemory`, it could come from the way bytes are dropped. If this happens again after increasing the speed of the serial connection with the PC, please try to log the input as described in [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).

Best regards,
Benoit

anhoang241998, 2020-03-25:
Hi @bblanchon ,
I just tried your suggestion programs, but it still had these problem, and I didn't know why the `"data2":2000`, I transferred `"data2":200` from the **arduino mega** to the **nodemcu ESP8266**.
This is my ***Serial Monitor***:
```
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
{"data1":100,"data2":2000}deserializeJson() failed: IncompleteInput
```
My ***Arduino Mega*** code:
```C
#include <ArduinoJson.h>

void setup() {
Serial1.begin(9600);
}
 
void loop() {
  
const size_t capacity = JSON_OBJECT_SIZE(2);
DynamicJsonDocument doc(capacity);

doc["data1"] = 100;
doc["data2"] = 200;

serializeJson(doc, Serial1);
delay(1000);
}
```
My ***nodemcu ESP8266*** code:
```C
#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(4,5);  //D2, D1 = SRX, STX

void setup() {
// Initialize Serial port
Serial.begin(38400);
mySerial.begin(9600);
while (!Serial) continue;
}
 
void loop() {
const size_t capacity = JSON_OBJECT_SIZE(2) + 20;
DynamicJsonDocument doc(capacity);

DeserializationError error = deserializeJson(doc, mySerial);
if (error) {
Serial.print(F("deserializeJson() failed: "));
Serial.println(error.c_str());
return;
}
int data1 = doc["data1"]; // 100
int data2 = doc["data2"]; // 200
serializeJson(doc, Serial);
delay(1000);
}
```

bblanchon, 2020-03-26:
Hi @anhoang241998,

[`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) looks like a timeout.
See [How to change the timeout when reading from a Stream?](https://arduinojson.org/v6/how-to/change-the-timeout/).

Regarding the `2000` instead of `200`, I'm not sure...
Please try:
1. Replace `deserializeJson()` with a simple `Serial.readString()`. Do you still see the issue?
2. Replace `serializeJson()` with a simple `Serial.println()`. Do you still see the issue?
3. Change the bauds rate (4800 for example)? Does it change anything?

Best regards,
Benoit

anhoang241998, 2020-03-27:
Hi, @bblanchon 
I had just done your suggestion about changing the bauds rate from 9600 to 4800. It worked perfectly (no more garbage data and `IncompleteInput` error).  
I also tried replacing `serializeJson()` with `Serial.println()` and replacing `deserializeJson()` with `Serial.readString()`. But it not worked. I think just like you said `the receiver (nodemcu ESP8266)` was too slow. Changing the bauds rate can solve this problem. I really thanks for your helping. 

My **result** in the `Serial Monitor`: 
```
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100,"data2":200}
```
My **Arduino Mega** code:
```C
#include <ArduinoJson.h>

void setup() {
Serial2.begin(4800);
}
 
void loop() {
  
const size_t capacity = JSON_OBJECT_SIZE(2);
DynamicJsonDocument doc(capacity);

doc["data1"] = 100;
doc["data2"] = 200;

serializeJson(doc, Serial2);
delay(1000);
}
```
My **nodemcu ESP8266** code:
```C
#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(4,5);  //D2, D1 = SRX, STX

void setup() {
// Initialize Serial port
Serial.begin(9600);
mySerial.begin(4800);
while (!Serial) continue;
}
 
void loop() {
const size_t capacity = JSON_OBJECT_SIZE(2) + 20;
DynamicJsonDocument doc(capacity);

DeserializationError error = deserializeJson(doc, mySerial);
if (error) {
Serial.print(F("deserializeJson() failed: "));
Serial.println(error.c_str());
return;
}
int data1 = doc["data1"]; // 100
int data2 = doc["data2"]; // 200
serializeJson(doc, Serial);
delay(1000);
Serial.println();
}
```
Best regards,
*Ân Nguyễn Hoàng*

bblanchon, 2020-03-27:
You're welcome @anhoang241998.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2020-03-27 by bblanchon