bblanchon/ArduinoJson #1220: Compilation error with NodeMCU ESP32 serializeJson

JonPhno, 2020-03-23:
Hi,

I have a problem when I try to serialize a json object to then send it. Basicly I'm trying to serialize it to a char array I have one error. 

```
  char json[128];
  serializeJson(doc,json);
```
The error:
```
ArduinoJson/Operators/VariantCasts.hpp:16:35: error: invalid conversion from 'ArduinoJson6150_0000010::VariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]

     return impl()->template as<T>();
```

Then I try to serialize it with char * I have another error. I found that putting const char * should fix it but I had no luck.

```
  char* json;
  serializeJson(doc,json);
```
error:
```
  ArduinoJson/Serialization/Writer.hpp:18:26: error: request for member 'write' in '*((ArduinoJson6150_0000010::Writer<char*, void>*)this)->ArduinoJson6150_0000010::Writer<char*, void>::_dest', which is of non-class type 'char*'

     return _dest->write(c);
```
or 
```
  const char* json;
  serializeJson(doc,json);
```
error:
```
ArduinoJson/Serialization/Writer.hpp:18:26: error: request for member 'write' in '*((ArduinoJson6150_0000010::Writer<const char*, void>*)this)->ArduinoJson6150_0000010::Writer<const char*, void>::_dest', which is of non-class type 'const char*'

     return _dest->write(c);

```
Thanks

bblanchon, 2020-03-24:
Hi @JonPhno,

The following code is correct and produces no error:

```c++
char json[128];
serializeJson(doc, json);
```

The error `invalid conversion from...` comes from another line in your program.
Read the rest of the compiler output; it will ultimately point to this line.

See also:

* [invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v6/error/invalid-conversion-from-const-char-to-char/)

Best regards,
Benoit

JonPhno, 2020-03-24:
Ah yes, the error came from the desirialize and I was able to make it work with the link.

Thank you!

closed 2020-03-24 by JonPhno

bblanchon, 2020-03-24:
You're welcome @JonPhno!
Good luck with your project.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)