bblanchon/ArduinoJson #1221: [Question] How to optimize my code for a smaller footprint?

effelle, 2020-03-24:
Benoît,

I have all my code in place and is working fine for all the type of JSON I must process but sadly the space occupied is really big: it takes 902 bytes. 
This is the complete void

```php
void HAssAnnounceSensors(void)
{
  uint8_t hass_xsns_index = 0;
  do
  {
    mqtt_data[0] = '\0';
    int tele_period_save = tele_period;
    tele_period = 2;                                 // Do not allow HA updates during next function call
    XsnsNextCall(FUNC_JSON_APPEND, hass_xsns_index); // ,"INA219":{"Voltage":4.494,"Current":0.020,"Power":0.089}
    tele_period = tele_period_save;

    char sensordata[512]; // Copy because we need to write to mqtt_data
    strlcpy(sensordata, mqtt_data, sizeof(sensordata));

    if (strlen(sensordata))
    {
      sensordata[0] = '{';
      snprintf_P(sensordata, sizeof(sensordata), PSTR("%s}"), sensordata); // {"INA219":{"Voltage":4.494,"Current":0.020,"Power":0.089}}
      // USE THE FOLLOWING LINE TO TEST JSON
      //snprintf_P(sensordata, sizeof(sensordata), PSTR("{\"HX711\":{\"Weight\":[22,34,1023.4]}}"));
      //snprintf_P(sensordata, sizeof(sensordata), PSTR("{\"TX23\":{\"Speed\":{\"Act\":8.6,\"Avg\":8.2,\"Min\":0,\"Max\":15.8},\"Dir\":{\"Card\":\"SSO\",\"Deg\":157.5,\"Avg\":145.5,\"AvgCard\":\"SO\",\"Min\":112.5,\"Max\":292.5,\"Range\":180}}}"));

      StaticJsonBuffer<500> jsonBuffer;
      JsonObject &root = jsonBuffer.parseObject(sensordata);
      if (!root.success())
      {
        AddLog_P2(LOG_LEVEL_ERROR, PSTR("HASS: jsonBuffer failed to parse '%s'"), sensordata);
        continue;
      }
      for (auto sensor : root)
      {
        const char *sensorname = sensor.key;
        JsonObject &sensors = sensor.value.as<JsonObject>();
        if (!sensors.success())
        {
          AddLog_P2(LOG_LEVEL_ERROR, PSTR("HASS: JsonObject failed to parse '%s'"), sensordata);
          continue;
        }

        for (auto subsensor : sensors)
        {
          if (subsensor.value.is<JsonObject&>()) {
            // If there is a nested json on sensor data, second level entitites will be created
            char NestedName[20];
            char NewSensorName[20];
            snprintf_P(NestedName, sizeof(NestedName), PSTR("%s"), subsensor.key);
            JsonObject& subsensors = subsensor.value.as<JsonObject>();
            for (auto subsensor : subsensors) {
              snprintf_P(NewSensorName, sizeof(NewSensorName), PSTR("%s %s"), NestedName, subsensor.key);
              HAssAnnounceSensor(sensorname, NestedName, NewSensorName, 0, 0, 1, subsensor.key);
            }              
          } else if (subsensor.value.is<JsonArray&>()) {           
            // If there is more than a value on sensor data, 'n' entitites will be created
            JsonArray& subsensors = subsensor.value.as<JsonArray&>();
            uint8_t subqty = subsensors.size();
            char MultiSubName[20];
            for (int i = 1; i <= subqty; i++) {
              snprintf_P(MultiSubName, sizeof(MultiSubName), PSTR("%s %d"), subsensor.key, i);
              HAssAnnounceSensor(sensorname, subsensor.key, MultiSubName, i, 1, 0, subsensor.key);
            }
          } else { HAssAnnounceSensor(sensorname, subsensor.key, subsensor.key, 0, 0, 0, subsensor.key);}
        }
      }
    }
    yield();
  } while (hass_xsns_index != 0);
}
```
Every time I use `value.as` the code increase at least 140 bytes and I'm starting to think I'm not processing the data as I should. Any suggestion will be really appreciated.

Thanks,

Federico

bblanchon, 2020-03-25:
Hi Federico,

Optimizing code size is an art!
It often comes with **compromises** on the code design and sometimes on the code behavior too.

For the sake of code size, ArduinoJson isn't very strict with the input and will accept non-conformant JSON documents. That's an example of a compromise.
Try to see, in your application, where you can add some slack in the specification.
That's for the "strategic" approach to code optimization.

Now, for the "tactical" side, here are a few things I can see:

1. Remove all the log
2. If you remove the log, you can drop several calls to `.success()`
3. Replace `.is<T>()`/`.as<T>()` with `.as<T>()`/`.success()`
4. Replace `if(strlen(sensordata))`  with `if(sensordata[0])`, or use the result of `strlcpy()`
5. Try to replace short Flash strings with regular strings
6. Try to replace `snprintf_P()` with `strcat()` and `strcpy()`
5. Try ArduinoJson 6

Best regards,
Benoit

effelle, 2020-03-25:
Yes Benoît, 
optimizing the code is really an art, especially on C. 

The logs are important to catch the error in JSON but I will probably remove them.
Can't implement AJ6 as the project owner (TASMOTA)  prefer to stay on AJ5 for a series of reasons.
I will tests all the rest of the suggestion as soon as possible but you gave me some good points to start optimizing.

Thanks,

Federico

bblanchon, 2020-03-25:
You're welcome.
Do you know why they don't want to upgrade ArduinoJson?

effelle, 2020-03-25:
Don't know for sure, I think the first cause was the size of the library and there was something about the need of a "fixed" dynamic size. 
Please, take it with a grain of salt, I cover mainly the integrations not the core code, perhaps there are more reasons and I'm not comfortable talking about things I don't know well.

bblanchon, 2020-03-26:
OK.
Don't hesitate to mention me (@bblanchon) in a conversation if you think I can help.

closed 2020-03-26 by bblanchon