bblanchon/ArduinoJson #1223: Problems updating more complex code from ArduinoJson 5 to ArduinoJson 6

GiampaMan, 2020-03-27:
Hi! 

I'm beginner and I need some help. I will appreciate it so much.
Recently I was updating my code of an Arduino project using your v5 library. Now that I decided to move to v6, all was going good until I arrived to the code that I put below. It's like a function, but don't know how to do the fixes to run Json part under v6.
```
JsonObject &getSchedules(DynamicJsonBuffer *jsonBuffer)
{
    String reqHeader = "";
    reqHeader += ("GET " + sch_url + " HTTP/1.1\r\n");
    reqHeader += ("Host: " + String(host) + ":" + String(httpsPort) + "\r\n");
    reqHeader += ("Authorization: Bearer "+ urlencode(deviceApi)+"\r\n");
    reqHeader += ("deviceids: "+urlencode(deviceID)+"\r\n"); 
    reqHeader += ("Connection: close\r\n");
    reqHeader += ("\r\n\r\n");
    
    return getRequest(host, httpsPort, reqHeader, jsonBuffer);
}

```

I hope you have a nice day! Take care.

Regards,
Giampaolo

bblanchon, 2020-03-27:
Hi @GiampaMan,

There isn't much to say about this code because most of the work is done by `getRequest()`.
Please see the [migration guide](https://arduinojson.org/v6/doc/upgrade/) for details.

Best regards,
Benoit

GiampaMan, 2020-03-27:
> Hi @GiampaMan,
> 
> There isn't much to say about this code because most of the work is done by `getRequest()`.
> Please see the [migration guide](https://arduinojson.org/v6/doc/upgrade/) for details.
> 
> Best regards,
> Benoit

Hi!

Thank you for your reply. I read that page yesterday (the migration guide) now I tested an idea that I had of what I was doing wrong to solve it and worked.

But now I have another problem, whats the best way to convert this to v6?
`JsonArray scheduleArray = schApiResponse->get<JsonArray>("s");` that´s v5. What do you think? I read the migration guide but don't now what to do, is not compiling for me.

And another question, how can I convert this 2 lines into v6 sintax?: 
`return jsonBuffer->createObject();`
`return jsonBuffer->parseObject(client);`

Sorry for the questions, I'm a beginner in process of learning. I appreciate your help. thanks a lot!

Regards,
Giampaolo

bblanchon, 2020-03-28:
Hi Giampaolo,

The best way to deal with the three statements above is to get rid of the pointer.
I don't think it ever makes sense to manipulate a `JsonDocument` via a pointer.

Once the pointer is removed, the code should be straightforward:

```c++
JsonArray scheduleArray = schApiResponse["s"];

return doc.to<JsonObject>();

deserializeJson(doc, client);
return doc.as<JsonObject>();
```

Best regards,
Benoit

GiampaMan, 2020-03-30:
> Hi Giampaolo,
>
> The best way to deal with the three statements above is to get rid of the pointer.
> I don't think it ever makes sense to manipulate a `JsonDocument` via a pointer.
>
> Once the pointer is [...]

Hi @bblanchon 
Thanks again for your reply. Thanks for all the help!

I tested removing the pointer but other code part is having problems (I think in this case the pointer should be there) supposing that we need to keep the pointer, how could be the best way to do that. 
Thanks for all. Without your help, this fixing couldn't be possible with my knowledge. 

Have a nice day and sorry for the delays responding.

Regards,
Giampaolo

bblanchon, 2020-04-01:
> other code part is having problems

Fix the other part.

> I think in this case the pointer should be there

I'm the author of the library; I know how it should be used.

If you don't listen to my advice, please don't ask for it.

closed 2020-04-08 by bblanchon