bblanchon/ArduinoJson #1224: Question: Partial Update of struct?

lbussy, 2020-03-28:
Hello Benoît,

Taking, for example, your `sketchbook6\SpiffsConfig` code again, I am wondering if there's an easy way to apply a partial update to the struct?

The flow would be something like:

1. POST a JSON message with `{"apconfig": {"ssid": "foo"}}`
2. Deserialize as `doc`
3. Load the JSON with ` config.load(doc.as<JsonObject>());`

Fairly certain that's a valid way to go - however, my logic in `load()` has some logic to establish default values if the values are null.  I believe that `doc["apconfig"]["passphrase"]` would be null in this case?  Or is there another way to determine that I should skip fields that are not present?

The end result would be an endpoint for the configuration of individual points.  I currently have a long string of `if (data->hasParam(F("ssid"), true))` statements but that seems unglamorous.

bblanchon, 2020-03-29:
Hi Lee,

I don't think I would use `load()` for this feature.
Instead, I would create a `patch()` method that would only apply diffs.

Alternatively, you could read the JSON file from disk, then [merge it](https://arduinojson.org/v6/how-to/merge-json-objects/) with the content of the message, and call `load()` as before.

Best regards,
Benoit

lbussy, 2020-03-29:
Brilliant!  merge is exactly what I was looking for.  Thank you, sir!  I even re-read the book last night - didn't see merge in there.

Thanks again for your assistance!

closed 2020-03-29 by lbussy