bblanchon/ArduinoJson #1225: Unable to acces elements in array

LocoMatt, 2020-03-28:
Hello, I've playing with data processing using ArduinoJson with ESP32 and I noticed problem with accesing "lower-level" elements in array. 
I've got Json data from server with such structure (it's quite long so I'm putting it on pastebin)
https://pastebin.com/iQmpHtEn - (https://rapidapi.com/api-sports/api/covid-193)
And now I want to acces in each response element cases -> active 
So I'm doing it like
`obj["response"][i]["cases"]["active"]`
Where **i** is number of element. For some reason I can only get acces to first 3 elements. 
Here's code for reference and testing. 

```
#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include "ArduinoJson.h"

const char *ssid = "ssid";
const char *password = "password";

String addr = "https://covid-193.p.rapidapi.com/history?country=";
int records_num; //number of records 

//connect to wifi network
void network_connection_setup()
{
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }

  Serial.println("Connected to the WiFi network");
}

//download and print data
void get_api_history_data(String country)
{
  if ((WiFi.status() == WL_CONNECTED))
  {
    HTTPClient http;

    http.begin(addr + country);                                            //Specify the URL
    http.addHeader("x-rapidapi-key", "e80de52abcmshe7ca5342101f11bp1e6891jsn17bd7b9061b1"); //get header api:key

    int httpCode = http.GET(); //Make the request

    if (httpCode > 0)
    {                                    
      DynamicJsonDocument doc(1024);     
      String payload = http.getString(); 
      deserializeJson(doc, payload);     
      JsonObject obj = doc.as<JsonObject>();
      Serial.println(payload); //print payload to make sure everyting is fine 
      records_num = obj["results"]; //return number of records 
      Serial.println(records_num);
      int total_cases[records_num]; //make an array for records 

      for (int i = 0; i < records_num; i++)
      {
        total_cases[i] = obj["response"][i]["cases"]["active"]; 
        Serial.println(total_cases[i]);
      }
    }

    else
    {
      Serial.println("Error on HTTP request");
    }

    http.end();
  }
}

void setup()
{
  Serial.begin(115200);
  network_connection_setup();
  get_api_history_data("china");
}

void loop()
{
}
```
Here's example output 

```
Connecting to WiFi..
Connected to the WiFi network
{"get":"history","parameters":{"country":"china"},"errors":[],"results":8,"response":[{"country":"China","cases":{"new":"+54","active":3128,"critical":886,"recovered":74971,"total":81394},"deaths":{"new":"+3","total":3295},"day":"2020-03-28","time":"2020-03-28T21:00:03+00:00"},{"country":"China","cases":{"new":"+55","active":3460,"critical":1034,"recovered":74588,"total":81340},"deaths":{"new":"+5","total":3292},"day":"2020-03-28","time":"2020-03-28T00:15:05+00:00"},{"country":"China","cases":{"new":"+67","active":3947,"critical":1235,"recovered":74051,"total":81285},"deaths":{"new":"+6","total":3287},"day":"2020-03-27","time":"2020-03-27T00:45:06+00:00"},{"country":"China","cases":{"new":"+47","active":4287,"critical":1399,"recovered":73650,"total":81218},"deaths":{"new":"+4","total":3281},"day":"2020-03-26","time":"2020-03-26T00:30:05+00:00"},{"country":"China","cases":{"new":"+78","active":4735,"critical":1573,"recovered":73159,"total":81171},"deaths":{"new":"+7","total":3277},"day":"2020-03-25","time":"2020-03-25T01:45:06+00:00"},{"country":"China","cases":{"new":"+39","active":5120,"critical":1749,"recovered":72703,"total":81093},"deaths":{"new":"+9","total":3270},"day":"2020-03-24","time":"2020-03-24T01:00:07+00:00"},{"country":"China","cases":{"new":"+46","active":5353,"critical":1845,"recovered":72440,"total":81054},"deaths":{"new":"+6","total":3261},"day":"2020-03-23","time":"2020-03-23T01:30:06+00:00"},{"country":"China","cases":{"new":null,"active":6013,"critical":1927,"recovered":71740,"total":81008},"deaths":{"new":null,"total":3255},"day":"2020-03-22","time":"2020-03-22T01:00:05+00:00"}]}
8
3128
3460
3947
0
0
0
0
0
```

I'm unsure I'm doing it wrong way or maybe it's more library based issue.

Even if I tried with manual accesing like 
`obj["response"][3]["cases"]["active"];`
It returns 0
When I try to get is .as<String>() it returns null

bblanchon, 2020-03-29:
Hi @LocoMatt,

Please check the result of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), it probably says [NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory).
You need to increase the capacity of the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/).

See also:

* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)
* [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)

Best regards,
Benoit

LocoMatt, 2020-03-29:
Thank you for quick answer! Just tested, it returned "no memory", increasing memory amout helped. Bit strange because I tried it before, but for some reason it didn't work.  Maybe I missed something? But I think it's because the json file I actually need is a bit "too big" 100s of elements.   I think I can close it.

closed 2020-03-29 by LocoMatt