bblanchon/ArduinoJson #1226: Suggestion for next version.

Michel-Ketelaars, 2020-03-29:
It would be nice if the functions / methods serialzeJson() and deserializeJson() would be extended, so that it is possible to use them directly on nested Objects (and not only on the Json document).
What I mean is an extra function overloadings like this:
`size_t serializeJson(const JsonObject& object, char output[size]);`
`deserializeJson(JsonObject& object, char* input);`

For my MQTT project it would be very handy, because I have to serialise a portion of a json object.

bblanchon, 2020-03-29:
Hi @Michel-Ketelaars,

The first is already supported, you can do that for example:

```c++
serializeJson(doc["config"], configFile);
```

Can you detail the use-case for `deserializeJson(JsonObject&,...)` ?

Best regards,
Benoit

Michel-Ketelaars, 2020-03-29:
Hi Benoit,

Thanks for your reply and good to hear that this is already supported for the serialization part. 

I indeed tested only the deserializeJson() function and assumed this was also not implemented in serializeJson() function, because I did not saw the particular overloaded function under Signatures.
(https://arduinojson.org/v6/api/json/serializejson/)

The use case is that I receive a Json serialized message from an MQTT server with several values that I want to add or change into an existing Json opject that is afterward is completely published to a webservice.

Best regards,
Michel

bblanchon, 2020-04-01:
> I want to add or change into an existing Json object

If you do this repeatedly, it will create [a memory leak](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/).
How do you plan to mitigate this?

bblanchon, 2020-09-20:
@Michel-Ketelaars, would #1343 provide an acceptable solution to your use case?

bblanchon, 2023-07-29:
I just implemented this feature in branch [7.x](https://github.com/bblanchon/ArduinoJson/tree/7.x).

commit 43eed00c (2023-07-29): Add `deserializeXxx(JsonVariant, ...)` (resolves #1226)
closed 2023-07-29 by bblanchon