bblanchon/ArduinoJson #1229: Sometimes garbage result

Hans-Zwiesel, 2020-04-02:
Version 6.15.0

WiFiBuffer = "{"name": "Robo Max", "id": "53C82A", "status": {"status": 4, "stopped": false, "duration": 278, "mode": 2, "battery": 83, "hours": 109}, "timer": {"status": 0}, "blades": {"quality": 5, "hours": 95, "days": 26}, "wlan": {"signal": -86}, "health": {"temperature": 24, "humidity": 0}, "error": {"error_code": 0, "error_message": "Robo Fehlercode!", "date": "2018-05-03", "time": "13:31:55", "unix": 1525354315}, "clock": {"date": "2018-05-04", "time": "10:05:41", "unix": 1525428341}, "successful": true}";

const size_t capacity = 2*JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 2*JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(7) + JSON_OBJECT_SIZE(9) + 630;

 DynamicJsonDocument root(capacity);
 DeserializationError Jerror = deserializeJson(root, WiFiBuffer);

RoDa_successful = root["successful"]; 
RoDa_name =       root["name"];        

After Parsing:
RoDa_name is sometimes "Robo Max",  sometimes "⸮?6", "⸮?6", ""

What happened?

bblanchon, 2020-04-03:
Hi Hans,

Please read:

* [Why does the output contain garbage?](https://arduinojson.org/v6/faq/why-does-the-output-contain-garbage/)

Best regards,
Benoit

Hans-Zwiesel, 2020-04-03:
Hi Benoit,
many thanks for the quick answer.
I copy this from your template at: https://arduinojson.org/v6/assistant/

I don't have much experience with objects, so it would very kind if you could improve my code
written above.

Many thanks,
All the best,
Hans

bblanchon, 2020-04-08:
Hi Hans,

You can learn from the [documentation on arduinojson.org](https://arduinojson.org/v6/doc/), or from the book [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

Hans-Zwiesel, 2020-04-09:
Hi Benoit,
many thanks for the tipp, but i switched from poll/parsing to mqtt.

closed 2020-04-09 by bblanchon