bblanchon/ArduinoJson #1230: uninitialized error similar to #1217

DasVoelkel, 2020-04-03:
Running platformio and esp-idf V4, I get an error i believe the same one as #1217.
I wanted to add ArduinoJson to my project using the single header, is the previous fix not yet in that file?
```
    DynamicJsonDocument doc(1024);
    const char *json = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.75608,2.302038]}";
    DeserializationError err = deserializeJson(doc, json);
    const char *event = NULL;

    event = doc["sensor"];
```
produces:
```

src/ArduinoJson.h: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson6150_1100000::VariantData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
src/ArduinoJson.h:5135:27: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     variant.setOwnedString(make_not_null(value));
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
src/ArduinoJson.h:5131:17: note: 'value' was declared here
     const char *value;
                 ^~~~~
Indexing .pio/build/esp32dev/esp-idf/app_trace/libapp_trace.a
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/bootloader_common.c.o
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/bootloader_flash.c.o
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/bootloader_flash_config.c.o
In file included from include/mg_protocoll.hpp:5,
                 from src/mg_protocoll.cpp:1:
src/ArduinoJson.h: In function 'bool mg::mg_protocoll::dispatch_command(std::__cxx11::string*)':
src/ArduinoJson.h:1695:12: error: '*((void*)(& doc)+24).ArduinoJson6150_1100000::VariantData::_flags' is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/bootloader_random.c.o
In file included from include/mg_protocoll.hpp:5,
                 from src/mg_protocoll.cpp:1:
src/ArduinoJson.h: In member function 'ArduinoJson6150_1100000::DeserializationError ArduinoJson6150_1100000::JsonDeserializer<TReader, TStringStorage>::parseObject(ArduinoJson6150_1100000::CollectionData&, TFilter, ArduinoJson6150_1100000::NestingLimit) [with TFilter = ArduinoJson6150_1100000::AllowAllFilter; TReader = ArduinoJson6150_1100000::Reader<const char*, void>; TStringStorage = ArduinoJson6150_1100000::StringCopier]':
src/ArduinoJson.h:5068:28: error: 'key' may be used uninitialized in this function [-Werror=maybe-uninitialized]
           slot->setOwnedKey(make_not_null(key));
           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/bootloader_utility.c.o
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/esp_image_format.c.o
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/flash_encrypt.c.o
cc1plus: some warnings being treated as errors
Compiling .pio/build/esp32dev/esp-idf/bootloader_support/src/flash_partitions.c.o
*** [.pio/build/esp32dev/esp-idf/src/mg_protocoll.cpp.o] Error 1
```

bblanchon, 2020-04-03:
Hi @DasVoelkel,

This warning is fixed by fc9e609ab5eab5cccc9cc6e947be8cf6d057e8d4, which I commited after 6.15.0.
Please use [this header](https://github.com/bblanchon/ArduinoJson/files/4428557/ArduinoJson-v6.15.0-4-gfc9e609a.zip) until 6.15.1 is out.

Best regards,
Benoit

DasVoelkel, 2020-04-03:
Thank you ! i suspected it was a duplicate ^^
Awesome library ! Thanks for the fast help !

closed 2020-04-03 by DasVoelkel