bblanchon/ArduinoJson #1231: Runtime crash with 6.15.0

labodj, 2020-04-04:
Hi, I think I have found a bug in 6.15 version, the problem doesn't occur in 6.14.1, i'm compiling with GCC 9.2.0 (https://blog.zakkemble.net/avr-gcc-builds/) using platformio and VSCode under linux 64bit for Controllino Maxi (ATmega2560).

I have this method that prepares a json to be sent:

```c++
void Serializer::serializeNetworkClick(uint8_t buttonIndex, constants::ButtonClickType clickType, bool confirm) const
{
    suing namespace constans;
    StaticJsonDocument<JSON_OBJECT_SIZE(4)> doc;

    // Write the Json
    doc[JSON_PAYLOAD] = serialized(JSON_PAYLOAD_COMMAND_NETWORK_CLICK_SERIALIZED);

    switch (clickType)
    {
    case ButtonClickType::LONG:
        doc[JSON_DATA_CLICK_TYPE] = serialized(JSON_DATA_CLICK_TYPE_LONG_SERIALIZED);
        break;
    case ButtonClickType::SUPER_LONG:
        doc[JSON_DATA_CLICK_TYPE] = serialized(JSON_DATA_CLICK_TYPE_SUPER_LONG_SERIALIZED);
        break;
    default:
        return;
    }

    doc[JSON_DATA_BUTTONS_IDS] = this->device.getButton(buttonIndex)->getId();
    doc[JSON_DATA_CONFIRM] = confirm;

    // Send the Json
    this->com->sendJson(doc);
}
```

Useful related code:
Merged from various constants .hpp files
```c++
namespace constants
{
static constexpr const char *JSON_PAYLOAD = "p";
static constexpr const char *JSON_DATA_CLICK_TYPE = "ct";
static constexpr const char *JSON_DATA_BUTTONS_IDS = "bi";
static constexpr const char *JSON_DATA_CONFIRM = "c";

static constexpr const char *JSON_PAYLOAD_COMMAND_NETWORK_CLICK_SERIALIZED = "\"c_nc\"";
static constexpr const char *JSON_DATA_CLICK_TYPE_LONG_SERIALIZED = "\"lc\"";
static constexpr const char *JSON_DATA_CLICK_TYPE_SUPER_LONG_SERIALIZED = "\"slc\"";

enum class ButtonClickType
{
    NONE,
    SHORT,
    LONG,
    SUPER_LONG
};
} // namespace constants
```

from device.cpp
```c++
Clickable *Device::getButton(uint8_t buttonIndex) const
{
    return this->buttons[buttonIndex]; // this->buttons is a std::vector<Clickable *>
}
```

from clickable.cpp
```c++
const char *Clickable::getId() const
{
    return this->id; // this->id is a const char *const
}
```

from espcom.cpp
```c++
void EspCom::sendJson(const JsonDocument &json)
{
    serializeJson(json, (*this->serial)); // this->serial is a HardwareSerial *const
    DJ(json);
    this->storeSentPayloadTime();
}
```

from debug.hpp
```c++
#define DJ(x)                     \
    do                            \
    {                             \
        serializeJson(x, Serial); \
        Serial.println();         \
    } while (0)

```

The problem is that when all 4 fields are set the program crash at runtime.
If any of these 4 fields are removed the program run fine.
I tried to use larger StaticJsonDocuments without success.

At the moment I'll stay with 6.14.1 until I find a fix or a workaround.
Another question: is serialized used like the code above actually faster than passing a normal (not preformatted) static constexpr const char *?
like
```c++
doc[JSON_PAYLOAD] = JSON_PAYLOAD_COMMAND_NETWORK_CLICK);
```

```c++
namespace constants
{
static constexpr const char *JSON_PAYLOAD_COMMAND_NETWORK_CLICK = "c_nc";
} // namespace constants
```

bblanchon, 2020-04-04:
Hi @labodj,

Thank you very much for reporting this issue.

Unfortunately, I'm not able to reproduce the error.
After reading your message, I conclude that the following program should be enough to reproduce the bug:

```c++
StaticJsonDocument<JSON_OBJECT_SIZE(4)> doc;
    
doc[JSON_PAYLOAD] = serialized(JSON_PAYLOAD_COMMAND_NETWORK_CLICK_SERIALIZED);
doc[JSON_DATA_CLICK_TYPE] = serialized(JSON_DATA_CLICK_TYPE_LONG_SERIALIZED);
doc[JSON_DATA_BUTTONS_IDS] = "1234";
doc[JSON_DATA_CONFIRM] = true;
    
serializeJson(doc, Serial);
```

But it's not.
Check this demo with GCC 9.3.0:
https://wandbox.org/permlink/yA5Hq0BUGtupvgd3

However, the following line caught my attention:

```c++
this->device.getButton(buttonIndex)->getId()
```

This line calls `getId()` on a *possibly* temporary `Clickable` returned by `getButton()`, so `getId()` could return a pointer from a temporary object, and, therefore, `doc` could store a dangling pointer.
It may not be a problem; it entirely depends on the way you implemented `Clickable` and `getButton()`.

It could also be a stack overrun, but I don't think so. How much free RAM do you have?

Best regards,
Benoit

labodj, 2020-04-05:
Hi @bblanchon and thanks for the fast reply.
I have spended many hours to find the clue of this bug.

Since I print a lot over the Serial (for debug using a PC) and Serial2 (to send JSONs to another controller) I tried to call a `Serial.flush()` **few line before** calling the method that caused the crash.
This solved the problem, but if I call `Serial.flush()` **next to** method call the crash happens.

_Very very strange, maybe is related to Serial use?_

Then I tried to build in release mode instead of debug one, problem solved, even without `Serial.flush()`, in release mode I only use Serial2 to send JSONs with all debug disabled and Serial never "beginned".

_Again... is this problem related to Serial?_

Then I checked your repo for recent changes and... I found that my problem was related with this commit: 5b812522faa992fff94e8bffff41fa9b03709cf7

Since I use platformio I set `build_type = debug` for debug builds and with ArduinoJson 6.15.0 that flag set ArduinoJson libray to be built in debug mode.

I fixed my problem setting `-D ARDUINOJSON_DEBUG=0` in my platformio.ini file.

I don't have any clue on what is causing the real problem when building this library in debug mode.

Hope this helped, feel free to ask any question.

Best regards,
Jacopo

bblanchon, 2020-04-06:
Hi Jacopo,

Thank you very much for taking the time to analyses this bug.

This is the second time we got an inexplicable crash with the debug mode.
The first time was on ESP32 (issue #1210) but 6.15 made the bug disappear (probably temporarily).

`ARDUINOJSON_DEBUG` adds some assertions to the code, so the program is probably crashing because of a failing assertion, but I don't know why it prints nothing of the serial port.

Anyway, there must be an undefined behavior somewhere in ArduinoJson and I have to fix it.

Can you create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) that I can run here?

Best regards,
Benoit

labodj, 2020-04-06:
Hi @bblanchon 
I've just finished trimming all unnecessary code to reproduce the bug.

main.cpp:

```c++
#include "Arduino.h"
#include "ArduinoJson.h"

class CrashClass
{
public:
    void crashTest();
};

void CrashClass::crashTest()
{
    StaticJsonDocument<JSON_OBJECT_SIZE(1)> doc;
    doc["c"] = true;
    serializeJson(doc, Serial);
}

void setup()
{
    Serial.begin(115200);
    Serial.println("Setup end");
}

void loop()
{
    static CrashClass crash;
    static uint32_t timer = 0;

    if (millis() - timer > 1000)
    {
        Serial.println("I'm working");
        crash.crashTest();
        timer = millis();
    }
}
```

platformio.ini:
```ini
[platformio]
default_envs = debug

[env]
platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino
board =  megaatmega2560
upload_port = /dev/ttyACM?
monitor_port = /dev/ttyACM?
monitor_speed = 115200
lib_deps = ArduinoJson@6.15.0

[common_settings]
build_flags = -D ARDUINOJSON_DEBUG=1

[env:release]
build_flags = ${common_settings.build_flags}

[env:debug]
build_type = debug
build_flags = ${common_settings.build_flags}
```

Compiled with GCC 9.3.0 (https://blog.zakkemble.net/avr-gcc-builds/comment-page-1/#comment-209216) for Arduino Mega 2560

Expected output:

> Setup end
> I'm working
> {"c":true}I'm working
> {"c":true}I'm working
> {"c":true}I'm working
> ...

Actual output:

> Setup end
> I'm

When ` ARDUINOJSON_DEBUG=0` is set it works just fine

bblanchon, 2020-04-07:
Hi Jacopo,

I don't own an Arduino Mega 2560, so I ran this code on an ATmega328.
Unfortunately, it worked like a charm.

I made sure that I compiled with the (shady) GCC 9.3 you provided.
I also made sure that the build was in debug mode, and verified the value of `ARDUINOJSON_DEBUG`.

It seems that the program above can be further simplified to:

```c++
#include "Arduino.h"
#include "ArduinoJson.h"

void setup()
{
    Serial.begin(115200);
    Serial.println("Setup end");
}

void loop()
{
    static uint32_t timer = 0;

    if (millis() - timer > 1000)
    {
        Serial.println("I'm working");
        StaticJsonDocument<JSON_OBJECT_SIZE(1)> doc;
        doc["c"] = true;
        serializeJson(doc, Serial);
        timer = millis();
    }
}
```

What happens in this case?

The only way I managed to block the program is by manually inserting a failing assert:

```c++
assert(false);
```

Unfortunately, it doesn't show any information.
Can you edit the file [src/ArduinoJson/Polyfills/assert.hpp](https://github.com/bblanchon/ArduinoJson/blob/6.x/src/ArduinoJson/Polyfills/assert.hpp#L7-L14) and replace the content with:

```c++
#define ARDUINOJSON_ASSERT(X)         \
  if (!(X)) {                         \
    Serial.println("ASSERT FAILED!"); \
    Serial.print(__FILE__);           \
    Serial.print(":");                \
    Serial.print(__LINE__);           \
    Serial.print(": ");               \
    Serial.print(#X);                 \
    while (true) continue;            \
  }
```

Hopefully, this will tell us which assert fails.

Best regards,
Benoit

labodj, 2020-04-07:
Hi,
at the moment I'm at work but I can do all the tests this afternoon.

I should (I'm not sure) have an Arduino Uno laying around, if I find it I'll try even with it.

BTW If you remove the CrashClass and call those instructions directly in the loop() the code runs fine, it only crashes when it's called via a method, very odd...

bblanchon, 2020-04-07:
Very odd indeed, but there is a difference between the two calls: the calling convention.

bblanchon, 2020-04-07:
Thanks, @labodj.

I now understand what's going on: my debug code checks that the pointers are correctly aligned, but the AVR architecture doesn't require the words to be aligned, so sometimes, it gives a non-aligned memory pool.

I used to [disable the alignment check back in v5](https://github.com/bblanchon/ArduinoJson/blob/5.x/src/ArduinoJson/Configuration.hpp#L124-L127), but I removed this check for some reason.
Now, it pops up because the Debug mode enables the assertions.

I simply need to disable the alignment checks on AVR.
Meanwhile, you can set `ARDUINOJSON_DEBUG` to 0.

closed 2020-04-08 by bblanchon

bblanchon, 2020-04-08:
Shipped in [ArduinoJson 6.15.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.1)

commit 67e78f07 (2020-04-08): Disabled alignment on AVR (fixes #1231)