bblanchon/ArduinoJson #1232: createNestedArray is clearing reference

pixelfriese, 2020-04-05:
Hi,

in my example I have a data structure like the following:

```json
{
    "header": {
        "mac": "xxxxxxxxxxxxxxxxxxx",
        "version": "1.0.1",
        "generated_at": 1585904405
    },
    "errors": [
        {
            "code": 1,
            "time": 1585904399
        }
    ],
    "data": [
        {
            "time": 1585904399,
            "temp": 19.4,
            "humidity": 100,
            "pressure": 1032.4,
            "wind": 34.6,
            "windgust": 42.8,
            "winddir": 246,
            "rain": 0.0,
            "israining": 1,
            "solar": 1021,
            "28BC2C799711031E": 18.7,
            "28BC2A799711031E": 14.1,
            "28BC23799711031E": 14.1,
            "28BC26799711031E": 14.1,
            "voltage": 12.4,
            "temp_case": -13.0,
            "hum_case": 100,
            "sddrivespace": 34.5
        }
    ]
}
```
The content is saved minified in a JSON file. Every minute a new entry is added to the data object or, in the event of an error, to the errors object. I am currently trying to use the "JSON Configuration File" example with moderate success. I can already change the information in the header object. But I don't succeed in inserting additional objects into the error or data object.

First I read in the Json file and write the values and arrays into the intended variables within the class. Then I create a new JSON document instance for later storage in the JSON. So far so good.

I'll show you the crucial point where I can't get any further. If I want to add the objects to the array using json.errors, I only succeed in the for loop before calling createNestedArray. After the call, the loop is no longer executed.

```cpp
  for (JsonObject obj : json.errors) {
    Serial.println("Inner Funktion cool before:");
  }

  JsonArray errors = doc.createNestedArray("errors");

  for (JsonObject obj : json.errors) {
    Serial.println("Inner Funktion cool after:");
  }
```

Everything works fine with the information in the header object. Where's my mistake in adding the objects in the error array? Why can't I use json.errors after calling createNestedArray ("errors")? 

### Output of the console:
```txt
The header obj:
3.1.2

The errors array:
1

Inner Funktion cool before:
[... 281 lines cut ...]
    RTC rtc;
    GSM gsm;
    Json json;
    Header header;
    Error error;
    JsonObject testit;
    const char* filename = "current.json";
    const char* archiveFolder = "archive";

    public:
    // Constructor
    SD();
    void begin();
    String read();
    void updateFile();
    void write(char* output);
    void createArchiveFolder();
    void moveToArchive();

    void loadCurrentFile();
    void writeCurrentFile();
};

#endif
```

bblanchon, 2020-04-06:
Hi @pixelfriese,

I didn't read your question completely because it's way too long.

From what I understood, `json.errors` contains a dangling pointer to a destructed `JsonDocument`.
The fact that `createNestedArray()` erase the original array is a pure coincidence.

Solution: don't keep a `JsonArray` without the corresponding `JsonDocument`.
Even better: don't use a `JsonDocument` to store the state of your application.

Best regards,
Benoit

pixelfriese, 2020-04-06:
Hi @bblanchon ,

many thanks for your response. I just tried to initialize the JsonDocument in the upper function updateFile and to pass it as a parameter to loadCurrentFile () and writeCurrentFile (). However, doc is always empty.

I try to use the JsonDocument as shortly as possible in the program code, as described.

With other words, I'm just looking for a way to add more objects to the errors and data array. This would be great if I could use separate setters, e.g. addError() and addData() to extend the json. The idea behind this was to first read the Json, expand it using a setter and then save it back to the SD card. Basically like the configuration file example. The only difference is that an array can be expanded to include additional objects. How is this possible?

### Before:
```json
{
    "header": {
        "mac": "xxxxxxxxxxxxxxxxxxx",
        "version": "1.0.1",
        "generated_at": 1585904405
    },
    "errors": [
        {
            "code": 1,
            "time": 1585904399
        }
    ],
    "data": [
        {
            "time": 1585904399,
            "temp": 19.4,
            "humidity": 100,
            "pressure": 1032.4,
            "wind": 34.6,
            "windgust": 42.8,
            "winddir": 246,
            "rain": 0.0,
            "israining": 1,
            "solar": 1021,
            "28BC2C799711031E": 18.7,
            "28BC2A799711031E": 14.1,
            "28BC23799711031E": 14.1,
            "28BC26799711031E": 14.1,
            "voltage": 12.4,
            "temp_case": -13.0,
            "hum_case": 100,
            "sddrivespace": 34.5
        }
    ]
}
```

### After:

```json
{
    "header": {
        "mac": "xxxxxxxxxxxxxxxxxxx",
        "version": "1.0.1",
        "generated_at": 1585904405
    },
    "errors": [
        {
            "code": 1,
            "time": 1585904399
        },
        {
            "code": 21,
            "time": 1587904399
        }       
    ],
    "data": [
        {
            "time": 1585904399,
            "temp": 19.4,
            "humidity": 100,
            "pressure": 1032.4,
            "wind": 34.6,
            "windgust": 42.8,
            "winddir": 246,
            "rain": 0.0,
            "israining": 1,
            "solar": 1021,
            "28BC2C799711031E": 18.7,
            "28BC2A799711031E": 14.1,
            "28BC23799711031E": 14.1,
            "28BC26799711031E": 14.1,
            "voltage": 12.4,
            "temp_case": -13.0,
            "hum_case": 100,
            "sddrivespace": 34.5
        },
        {
            "time": 1585924399,
            "temp": 19.4,
            "humidity": 100,
            "pressure": 1032.4,
            "wind": 34.6,
            "windgust": 42.8,
            "winddir": 246,
            "rain": 0.0,
            "israining": 1,
            "solar": 1021,
            "28BC2C799711031E": 18.7,
            "28BC2A799711031E": 14.1,
            "28BC23799711031E": 14.1,
            "28BC26799711031E": 14.1,
            "voltage": 12.4,
            "temp_case": -13.0,
            "hum_case": 100,
            "sddrivespace": 34.5
        }     
    ]
}
```

bblanchon, 2020-04-07:
Hi @pixelfriese,

I don't know how to help you, because I don't know what your problem is.
Can you rephrase your question in one sentence only?

Best regards,
Benoit

closed 2020-04-20 by bblanchon