bblanchon/ArduinoJson #1233: Warnings in IAR compiler

richnwhit, 2020-04-06:
I know the IAR Compiler isn't a supported option, but the project is working fine on it and compiles with just two warnings, so I thought it worth reporting these especially as they look sensible.

`template <typename TAdaptedString>
inline bool slotSetKey(VariantSlot* var, TAdaptedString key, MemoryPool* pool,
                       storage_policy::decide_at_runtime) {
  if (key.isStatic()) {
    return slotSetKey(var, key, pool, storage_policy::store_by_address());
  } else {
    return slotSetKey(var, key, pool, storage_policy::store_by_copy());
  }
  return true;       // <-- warning here
}`
Warning : Statement is unreachable

`  void visitPositiveInteger(UInt value) {
    if (value <= 0x7F) {
      writeInteger(uint8_t(value));
    } else if (value <= 0xFF) {
      writeByte(0xCC);
      writeInteger(uint8_t(value));
    } else if (value <= 0xFFFF) {
      writeByte(0xCD);
      writeInteger(uint16_t(value));
    } else if (value <= 0xFFFFFFFF) {     // <-- Warning here
      writeByte(0xCE);
      writeInteger(uint32_t(value));
    }
#if ARDUINOJSON_USE_LONG_LONG
    else {
      writeByte(0xCF);
      writeInteger(uint64_t(value));
    }
#endif
  }
`
Warning: Pointless integer comparison, the result is always true

bblanchon, 2020-04-06:
Hi @richnwhit,

Thank you for reporting these warnings.
IAR compiler **is** a supported option.
Can you give me the exact version/architecture of the compiler so I can try to repro here?

Best regards,
Benoit

richnwhit, 2020-04-06:
Benoit,

Sorry I misspoke - I am using Embedded Workbench, which is indeed supported. But I am using the ARM compiler (version 8.40.1), which you may not be supporting (it isn't free).

The warnings looked pretty sensible and straightforward (I'm surprised that other tools aren't complaining); if you want me to check an unofficial version for you on the IAR compiler, please let me know.

BTW I did successfully implement a memory test, by storing a dummy value and then checking that it is present.

Thanks a lot for your excellent work,

Richard

_____________ Servomex incoming mail advice _____________
Servomex may, as part of its normal activities, monitor, edit or censor the content of any information and software, transmitted through, or stored on, its facilities. Servomex reserves the right to monitor communications, as required, to protect the legitimate interests of Servomex and its business partners.

CORPORATE DOCUMENT

----Servomex Group Limited, UK Technical and Service Centre, Jarvis Brook, Crowborough, East Sussex, TN6 3FB, England. Company Registered in England: No.2170458 VAT No.: GB 522 6077 63---- EXPORT CONTROL INFORMATION: Resale, onward shipment or diversion of Servomex products to embargoed destinations including but not limited to Iran, Syria, Sudan and North Korea is strictly prohibited. This email has been scanned for all viruses by MessageLabs and contains information from Servomex which may be privileged or confidential. The information is intended to be for the use of the individual(s) or entity named above. If you are not the intended recipient be aware that any disclosure, copying, distribution or use of the contents of this information is prohibited. If you have received this electronic message in error, please notify us immediately. Servomex may, as part of its normal activities, monitor, edit or censor the content of any information and software, transmitted through, or stored on, its facilities.

bblanchon, 2020-04-06:
@richnwhit, can you check out the branch "issue1233" to verify that the warnings are gone?

richnwhit, 2020-04-07:
Benoit,

Would it be possible for you to create the single-header version for me? I can try to build the library from source in ARM compiler, but that's not what I was doing, and I may get different issues.

Thanks,

Richard

bblanchon, 2020-04-07:
[ArduinoJson-v6.15.0-6-g40d1cfe7.zip](https://github.com/bblanchon/ArduinoJson/files/4442857/ArduinoJson-v6.15.0-6-g40d1cfe7.zip)

richnwhit, 2020-04-07:
Benoît,

That's fixed it, thank you!

Richard

closed 2020-04-07 by bblanchon

bblanchon, 2020-04-08:
Shipped in [ArduinoJson 6.15.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.1)