bblanchon/ArduinoJson #1234: deserializeJson results in an empty Json document

mavenik, 2020-04-06:
I am using `deserializeJson` to parse a json file from SPIFFS on an ESP32 board. I have tried using the `file` stream as illustrated in [JsonConfigFile example](https://arduinojson.org/v6/example/config/) except for the fact that I am using SPIFFS instead of SD as in the example. Every time I deserialize the file, it returns an empty JSON with size 0. I then tried to read contents of the file using `Stream::readBytes` method and did an `ets_printf` to check if the file was actually being read or not. The file contains valid JSON (I tested the code with invalid JSON and it fails with a `DeserializationError` in that case). `ets_printf` prints the exact JSON that's on the file. But my document object gives size 0.

Target Platform: ESP32 on DOIT DEVKITV1 board
Compiler model and version: Can't be sure of this, but it shows xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 in my compiler logs.

MVCE:
````C
#include <ArduinoJson.h>
#include <SPIFFS.h>
const char spacesFilePath[] = "/spaces.json";
DynamicJsonDocument loadSpaces()
{
  DynamicJsonDocument spacesJsonDoc(2048);

  if(SPIFFS.exists(spacesFilePath))
  {
    File file = SPIFFS.open(spacesFilePath, "r");
    if(!file)
    {
      Serial.println("Could not open spaces file to load current status");
      } // if
      else
      {
        // I resorted to reading the file after deserializeJson did not work with file
        char fileBuffer[file.size()+1];
        file.readBytes(fileBuffer, file.size());
        fileBuffer[file.size()]='\0';
        ets_printf("File read:\n%s\n", fileBuffer);
        DeserializationError error = deserializeJson(spacesJsonDoc, fileBuffer, file.size()+1);
        if(error)
        {
          Serial.println("Failed to read json data from spaces file");
          }
              JsonObject spaces = spacesJsonDoc.to<JsonObject>();
    size_t spacesLength = measureJson(spacesJsonDoc);
    char * jsonString = (char*)malloc(spacesLength*sizeof(char));
    serializeJson(spacesJsonDoc, jsonString, spacesLength+1);
    ets_printf("Space JSON Document length is %d\n", spacesJsonDoc.size());
    ets_printf("Spaces JSON Object length is %d\n", spaces.size());
    ets_printf("Spaces JSON String is %s\n", jsonString);
        } // else (!file)

            file.close();
    }

    return spacesJsonDoc;
  }

void setup() {
  Serial.begin(115200);
  SPIFFS.begin();
  delay(5000);
  Serial.println("Starting...");
  // put your setup code here, to run once:
  DynamicJsonDocument spacesJsonDoc = loadSpaces();
  JsonObject spaces = spacesJsonDoc.to<JsonObject>();
    size_t spacesLength = measureJson(spacesJsonDoc);
    char * jsonString = (char*)malloc(spacesLength*sizeof(char));
    serializeJson(spacesJsonDoc, jsonString, spacesLength+1);
    ets_printf("Space JSON Document length is %d\n", spacesJsonDoc.size());
    ets_printf("Spaces JSON Object length is %d\n", spaces.size());
    ets_printf("Spaces JSON String is %s\n", jsonString);
  
}

void loop() {
  // put your main code here, to run repeatedly:
}
````

Outputs:
````JSON
Starting...
File read:
{
  "0e1f3c95-75b9-877b-49b9-17cf4d9ddabc": {
    "name": "Bedroom",
    "icon": "home"
  }
}

Space JSON Document length is 0
Spaces JSON Object length is 0
Spaces JSON String is {}
Space JSON Document length is 0
Spaces JSON Object length is 0
Spaces JSON String is {}
````

In another context, `deserializeJson` works well for string data. So I guess I am missing something obvious that I am unable to figure for the last few days. It would greatly help if someone could point it out to me.

bblanchon, 2020-04-06:
Hi @mavenik,

[JsonDocument::to<T>()](https://arduinojson.org/v6/api/jsondocument/to/)  clears the `JsonDocument`.
You must use [JsonDocument::as<T>()](https://arduinojson.org/v6/api/jsondocument/as/) instead.

Best Regards,
Benoit

mavenik, 2020-04-07:
Thanks @bblanchon! This worked. I'll close this issue.

closed 2020-04-07 by mavenik

bblanchon, 2020-04-07:
You're welcome @mavenik!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: