bblanchon/ArduinoJson #1235: ArduinoJson with ESPAsyncWebServer

lbussy, 2020-04-10:
Hello again,

I noted that you contributed some to [AsyncJson.h](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncJson.h).  It's apparent that your work has marched on and maybe the references/capabilities in ESPAsyncWebServer have not.  The header referenced seems to be an attempt as a bridge between the two projects, however, the examples seem to provide an inconsistent level of success.

My question, to the writer of ArduinoJson, is what might be the best approach to receive data and process it as JSON, and to send JSON data?  Neither of the examples in the AsyncJson.h header work.

I do realize that you are not responsible for that project, but I thought you might have some insight that's specific to your project.

ewaldc, 2020-04-11:
I have extensively tested ESPAsyncWebServer with both AJ5 & 6, but despite recoding several parts of ESPAsyncWebServer, I was not able to make it work in a stable fashion (unexplained crashes, running out of memory, SW watchdog resets).  It also consumes significantly more memory.  I had to make a few modifications to AsyncJsonResponse, AsyncWebServerResponse etc. to make it work but no issues using it with AJ.

In the end, the ESP8266 only has 2 threads to work with, and I found that the regular webserver does a very fine job with up to 10 web server sessions, including web sockets.  So I wrote a WebSockets and Server Sent Events (SSE's) example and contributed those to the Arduino ESP library via this [PR](https://github.com/esp8266/Arduino/pull/7012/files).

Happy to post my code and changes, all is working but it does not survive a 24 hour torture test...
I would have contributed it all back to the ESPAsyncWebServer library but because it never ran in a stable fashion, it made no sense...

bblanchon, 2020-04-11:
Hi Lee,

Let's make things clear: I didn't contribute to `AsyncJson.h`.

I never used ESPAsyncWebServer, and as you can see, I don't use it in any of the examples in arduinojson.org, nor in the book.
It doesn't mean that this library is bad; it's just that I didn't try it.

Now, supposing that the documentation on the README page is up-to-date, I think I'd use `handleBody()` like that:

```c++
void handleBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
  DynamicJsonDocument doc(4096);

  deserializeJson(doc, data, len);
  // handle the request here...

  // prepare response
  doc.clear();
  doc["result"] = "success";

  // serialize response
  String response;
  response.reserve(4096);
  serializeJson(doc, response);

  request->send(200, "application/json", response);  
}
```

You could reduce memory usage by scoping the `DynamicJsonDocument`s properly.

Best regards,
Benoit

lbussy, 2020-04-11:
Thanks, @ewaldc ... yes it's not perfect but as far as I know, it's all we've got.  My specific use case has the controller handling a few gets from SPIFFS as well as interrupts at the same time, and Async helped a bit.  I am also _trying_ to standardize my app framework across some esp8266 as well as esp32 projects where that's a larger advantage.

The developer for Async is rather notoriously out of pocket on those libraries which a lot of people use.  I might be late to the party on this idea, but I've started forking the libs I depend upon once I get everything running.  To that end, I'd LOVE to see any improvements you've made.  I checked your repos and did not see it.  If you want to point me to them or drop them on [my fork](https://github.com/lbussy/ESPAsyncWebServer), I'd owe you a beer if you ever get to Kansas City.

@bblanchon, thank you.  Your name was used in that header - maybe you helped the guy who contributed that indirectly?  I know my last three or four questions have nothing to do with you maintaining your library - I really appreciate your assistance.

ewaldc, 2020-04-11:
I posted my application using ESPAsyncWebServer [here](https://github.com/ewaldc/OpenHAB-ESP/blob/master/OpenHab_ESP_Async.7z).  The same project using regular web server is [here](https://github.com/ewaldc/OpenHAB-ESP).  You will find AsynArduinoJson.h with some modified sections e.g. ChunkedPrint and class AsyncJsonResponse.
Ewald

lbussy, 2020-04-11:
Duh, didn't think to actually search your apps. :)  Thank you sir.

closed 2020-04-20 by bblanchon