bblanchon/ArduinoJson #1236: Soft WDT Reset

Razzo1987, 2020-04-12:
I'm using ArduinoJson 6.15.1 on NodeMCU 1.0

I do a webservice API call and then parse the response.
The `bodyLine` have 3191 charaters: the data of 20 users

I would like to print on serial all the response as JsonPretty ( but with serializeJson I have the same problem): after half ot fhe parsed text I recieve the error "Soft WDT Reset "

CODE:
```
          WiFiClientSecure client;
          ....
      // Allocate the JSON document
      // Use arduinojson.org/v6/assistant to compute the capacity.
      const size_t capacity = JSON_ARRAY_SIZE(20) + JSON_OBJECT_SIZE(1) + 22*JSON_OBJECT_SIZE(3) + 20*JSON_OBJECT_SIZE(6) + 2850;
      DynamicJsonDocument doc(capacity);
          ...
          String bodyLine = client.readStringUntil('\n');
          ....
          // Deserialize the JSON document
          DeserializationError err = deserializeJson(doc, bodyLine);
          //ReadLoggingStream loggingStream(bodyLine, Serial);
          //DeserializationError err = deserializeJson(doc, loggingStream);
    
          // Test if parsing succeeds.
          if (err) {
            Serial.print(F("deserializeJson() failed: "));
            Serial.println(err.c_str());
            return;
          }

          // Print the body response
          serializeJsonPretty(doc, Serial);
```

ERROR:
```
Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffb70 end: 3fffffc0 offset: 01b0
3ffffd20:  00000000 00000065 3ffefb7c 40208c35  
3ffffd30:  3ffeedf8 3fff76e6 3ffffecc 40206c7b  
3ffffd40:  3ffe86ab 3fff8388 3ffffecc 402098f1  
```

ewaldc, 2020-04-13:
Hard to say for sure where the problem is without [exception decoding](https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html).
My guess would be the _readStringUntil()_ call as it is known to be a common cause for Soft WDT resets as it's a blocking call.  
Try to put a yield() statement before of after it and add a timeout to make sure it's not blocking.
```
yield();
client.setTimeout(2000);
input = client.readStringUntil('\n');
```
Ewald

Razzo1987, 2020-04-13:
Thanks @ewaldc but I didn't solve :(
I have uploaded the full code here: https://github.com/JoomlaLABS/esp8266-joomla/blob/master/users/users-count/users-count.ino

And this is my stack:
```
Decoding stack results
0x40208c9d: uart_write(uart_t*, char const*, size_t) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/uart.cpp line 509
0x40208c9d: uart_write(uart_t*, char const*, size_t) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/uart.cpp line 509
0x40206c9c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/HardwareSerial.h line 165
0x402098de: ArduinoJson6151_0000010::Writer ::write(unsigned char const*, unsigned int) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/Writers/PrintWriter.hpp line 22
0x40201cfc: ArduinoJson6151_0000010::TextFormatter   >::writeRaw(char const*) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp line 131
0x40201cfc: ArduinoJson6151_0000010::TextFormatter   >::writeRaw(char const*) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp line 131
0x40201d2c: ArduinoJson6151_0000010::PrettyJsonSerializer   >::indent() at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp line 63
0x40202390: ArduinoJson6151_0000010::PrettyJsonSerializer   >::visitObject(ArduinoJson6151_0000010::CollectionData const&) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp line 62
0x402022e9: ArduinoJson6151_0000010::VariantData::accept    > >(ArduinoJson6151_0000010::PrettyJsonSerializer   >&) const at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp line 46
0x402023a7: ArduinoJson6151_0000010::PrettyJsonSerializer   >::visitObject(ArduinoJson6151_0000010::CollectionData const&) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp line 53
0x402022e9: ArduinoJson6151_0000010::VariantData::accept    > >(ArduinoJson6151_0000010::PrettyJsonSerializer   >&) const at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp line 46
0x402022b3: ArduinoJson6151_0000010::VariantData::accept    > >(ArduinoJson6151_0000010::PrettyJsonSerializer   >&) const at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp line 32
0x402023a7: ArduinoJson6151_0000010::PrettyJsonSerializer   >::visitObject(ArduinoJson6151_0000010::CollectionData const&) at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp line 53
0x402022e9: ArduinoJson6151_0000010::VariantData::accept    > >(ArduinoJson6151_0000010::PrettyJsonSerializer   >&) const at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp line 46
0x40202e59: loop() at /home/razzo/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp line 221
0x40206c00: fs::FS::open(char const*, char const*) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/FS.cpp line 355
0x40206c9c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/HardwareSerial.h line 165
0x40206c00: fs::FS::open(char const*, char const*) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/FS.cpp line 355
0x40206f00: IPAddress::fromString(char const*) at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/IPAddress.cpp line 68
0x4020131e: setup() at /home/razzo/Arduino/Joomla/J4-rest-call-esp8266/J4-rest-call-esp8266.ino line 57
0x40207f4c: loop_wrapper() at /home/razzo/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 180
```

If I comment line 172 it works without problem

bblanchon, 2020-04-14:
Hi @Razzo1987,

Thank you very much for sharing this.

I really don't understand what is going on.
According to the call stack, the crash occurs when ArduinoJson prints the indentation spaces.
From my experience, these call stacks are often misleading, so we mustn't put to much credit on that.

Unfortunately, I cannot run your source code here.
Please write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) in the following form:

```c++
String bodyLine = "INSERT CONTENT HERE";
deserializeJson(doc, bodyLine);
serializeJson(doc, bodyLine);
```

From what I understand, it should be enough to trigger the crash.

BTW, I see that you're jumping through hoops to get around Chunk Transfer Encoding.
You could greatly simplify your code by using HTTP 1.0.
Also, you could reduce the memory usage by letting ArduinoJson pull the body from the `WifiClientSecure`.
See [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) for an example.

Best regards,
Benoit

Razzo1987, 2020-04-14:
Hi @bblanchon I create a MCVE:
```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  delay(5000);
}

void loop() {

  const size_t capacity = JSON_ARRAY_SIZE(20) + JSON_OBJECT_SIZE(1) + 22*JSON_OBJECT_SIZE(3) + 20*JSON_OBJECT_SIZE(6) + 2850;
  DynamicJsonDocument doc(capacity);
  
  String bodyLine = "{\"links\":{\"self\":\"https:\/\/users.jugmi.it\/api\/index.php\/v1\/users\",\"next\":\"https:\/\/users.jugmi.it\/api\/index.php\/v1\/users?page%5Boffset%5D=20&page%5Blimit%5D=20\",\"last\":\"https:\/\/users.jugmi.it\/api\/index.php\/v1\/users?page%5Boffset%5D=20&page%5Blimit%5D=20\"},\"data\":[{\"type\":\"users\",\"id\":\"56\",\"attributes\":{\"id\":56,\"name\":\"Luca Racchetti\",\"username\":\"Razzo\",\"email\":\"razzo@jugmi.it\",\"group_count\":1,\"group_names\":\"Super Users\"}},{\"type\":\"users\",\"id\":\"57\",\"attributes\":{\"id\":57,\"name\":\"user1\",\"username\":\"user1\",\"email\":\"user1@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"66\",\"attributes\":{\"id\":66,\"name\":\"user10\",\"username\":\"user10\",\"email\":\"user10@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"67\",\"attributes\":{\"id\":67,\"name\":\"user11\",\"username\":\"user11\",\"email\":\"user11@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"68\",\"attributes\":{\"id\":68,\"name\":\"user12\",\"username\":\"user12\",\"email\":\"user12@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"69\",\"attributes\":{\"id\":69,\"name\":\"user13\",\"username\":\"user13\",\"email\":\"user13@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"70\",\"attributes\":{\"id\":70,\"name\":\"user14\",\"username\":\"user14\",\"email\":\"user14@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"71\",\"attributes\":{\"id\":71,\"name\":\"user15\",\"username\":\"user15\",\"email\":\"user15@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"72\",\"attributes\":{\"id\":72,\"name\":\"user16\",\"username\":\"user16\",\"email\":\"user16@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"73\",\"attributes\":{\"id\":73,\"name\":\"user17\",\"username\":\"user17\",\"email\":\"user17@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"74\",\"attributes\":{\"id\":74,\"name\":\"user18\",\"username\":\"user18\",\"email\":\"user18@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"75\",\"attributes\":{\"id\":75,\"name\":\"user19\",\"username\":\"user19\",\"email\":\"user19@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"58\",\"attributes\":{\"id\":58,\"name\":\"user2\",\"username\":\"user2\",\"email\":\"user2@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"76\",\"attributes\":{\"id\":76,\"name\":\"user20\",\"username\":\"user20\",\"email\":\"user20@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"59\",\"attributes\":{\"id\":59,\"name\":\"user3\",\"username\":\"user3\",\"email\":\"user3@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"60\",\"attributes\":{\"id\":60,\"name\":\"user4\",\"username\":\"user4\",\"email\":\"user4@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"61\",\"attributes\":{\"id\":61,\"name\":\"user5\",\"username\":\"user5\",\"email\":\"user5@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"62\",\"attributes\":{\"id\":62,\"name\":\"user6\",\"username\":\"user6\",\"email\":\"user6@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"63\",\"attributes\":{\"id\":63,\"name\":\"user7\",\"username\":\"user7\",\"email\":\"user7@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}},{\"type\":\"users\",\"id\":\"64\",\"attributes\":{\"id\":64,\"name\":\"user8\",\"username\":\"user8\",\"email\":\"user8@jugmi.it\",\"group_count\":1,\"group_names\":\"Registered\"}}],\"meta\":{\"total-pages\":2}}";
  
  deserializeJson(doc, bodyLine);
  serializeJson(doc, bodyLine);
  Serial.print(bodyLine);
  
}
```

PS = Tanks for your example, I'm reading the http-client example :+1:

ewaldc, 2020-04-15:
Hmm, on a NodeMCU ESP8266 it's running fine for one hour now.   Can you check your power supply and/or USB cable?  That could explain why it crashes in an unlikely location...

bblanchon, 2020-04-15:
Hi @Razzo1987,

I managed to reproduce the crash on my D1 mini.
I fixed it by changing the baud rate of the serial port from 9600 to 115200.
I believe it's a bug in the ESP8266 core that occurs when the program saturates the serial buffer.

Best regards,
Benoit

bblanchon, 2020-04-15:
I can reproduce the bug without ArduinoJson:

```c++
while(true) {
  Serial.write('X');
}
```

ewaldc, 2020-04-15:
Ah, changing the baud rate to 115200 is also the only change I made.  
This works for me @9600 baud:
```
  deserializeJson(doc, bodyLine);
  serializeJson(doc, bodyLine);
  ESP.wdtDisable();
  Serial.print(bodyLine);
  ESP.wdtEnable(1);
  yield();
```
In general though it is not recommended to do blocking calls (Serial, TCP/IP etc.) in the main _loop()_ or in callbacks and use the Ticker() library (TickerScheduler) instead (see [here](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html)).  That is what I do in my code as as well.

9600, 2020-04-15:
You called?

ewaldc, 2020-04-15:
Oops, did not realize there was a user called **9600**.  Side effect of referring to 9600 baud and typing @ 9600 in (one word) I guess.

Razzo1987, 2020-04-16:
Thanks to @bblanchon 
Setting the baud rate of the serial port from 9600 to 115200 solved for me!

The solution of @ewaldc truncates the output:

```
...
     "type": "users",
      "id": "61",
      "attributes": {
        "id": 61,
        >B⸮b⸮P8⸮⸮S⸮⸮
```

ewaldc, 2020-04-17:
Interesting, on my NodeMCU 1.0 I see no truncation, nor strange characters.  Maybe the difference is with the board serial implementation, cable, or settings (stop bits, hardware line control etc.).
Anyhow, increasing baud rate is the easiest solution, by default I always use 115200 or higher (e.g. I upload always at 921600 baud), which is why I wasn't able to reproduce the issue initially.

closed 2020-04-20 by bblanchon