bblanchon/ArduinoJson #1237: Json Beginner -> IncompleteInput/NoMemory on large JSON

MikeAUT, 2020-04-13:
Hi all,
I'm pretty new on ArduinoJson and Json itself and string handling with arduino. I try to write a code on the NodeMCU (ESP8266) with the Arduino IDE to fetch weather data from Openweathermap.

I want to use the oneCall API because there is everything in it I want to use, even the hourly forecast for which you have to pay for when using it's own API. I've tried two other API's (current weather and forecast). Parsing this API's work. But when I try to parse the oneCall API it results in IncompleteInput or NoMemory depending on the code i use to read the incoming string. According to arduinojson the oneCall api has a lenght of approximately 15000 Bytes. 

Here is the code i use for parsing:
`Serial.println("\nStarting connection to server...");
  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected to server");
    client.println("GET " + oneCall);
    client.println("Host: api.openweathermap.org");
    client.println("Connection: close");
    client.println();
  } else {
    Serial.println("unable to connect");
  }
  delay(1000);
  String input = "";
  while (client.available()) {
    input = client.readStringUntil('\n');
    Serial.println(input);
    Serial.println("parsingValues");
    DynamicJsonDocument doc(25000); <- also tried 30000, which should be enough

    //DeserializationError err = deserializeJson(doc, client.readStringUntil('\n')); <- results in IncompleteInput
    DeserializationError err = deserializeJson(doc, input); <- results in NoMemory
    if (err) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(err.c_str());
    }`

I hope you can help ;)

ewaldc, 2020-04-13:
Working with 25K strings is always a bit of a challenge.
First problem is the String class which often causes duplication e.g. when passed as an argument. Secondly, AJ will make another copy (see [here](https://arduinojson.org/v6/api/json/deserializejson/)) if not passed a _char *_ (zero copy mode).  
You might try to pass input.c_str() to deserializeJson, but in that case you need to keep _input_ in memory.  

A better solution is to let AJ deserialize directly from the HTTPClient stream (see [here](https://arduinojson.org/v6/example/http-client/)) after having skipped the headers. 

Another possibility is to read the content from the web server into SPIFFS first  (httpClient.writeToStream(&file)) and deserialize from there (take a look [here](https://github.com/ewaldc/OpenHAB-ESP/blob/master/src/main.cpp), function _GetSitemap()_), but this approach generally makes more sense if you want to cache reads from the web server (not the case for weather data I guess)

It is very easy to end up with multiple copies of the data in memory (or fragments thereof) e.g. in the network stack buffers, the client.readStringUntil() buffer string, another copy when passing it as an argument, and finally AJ making its own copy.  Add to that the problem memory fragmentation, and it's easy to see why with about 50K free memory you can run into issues.

Ewald

bblanchon, 2020-04-13:
Hi @MikeAUT,

[IncompleteInput](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) usually means that the connection dropped or timed out.

[NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory) means that the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is too small, or that the allocation of the memory pool failed.

In addition to @ewaldc, please see: [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)

Best regards,
Benoit

MikeAUT, 2020-04-13:
Thank you guys, really kind.
I tried the method of deserializing directly from HTTP and skipping headers. And it's working (FOR NOW). Only those three code parts where added respectively changed

 ```
// Send HTTP request
  client.println(F("GET /data/2.5/onecall?lat=48.262035&lon=13.867318&appid=XXXXXXXXXXXXXXXXXXX&mode=json&units=metric&lang=de HTTP/1.0"));
  client.println(F("Host: api.openweathermap.org"));
  client.println(F("Connection: close"));
  if (client.println() == 0) {
    Serial.println(F("Failed to send request"));
    return;
  }

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response: "));
    Serial.println(status);
    return;
  }

  // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    Serial.println(F("Invalid response"));
    return;
  }
```

Can you explain what those parts effectively do?
what exactly does HTTP/1.0 in the GET command do?
What does Check HTTP Status do?
What does skip HTTP exactly do?

ewaldc, 2020-04-13:
@MikeAUT, a HTTP(S) response to a GET command consists of multiple parts (see [here](https://developer.mozilla.org/en-US/docs/Web/HTTP/Messages)).  The "_body_" section is where you (typically) find the JSON payload (the data you want to pass to ArduinoJson), but the headers and status line are not valid JSON, so you want to skip these, except for the status line where you want to make sure code _200_ was returned which means that the GET request completed successfully (e.g. a code of _404_ would mean that the object you requested was not _Not Found_). 
If you turn on developer mode of your browser, you will be able to see the header and status line.

These are the fundamentals of HTTP, just needs a little bit of study work :-).

MikeAUT, 2020-04-13:
Ok, that clears some things up ;) Many thanks ;)

closed 2020-04-14 by MikeAUT