bblanchon/ArduinoJson #1238: Add json object to array and serializeJson json to string have problem crash esp8266

thiminhnhut, 2020-04-13:
* I'm using ArduinoJson 6.15.1 on ESP8266 Wemos D1 Mini.
* I'm want create json object with format: 
```json
{
	"cmd_class": "03",
	"cmd_type": "03",
	"cmd_id": 29173199,
	"ts": 1581302429,
	"client_id": "e3747944",
	"duuid": "3b4d8ccc",
	"did": "84F3EBB9FFD6",
	"err_code": 0,
	"size": 2,
	"filesystem": [{
			"file": "serial.json",
			"size": "20B"
		},
		{
			"file": "bootstrap.json",
			"size": "80B"
		}
	]
}
```
and convert json object to string with statements  `serializeJson(json, command)` but not success.
* Code:
```cpp
String getResListFileSystem(String serial, String duuid, String clientID,
                              unsigned long cmdID, unsigned long timestamp,
                              unsigned char size, StructFS fs[], int error) {
    DynamicJsonDocument response(1024);
    response[CMD_CLASS] = CMD_CLASS_DEV;
    response[CMD_TYPE] = CMD_LIST_FILE_SYSTEM;
    response[CLIENT_ID] = clientID;
    response[DEVICE_UUID] = duuid;
    response[CMD_ID] = cmdID;
    response[INFO_DEVICE_ID] = serial;
    response[ERROR_CODE] = error;
    response[SIZE] = size;
    JsonArray arrayFS = response.createNestedArray(LIST_FILE_SYSTEM);
   // <<<Open PROBLEM
    for (unsigned char i = 0; i < size; i++) {
      JsonObject objectFS = arrayFS.createNestedObject();
      objectFS[FILE_NAME] = fs[i].name;
      objectFS[SIZE] = fs[i].size;
    }
    // Close PROBLEM >>>
    String command;
    serializeJson(response, command);
    return command;
  }
```
* When use part code  bellow:
```cpp
// <<<Open PROBLEM
  for (unsigned char i = 0; i < size; i++) {
    JsonObject objectFS = arrayFS.createNestedObject();
    objectFS[FILE_NAME] = fs[i].name;
    objectFS[SIZE] = fs[i].size;
  }
  // Close PROBLEM >>>
```
and using statements `serializeJson(response, command)` convert json to String, make crash esp8266 with exception:
```
Exception (2):
epc1=0x3fff49c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff49c0 depc=0x00000000

InstructionFetchError: Processor internal physical address or data error during instruction fetch
```
when I don't add part code above. Statements  `serializeJson(response, command)` success and everything works normally, esp8266 not crash.

I hope you can help,
Thanks!

bblanchon, 2020-04-13:
Hi @thiminhnhut,

At first sight, it looks like a dangling pointer.
What is in `fs[i].name`?

Best regards,
Benoit

thiminhnhut, 2020-04-13:
> Hi @thiminhnhut,
> 
> At first sight, it looks like a dangling pointer.
> What is in `fs[i].name`?
> 
> Best regards,
> Benoit

Thank @bblanchon, 
Variable `fs` have type bellow:
```cpp
struct StructFS {
  String name;
  String size;
};
```

bblanchon, 2020-04-13:
Well, this should work...
Can you write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?
I have a D1 mini to test with.

bblanchon, 2020-04-20:
@thiminhnhut, did you make any progress?

closed 2020-05-04 by bblanchon