bblanchon/ArduinoJson #1240: [Help] How to check if serializeJson has correctly sent json over WiFi

GTP95, 2020-04-16:
Hello everyone.
I'm using your library in a simple sketch that reads data from a DHT22 sensor and sends them over WiFi as a json containing 3 fileds: a string identifying the device, a float number for humidity and another float for temperature. For some reason I wasn't able to determine yet, when I add the code to read values from the sensor (without modifying the code that connects to my server and sends the json!) I have to call serializeJson(doc, client) multiple times before the json gets sent correctly (I'm checking using netcat on the receiving side). My workaround was to use a do-while loop to retry until the data gets sent, but here's my problem: to which value shoud I compare serializeJson()'s return value? I've already tryied to compare it to JSON_OBJECT_SIZE(3), to some values obtained from the assistant (https://arduinojson.org/v6/assistant/) and other values determined in other ways but I wans't able to to find the right value. According to the documentation serializeJson() should return the number of bytes written, but I didn't find a way to obtain the json's size in order to compare the bytes written to the bytes that should have been written. Could somebody please tell me how to check if serializeJson() has correctly sent the json?
Thanks for your help

bblanchon, 2020-04-17:
Hi @GTP95,

As explained in the documentation of [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/), you can use the `WriteLoggingStream` decorator from the [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) library:

```c++
WriteLoggingStream wifiClientWithLog(wifiClient, Serial);
serializeJson(doc, wifiClientWithLog);
```

This will print exactly what ArduinoJson sends over the `WiFiClient`.

It's very likely that the problem is in the WiFi connection and not with the JSON serializer, so I recommend that you check what happens when you send a hard-coded string. I bet that the same problem occurs.

Best regards,
Benoit

ewaldc, 2020-04-17:
For larger strings, definitely writing to WriteLoggingStream is the way to go as suggested by @bblanchon.
For smaller strings, you may get away with [size_t measureJson(const JsonDocument& doc);](https://arduinojson.org/v6/api/json/measurejson/), it provides the length excluding the '0' termination byte, which you may have to sent as well.

GTP95, 2020-04-17:
> Hi @GTP95,
>
> As explained in the documentation of [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/), you can use the `WriteLoggingStream` decorator from the [...]

Yes, I too think this isn't a problem with the serializer, because when I was testing the library sending a json with hardcoded values worked, while now it fails even sending a json with hardcoded values.

> For larger strings, definitely writing to WriteLoggingStream is the way to go as suggested by @bblanchon.
> For smaller strings, you may get away with [size_t measureJson(const JsonDocument& [...]

This doesn't seem to work either: if I keep calling serializeJson while the sent bytes count is strictly less than the bytes computed by measureJson sometimes it correctly sends the json and then exits the loop, but some others on thje receiving end I just see the opening '{'. If I instead check if the value reported by serializeJson is less or equal the value computed by measureJson the json gets sent multiple times.
I will try with WriteLoggingStream and report back any interesting findings

bblanchon, 2020-04-18:
Hi @GTP95,

It reminds me that some implementations of `WiFiClient` don't implement [Nagle's algorithm](https://en.wikipedia.org/wiki/Nagle%27s_algorithm), which can lead to terrible performance with ArduinoJson (and may also explain your problem).

As a simple workaround, you can insert a buffer between `serializeJson()` and `WiFiClient` using the `WriteBufferingStream` of the [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) library;

```c++
WriteBufferingStream bufferedWifiClient{wifiClient, 64};
serializeJson(doc, bufferedWifiClient);
bufferedWifiClient.flush();
```

Note that  `WriteBufferingStream`'s destructor calls `flush()`, so depending on the way you write your program, you may drop the last line.

Best regards,
Benoit

GTP95, 2020-04-18:
This worked, thanks! It's still not 100% reliable but failed only once in multiple tests, so this is enough for my pruposes. Thanks again and have a nice weekend!

closed 2020-04-18 by GTP95