bblanchon/ArduinoJson #1241: two way communication of nrf24l01 model using ArduinoJson

soumiksinha19, 2020-04-18:
I want to transmit and receive a json file using arduinojson in nrf24l01
but i am not getting the output..
if anyone can help me giving a very simple code (i.e generator and parsar) will be very thnkful..

bblanchon, 2020-04-18:
Hi @soumiksinha19,

Since `RF24` doesn't implement the `Stream` interface, you must use [custom reader and writer classes](https://arduinojson.org/news/2019/11/01/version-6-13-0/).
In other words, you must create one or two "adapter" classes to adapt the interface of `RF24` to the expectations of [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/) and [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/).

In practice, we can bundle the reader class and the writer class together in a single adapter, which could look like this:

```c++
class RF24Adapter {
 public:
  RF24Adapter(RF24 &radio) : _radio(&radio) {}

  int read() {
    char buf[1];
    _radio->read(buf, 1);
    return buf[0];
  }

  size_t readBytes(char *buffer, size_t length) {
    _radio->read(buffer, static_cast<uint8_t>(length));
    return length;
  }

  size_t write(uint8_t c) {
    return _radio->write(&c, 1) ? 1 : 0;
  }

  size_t write(const uint8_t *buffer, size_t length) {
    return _radio->write(buffer, static_cast<uint8_t>(length)) ? length : 0;
  }

 private:
  RF24 *_radio;
};
```

Now, you can send JSON over RF24 like this:

```c++
RF24Adapter radioForArduinoJson(radio);
serializeJson(doc, radioForArduinoJson);
```

Similarly, you can read JSON from RF24 like this:

```c++
RF24Adapter radioForArduinoJson(radio);
deserializeJson(doc, radioForArduinoJson);
```

See an online proof-of-concept here:
https://wandbox.org/permlink/w1Vz4B0rSNDYcbEl

You can also use `RF24Adapter` with [`serializeMsgPack()`](https://arduinojson.org/v6/api/msgpack/serializemsgpack/) and [`deserializeMsgPack()`](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/).

Best regards,
Benoit

soumiksinha19, 2020-04-18:
thanks a lot

soumiksinha19, 2020-04-18:
i'll let you know

soumiksinha19, 2020-04-19:
sorry to say i couldn't find the way of using this..I mean where to paste the above code  and all..if you kindly elaborate it will be helpful..
Thank You in advance

soumiksinha19, 2020-04-19:
i was working with arduinojson v6..is the above code ok with that?

bblanchon, 2020-04-20:
Yes, the code is intended for ArduinoJson 6.

Simply copy and paste the snippet above into your program.
I really don't see how one could elaborate more...

What did you try? What result did you get?

soumiksinha19, 2020-04-20:
Hi,
Actually i used the examples of generators and parser..but using the above (rf24Adapter)  I changed in transmitter section but in the receiver section the change was not there...thats the prblm i'm facing

soumiksinha19, 2020-04-20:
both the codes are given i am using...but after using the RFAdepter the output is same...
[image]
[image]

soumiksinha19, 2020-04-20:
1st one  is for transmitter section

bblanchon, 2020-04-20:
@soumiksinha19, do you manage to transmit plain text?
In other words, does your program work if you remove ArduinoJson?

soumiksinha19, 2020-04-20:
yes but there were //const char binr[] = "helllowo"; // you can customize this text to your wish
//radio.write(&binr, sizeof(binr));portion in transmitter part and /char inp[32] = "";

//radio.read(&inp, sizeof(inp));Serial.println(binr[2]);
Serial.println(binr[3]);
Serial.println(binr[4])
this was in receiver portion

soumiksinha19, 2020-04-20:
[image]
transmitter section

soumiksinha19, 2020-04-20:
[image]
receiver section

bblanchon, 2020-04-25:
I don't understand.
Can you confirm that you manage to send a string and receive it on the other end?

soumiksinha19, 2020-04-25:
In the transmitter section  just what you write it will print on the receiver side...i just wrote "100" in the void loop section...you can write anything instead of that..that will print

bblanchon, 2020-04-25:
Now, please try to add ArduinoJson on the transmitter side.
Do you receive the expected JSON document on the receiver side?

soumiksinha19, 2020-04-25:
No...it is not working...when i am trying to change in the transmitter side
it is not receiving in the  receiver side

bblanchon, 2020-04-25:
Can you copy/paste the few lines you changed?

soumiksinha19, 2020-04-25:
Okkk

soumiksinha19, 2020-04-25:
[image]
json trans

soumiksinha19, 2020-04-25:
[image]
json recv

soumiksinha19, 2020-04-25:
now you pls hlp me using that rf24Adepter..how to use that..i also used that.but perhaps couldn't work properly..

bblanchon, 2020-04-26:
In the code above, ArduinoJson is writing to the serial port, not to the RF24.
Please re-read my first message above; I explained how you could send to RF24.

soumiksinha19, 2020-04-26:
if i don't use serial port then how can i check it's working or not?

bblanchon, 2020-04-27:
If you don't use RF24 then how can you transmit the data?

soumiksinha19, 2020-04-27:
I used...but i have sent you both of the sketches..

bblanchon, 2020-04-27:
In [this screen capture](https://github.com/bblanchon/ArduinoJson/issues/1241#issuecomment-619346122), you don't seem to use RF24.

bblanchon, 2020-05-10:
@soumiksinha19, did you make any progress?

soumiksinha19, 2020-05-11:
Hi
I hv tried to change the way of doing that...i mean instd of using JSON i
am trying to use DMX

bblanchon, 2020-05-11:
I don't think JSON was a good fit anyway.
If I were you, I would simply send raw `struct`s.

Well, good luck with your project, @soumiksinha19.

closed 2020-05-11 by bblanchon