bblanchon/ArduinoJson #1242: cannot send data from arduino to nodemcu

fatahillahardhi, 2020-04-18:
Hi! im trying to sending data from arduino uno r3 to nodemcu using arduinoJSON 6.15.1 but its always failed, the arduino didnt sent the data.

i use this tutorial https://engineeringprojectshub.com/serial-communication-between-nodemcu-and-arduino/ but the tutorial using arduinoJSON 5 and then i implemented to arduinoJSON 6. When i try use arduinoJSON 5 i got failed too, so i try to implemented arduinoJSON 6, but again, im failed too :(

can u help me?

here's my arduino code

`
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial s(5,6);
 
void setup() {
  s.begin(9600);
}
 
void loop() {
   StaticJsonDocument<100> doc;
   JsonObject root = doc.to<JsonObject>();
   root["data1"] = 100;
   root["data2"] = 200;
  
   if(s.available()>0){
     serializeJson(doc,s);
  }
  
}
`

`
#include <SoftwareSerial.h>
SoftwareSerial s(D6,D5);
#include <ArduinoJson.h>
 
void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  s.begin(9600);
  while (!Serial) continue;
 
}
 
void loop() {
  StaticJsonDocument<100> doc;
  deserializeJson(doc, s);
  if (doc == doc.isNull())
    return;
 
  Serial.println("JSON received and parsed");
  serializeJsonPretty(doc,Serial);
  Serial.print("Data 1 ");
  Serial.println("");
  int data1=doc["data1"];
  Serial.print(data1);
  Serial.print("   Data 2 ");
  int data2=doc["data2"];
  Serial.print(data2);
  Serial.println("");
  Serial.println("---------------------xxxxx--------------------");
 
}
`

anyway sorry for my bad english :))

bblanchon, 2020-04-19:
Hi @fatahillahardhi,

The following lines are curious:

```c++
 if(s.available()>0){
     serializeJson(doc,s);
  }
```

Why are you waiting for `available()`?
It seems that you should always call `serializeJson()`.

Best regards,
Benoit

anhoang241998, 2020-04-19:
Hi, @fatahillahardhi,
I think your question the same like my question 27 days ago, I found the [tutorial]( https://engineeringprojectshub.com/serial-communication-between-nodemcu-and-arduino/) on that link. And tried to pass a json from the **Arduino Mega** to **NodeMCU** and failed. But by @bblanchon suggestion, I thought it can't send because the **baud rate**  between two boards. The receiver **NodeMCU** runs slower than the **Mega**.
So, if I can change your code in ArduinoJson, It will be like this:
Your ***Arduino Uno R3*** code should be:
```C++
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial s(5,6);
 
void setup() {
  s.begin(4800);
}
 
void loop() {
   StaticJsonDocument<100> doc;
   JsonObject root = doc.to<JsonObject>();
   root["data1"] = 100;
   root["data2"] = 200;
   serializeJson(doc,s);
}
```
And your code in ***NodeMCU*** code should be:
```C++
#include <SoftwareSerial.h>
SoftwareSerial s(D6,D5);
#include <ArduinoJson.h>
 
void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  s.begin(4800);
  while (!Serial) continue;
 }
 
void loop() {
  const size_t capacity = JSON_OBJECT_SIZE(2) + 20; 
  DynamicJsonDocument doc(capacity);
  DeserializationError error = deserializeJson(doc, mySerial);
  if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.c_str());
  return;
  }
  Serial.println("JSON received and parsed");
  serializeJsonPretty(doc,Serial);
  Serial.print("Data 1 ");
  Serial.println("");
  int data1=doc["data1"];
  Serial.print(data1);
  Serial.print("   Data 2 ");
  int data2=doc["data2"];
  Serial.print(data2);
  Serial.println("");
  Serial.println("---------------------xxxxx--------------------");
 
}
```
*Note:* My question is #1219 
Best regards,
*Ân Nguyễn Hoàng*

fatahillahardhi, 2020-04-19:
Thank you for the answer

> 
> 
> Hi @fatahillahardhi,
> 
> The following lines are curious:
> 
> ```c++
>  if(s.available()>0){
>      serializeJson(doc,s);
>   }
> ```
> 
> Why are you waiting for `available()`?
> It seems that you should always call `serializeJson()`.
> 
> Best regards,
> Benoit

> 
> 
> Hi @fatahillahardhi,
> 
> The following lines are curious:
> 
> ```c++
>  if(s.available()>0){
>      serializeJson(doc,s);
>   }
> ```
> 
> Why are you waiting for `available()`?
> It seems that you should always call `serializeJson()`.
> 
> Best regards,
> Benoit

Hi, @bblanchon  thank you for ur answer, i try to remove s.available but the result in the nodemcu sometimes not same with data that i send or the data is not receive in nodemcu, here's my serial monitor

```
---------------------xxxxx--------------------
JSON received and parsed
2000Data 1 
0   Data 2 0
---------------------xxxxx--------------------
JSON received and parsed
{
  "data1": 100,
  "data2": 2000
}Data 1 
100   Data 2 2000
---------------------xxxxx--------------------
JSON received and parsed
nullData 1 
0   Data 2 0
---------------------xxxxx--------------------
JSON received and parsed
nullData 1 
0   Data 2 0
---------------------xxxxx--------------------
JSON received and parsed
"data1"Data 1 
0   Data 2 0
```

> Hi, @fatahillahardhi,
> I think your question the same like my question 27 days ago, I found the [tutorial](https://engineeringprojectshub.com/serial-communication-between-nodemcu-and-arduino/) on [...]

Hi, @anhoang241998  thank you for your help, i tried to implement this code, but the result in serial monitor is still wrong, sometimes the data is received but sometimes not too, here's the result

```
JSON received and parsed
{
  "data1": 100,
  "data2": 200
}Data 1 
100   Data 2 200
---------------------xxxxx--------------------
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
JSON received and parsed
{
  "data1": 100,
  "data2": 200
}Data 1 
100   Data 2 200
---------------------xxxxx--------------------
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
deserializeJson() failed: InvalidInput
JSON received and parsed
"data2"Data 1 
0   Data 2 0
---------------------xxxxx--------------------
```

anhoang241998, 2020-04-19:
Hi, @fatahillahardhi,
I think it cause because that the pin of TX, RX you put in the **NodeMCU** is not correct. Tried to change it `D2= RX, D1= TX`.
And again, you can tried my **Mega** code:
```C++
#include <ArduinoJson.h>

void setup() {
Serial2.begin(4800);
}
 
void loop() {
  
const size_t capacity = JSON_OBJECT_SIZE(2);
DynamicJsonDocument doc(capacity);

doc["data1"] = 100;
doc["data2"] = 200;

serializeJson(doc, Serial2);
delay(1000);
}
```
My **NodeMCU** code:
```C++
#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(4,5);  //D2, D1 = SRX, STX

void setup() {
// Initialize Serial port
Serial.begin(9600);
mySerial.begin(4800);
while (!Serial) continue;
}
 
void loop() {
const size_t capacity = JSON_OBJECT_SIZE(2) + 20;
DynamicJsonDocument doc(capacity);

DeserializationError error = deserializeJson(doc, mySerial);
if (error) {
Serial.print(F("deserializeJson() failed: "));
Serial.println(error.c_str());
return;
}
int data1 = doc["data1"]; // 100
int data2 = doc["data2"]; // 200
serializeJson(doc, Serial);
delay(1000);
Serial.println();
}
```
 ***Note***: 
If it not wored, please try:
- Replace deserializeJson() with a simple Serial.readString(). Do you still see the issue?
- Replace serializeJson() with a simple Serial.println(). Do you still see the issue?
- Change the bauds rate more lower? Does it change anything?
Best regards,
*Ân Nguyễn Hoàng*

fatahillahardhi, 2020-04-19:
Hi, @anhoang241998 i try to change softwareserial, but the result still same, i try change baudsrate too but the result still same.

but i dont understand for replace deserializeJson() with a simple Serial.readString() and serializeJson() with a simple Serial.println(), is it replace only in nodemcu? or both? 

for replacing deserializeJson()  whats the variable inside Serial.readString() ? because if the variable same with deserializeJson() i got an error like this 
```no matching function for call to 'HardwareSerial::readString(ArduinoJson6151_0000010::StaticJsonDocument<100u>&, SoftwareSerial&)'```

im new in arduino and i dont understand too much about it, i still learning it :)

anhoang241998, 2020-04-20:
Hi @fatahillahardhi ,
I just tried these programs with my Arduino **MEGA** and my **NodeMCU**. They worked fine.

The speed issue explains the InvalidInput error. The problem I get in my  #1219 is fixed the same way by lowering the baund rate. You can try another baund rate to find which is suit for your component. 

And @bblanchon, can you check this issues for him, I didn't really know why his code don't work anymore? 

Best regards,
Ân Nguyễn Hoàng

fatahillahardhi, 2020-04-21:
i change baud rate to 1200 and i put the delay for 3sec, the result is much better from before, but i got garbage result

this code will be use on my project, i need real-time data, so i must reduce the delay
this the result of Serial Monitor on my **NodeMCU**
```
null
{"data1":100,"data2":2000}
{"data1":100,"data2":2000}
{"data1":100,"data2":2000}
{}
null
100
"data2"
null
2000
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100}
null
{"data1":100,"data2":200}
{"data1":100,"data2":200}
{"data1":100}

```

my **Arduino** Code
```
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial s(5,6);
 
void setup() {
  s.begin(1200);
}
 
void loop() {
  const size_t capacity = JSON_OBJECT_SIZE(2);
  DynamicJsonDocument doc(capacity);
  doc["data1"] = 100;
  doc["data2"] = 200;

  serializeJson(doc, s);
  delay(3000);
}
```

my ***NodeMCU*** Code
```
#include <SoftwareSerial.h>
SoftwareSerial s(D1,D2);
#include <ArduinoJson.h>
 
void setup() {

  Serial.begin(9600);
  s.begin(1200);
  while (!Serial) continue;
 }
 
void loop() {
    const size_t capacity = JSON_OBJECT_SIZE(2) + 20;    
    DynamicJsonDocument doc(capacity);

    delay(3000);
    deserializeJson(doc, s);
   
    int data1 = doc["data1"]; // 100
    int data2 = doc["data2"]; // 200
    serializeJson(doc, Serial);
    
    Serial.println();
    }
```

bblanchon, 2020-04-21:
@fatahillahardhi and I talked through the live chat and my conclusion is that these errors come from a noisy communication between the two boards.
Reducing the data rate improves the error ratio but it is still quite high.
I don't know how to fix this issue, but I'm sure there is nothing I can do on the library side.

@fatahillahardhi, feel free to comment or close this issue.

LenirSantiago, 2020-06-09:
I have the exact same issue... Actually I tried the code from the ArduinoJson tutorial [https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/](url) using two Arduino UNOs and the code works perfectly... As a matter of fact, in order to keep the code exactly the same, I used pins 4 and 5 for Rx and TX respectively. Then I tried the same receiver code on the nodemcu and i get failures... On the SoftwareSerial I went down on baudrate as low as 300 and showed no improvement.

So, after some troubleshooting, I really do not think  that these errors come from a noisy communication between the two boards.

Here are my finding using the following code:
**Sender:**
```
#include <ArduinoJson.h>
#include <SoftwareSerial.h>

// Declare the "link" serial port
// Please see SoftwareSerial library for detail
SoftwareSerial linkSerial(4, 5); // RX, TX

void setup() {
	// Initialize "debug" serial port
	// The data rate must be much higher than the "link" serial port
	Serial.begin(115200);
	while (!Serial) continue;

	// Initialize the "link" serial port
	// Use the lowest possible data rate to reduce error ratio
	linkSerial.begin(4800);
}

void loop() {
	// Values we want to transmit
	long timestamp = millis();
	int value = random(100);

	// Print the values on the "debug" serial port
	Serial.print("timestamp = ");
	Serial.println(timestamp);
	Serial.print("value = ");
	Serial.println(value);
	Serial.println("---");

	// Create the JSON document
	StaticJsonDocument<200> doc;
	doc["timestamp"] = timestamp;
	doc["value"] = value;

	// Send the JSON document over the "link" serial port
	serializeJson(doc, linkSerial);

	// Wait
	delay(5000);
}
```
**Receiver:**
```
#include <ArduinoJson.h>
#include <SoftwareSerial.h>

// Declare the "link" serial port
// Please see SoftwareSerial library for detail
SoftwareSerial linkSerial(10, 11); // RX, TX

void setup() {
  // Initialize "debug" serial port
  // The data rate must be much higher than the "link" serial port
  Serial.begin(115200);
  while (!Serial) continue;

  // Initialize the "link" serial port
  // Use the lowest possible data rate to reduce error ratio
  linkSerial.begin(4800);
}
 
void loop() {
[... 152 lines cut ...]
      Serial.println(err.c_str());
  
      // Flush all bytes in the "link" serial port buffer
      while (linkSerial.available() > 0)
        linkSerial.read();
    }
  }
}
```
Serial Monitor output:
```
14:01:01.446 -> {"timestamp":448494,"value":98}
14:01:01.479 -> timestamp = 448494
14:01:01.479 -> value = 98
14:01:06.525 -> {"timestamp":453533,"value":88}
14:01:06.525 -> timestamp = 453533
14:01:06.525 -> value = 88
14:01:11.578 -> {"timestamp":458573,"value":23}
14:01:11.578 -> timestamp = 458573
14:01:11.578 -> value = 23
14:01:16.670 -> {"timestamp":463613,"value":5}
14:01:16.670 -> timestamp = 463613
14:01:16.670 -> value = 5

```

bblanchon, 2020-06-25:
@fatahillahardhi, can you answer my questions?

closed 2020-07-17 by bblanchon