bblanchon/ArduinoJson #1245: Send Json data from client to server

lcfelippe, 2020-04-23:
Hi bblanchon, 

how can i send a serialized json data to a server(esp8266)?. Can you help me with it?

Thanks for the good job on ArduinoJson.

Regards,

lcfelippe.

bblanchon, 2020-04-23:
Hi @lcfelippe,

Please look at:

* [JsonServer.ino](https://arduinojson.org/v6/example/http-server/)
* [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/)

Best regards,
Benoit

lcfelippe, 2020-04-23:
bblancon, Thank you for your reply...

I have already tried this examples, but in this case the server send json data to a client. I need that the client read all the sensors and send json to a server. Maybe it's easy to do, but I'm newbie on http protocols.

Can you help me, how to make this?

Regards,

lcfelippe.

bblanchon, 2020-04-24:
Hi @lcfelippe,

Please see:

* [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/)

Best regards,
Benoit

lcfelippe, 2020-04-24:
Thank you for your patience.

I have no doubts about how to serialize or deserialize the json. My only problem is how to send it from client to server.

Regards,

lcfelippe.

bblanchon, 2020-04-25:
@lcfelippe, the serialization tutorial shows how to POST the data to a Web service.
What else do you need?

lcfelippe, 2020-04-27:
@bblanchon, i need to know how can i prepare the server to receive the Json data. My project is a Weather Station that collect data and send it to wunderground. Now, i would like to serialize this data  and send it to a Esp8266 (server) to display the weather information on ILI9341TFT Display. 

Below a example of my serialized data:

{"Weather":"Sensors","Date":{"Day":165,"Month":165,"Year":2165,"Weekday":"Monday"},"Time":{"Hour":165,"Minute":165,"Second":85},"WiFi":{"WiFi Signal Strenght":45},"BME280":{"Temperature":0,"Humidity":0,"DewPoint":0,"Pressure":0,"Pressure Equivalent":0},"VEML-6075":{"UVA":0,"UVB":0,"UV Index":0},"PMS 5003":{"Last Read Sample":{"PM 1.0":0,"PM 2.5":0,"PM 10":0},"Total of 24hs PM 2.5":0,"Average of 24hs PM 2.5":0,"PM 2.5 AQI":0,"Air Quality PM 2.5":"","Total of 24hs PM 10":0,"Average of 24hs PM 10":0,"PM 10 AQI":0,"Air Quality PM 10":"","Air Quality Comparison":""},"Rain Bucket":{"Rain Rate":0,"Rain Total":0},"Wind":{"Speed":0,"Direction Cardinal":"","Direction Degrees":0},"Circuit Temperature":{"Circuit":0,"Battery 1":0,"Battery 2":0}}

bblanchon, 2020-04-27:
Hi @lcfelippe,

I don't have a ready-made example of an HTTP server that deserializes a JSON document from the request body.
However, you can easily build from the two following examples:

* [JsonServer.ino](https://arduinojson.org/v6/example/http-server/) shows how to implement an HTTP server
* [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) shows how to deserialize a JSON document from the *response* body.

Combining the two programs should be fairly easy.

Best regards,
Benoit

lcfelippe, 2020-04-27:
Hi @bblanchon,

I don't have good skills with HTTP protocols, therefore the examples didn't help me to solve my necessity. I'll try to manage the data between the Esp8266 using a 433Mhz transmitter and send a structure instead of the Json data.

struct sensor {
  float temp;
  float pressure;
  float humidity;
};

Thank you for the help. 👍

ewaldc, 2020-04-28:
@lcfelippe,
As your client will submit the data to your server you need to use HTTP POST (or PUT) requests from the client side, rather than GET commands.
You can follow these examples:
- for the client: [client post](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient/examples/PostHttpClient)
- for the server: [simple (rest) server](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino)

First, you need to settle on an _URI_ where the client will post the JSON data and the server will accept it e.g. "_/rest/post_data_"
Then on the client code change the line accordingly to the chosen URI
`    http.begin(client, "http://"<SERVER_IP>"/postplain/"); //HTTP`
to
`    http.begin(client, "http://<SERVER_IP>/rest/post_data"); //HTTP`
Change <SERVER_IP> to your web server address

On the client side, also change the line of code
`    int httpCode = http.POST("{\"hello\":\"world\"}"); `
to your serialized json.   You can build your JSON data by simple string concatenation of keys and values.

On the server side, change these lines of code:
```
  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });
```
to
```
server.on("/rest/post_data", []() {
if (server.hasArg("plain")== false){ //Check if body received
    server.send(400, "text/plain", "Bad request: body not received");
} else { // we received a body, hopefully our JSON data
    String message = server.arg("plain");
    message += "\n";
    // Allocate the JSON document
    // Use arduinojson.org/v6/assistant to compute the capacity.
    const size_t capacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
    DynamicJsonDocument doc(capacity);

    // Parse JSON object
    DeserializationError error = deserializeJson(doc, message);
    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      return;
    }
    // incorporate the parsing of your JSON data as per the [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) example
}
}
```
Get rid of other _server.on()_ pieces of code, unless you want to extend your web server to serve more requests...

There is a good tutorial [here](https://techtutorialsx.com/2017/03/26/esp8266-webserver-accessing-the-body-of-a-http-request/)

Good luck...
Ewald

lcfelippe, 2020-05-04:
Hi @ewaldc, Thank you for the examples, and for the instructions to modify the code.

The only issue is that the Client only send the Json data to a server once. To send it again i have to reboot the client.

After read all sensors i have the code below on void loop

buffer was declared as global variable (String).

Code:
```
void loop(){
.
.
.

DynamicJsonDocument weather(2048);
   
  weather["Weather"]= "Sensors";
  weather["Date"]["Day"] = now.day();
  weather["Date"]["Month"] = now.month();
  weather["Date"]["Year"] = now.year();
  weather["Date"]["Weekday"] = daysOfTheWeek[now.dayOfTheWeek()];
  weather["Time"]["Hour"] = now.hour();
  weather["Time"]["Minute"] = now.minute();
  weather["Time"]["Second"] = now.second();
  weather["WiFi"]["WiFi Signal Strenght"] = Signal_Strength;
  weather["BME280"]["Temperature"] = tempc;
  weather["BME280"]["Humidity"] = humidity;
  weather["BME280"]["DewPoint"] = dewpointc;
  weather["BME280"]["Pressure"] = pressure;
  weather["BME280"]["Pressure Equivalent"] = slpressure_mB;
  weather["BME280"]["Heat Index"] = Heat_Index;
  weather["VEML-6075"]["UVA"] = uvia;
  weather["VEML-6075"]["UVB"] = uvib;
  weather["VEML-6075"]["UV Index"] = uvi;
  weather["PMS 5003"]["Last Read Sample"]["PM 1.0"] = PM_1wu;
  weather["PMS 5003"]["Last Read Sample"]["PM 2.5"] = PM_2_5wu;
  weather["PMS 5003"]["Last Read Sample"]["PM 10"] = PM_10wu;
  weather["PMS 5003"]["Total of 24hs PM 2.5"] = total24hs_PM_2_5;
  weather["PMS 5003"]["Average of 24hs PM 2.5"] = media24hs_PM_2_5;
  weather["PMS 5003"]["PM 2.5 AQI"] = AQI_PM_2_5;
  weather["PMS 5003"]["Air Quality PM 2.5"] = air_quality_PM_2_5;
  weather["PMS 5003"]["Total of 24hs PM 10"] = total24hs_PM_10;
  weather["PMS 5003"]["Average of 24hs PM 10"] = media24hs_PM_10;
  weather["PMS 5003"]["PM 10 AQI"] = AQI_PM_10;
  weather["PMS 5003"]["Air Quality PM 10"] = air_quality_PM_10;
  weather["PMS 5003"]["Air Quality Comparison"] = range_index;
  weather["Rain Bucket"]["Rain Rate"] = rainrate;
  weather["Rain Bucket"]["Rain Total"] = raintotal;
  weather["Wind"]["Speed"]= wind_speed_kmh;
  weather["Wind"]["Direction Cardinal"]= wind_direction_cardinal;
  weather["Wind"]["Direction Degrees"]= wind_direction_degrees;
  weather["Wind"]["Chill"]= Wind_Chill;  
  weather["Circuit Temperature"]["Circuit"]= temperature;
  weather["Circuit Temperature"]["Battery 1"] = batt_temp1;
  weather["Circuit Temperature"]["Battery 2"] = batt_temp2;

  Serial.print(F("Sending: "));
  serializeJson(weather, Serial);
  Serial.println();
  
  WiFiClient client;
  HTTPClient http;
  serializeJson(weather, buffer);
  
    Serial.print("[HTTP] begin...\n");
    // configure traged server and url
    http.begin(client, "http://"SERVER_IP"/rest/post_data"); //HTTP
    http.addHeader("Content-Type", "application/json");

[... 153 lines cut ...]
int Circuit_Temperature_Battery_2 = Circuit_Temperature["Battery 2"]; // 0

server.send(200, "text/plain", "Successfully Received...");

}
});

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  MDNS.update();

}
```

what could be wrong?

Thanks,

lcfelippe.

bblanchon, 2020-05-10:
Hi Icefelippe,

I don't know why you need to reboot; maybe you should try with a simpler example.
Anyway, I don't think this issue is related to ArduinoJson.

Best regards,
Benoit

lcfelippe, 2020-05-10:
Hi @bblanchon,

I solved it clean the buffer variable every time before serialize the JSON on client.

```
void loop(){
buffer = "";
.
.
.
}
```
Regards,

lcfelippe.

bblanchon, 2020-05-11:
Hi @lcfelippe,

Indeed, `serializeJson()` appends to `String`, just as it would do with a `Stream`.
You didn't show the declaration of `buffer`, so I didn't realize you were (re-)using a `String`.

Best regards,
Benoit

closed 2020-05-11 by bblanchon