bblanchon/ArduinoJson #1246: doc[StringVariable]; Is it possible?

Paytah232, 2020-04-23:
I'm using this on an ESP32 and I have basically this code:

char variableName[25] = "SomeVariable";
DynamicJsonDocument doc(config_size);
if (doc[variableName] != NULL) {
  //Do Stuff
}

When I get to the if statement the ESP32 crashes with this
[image]

I'm wondering if it is at all possible to pass 'variableName' to doc[variableName]?
Is it and I can't just use char[], because doc["SomeVariable"] where "SomeVariable" is a const char * would work I'd assume, however I am building the string in the char[] before I look for it in the doc.

Any help would be appreciated. Cheers!

bblanchon, 2020-04-24:
Hi @Paytah232,

I don't think the problem comes from the `char` array, but from `NULL`.

This issue required an extended response, so I wrote an article.
Please see:

* [Why can't I use `NULL`?](https://arduinojson.org/v6/faq/why-cant-i-use-null/)

Best regards,
Benoit

Paytah232, 2020-04-24:
You're right. I used nullptr instead and it works! Thanks!

bblanchon, 2020-04-24:
You're welcome.
Thank your for using ArduinoJson.

closed 2020-04-24 by bblanchon