bblanchon/ArduinoJson #1247: Can't return a JsonObject  on the ESP8266

sblantipodi, 2020-04-24:
Thanks for this wonderful library, for the wonderful documentation, and for the great work you have done here @bblanchon .

It's probably my fault but I'm not able to solve the problem at the moment, I "lack some piece" probably.

I have this easy function, this function get in input what I get in input from the pubsubclient mqtt library in the callback function AND it returns a json object with the deserialized output.

If the input received is not a json, it create a json and it return the received value in this format

`{"value":"receivedvalue"}`

here the function
```c++
JsonObject BootstrapManager::parseQueueMsg(char* topic, byte* payload, unsigned int length) {
  
  JsonDocument jsonDoc;
  
  char message[length + 1];
  for (unsigned int i = 0; i < length; i++) {
    message[i] = (char)payload[i];
  }
  message[length] = '\0';

  DeserializationError error = deserializeJson(jsonDoc, payload);

  // non json msg
  if (error) {    
    JsonObject root = jsonDoc.to<JsonObject>();
    root[VALUE] = message;
    String msg = jsonDoc[VALUE];
    return root;
  } else { // return json doc
    JsonObject root = jsonDoc.to<JsonObject>();
    return root;
  }

}
```

this function makes my ESP9266 restart with a stack trace.

this is the error:

[image]

bblanchon, 2020-04-25:
Hi @sblantipodi,

I can spot two errors in your program.

First, you cannot use [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) directly; you must use one of the two derivated classes [StaticJsonDocument](https://arduinojson.org/v6/api/staticjsondocument/) or [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/).
I'm surprised this code even compiles; when I do the same, I get `error: 'JsonDocument::JsonDocument()' is protected`.

Second, [JsonObject](https://arduinojson.org/v6/api/jsonobject/) is a reference that points to the data in the `JsonDocument`, so the value returned by this function points to a destructed object.
The solution is to return a `DynamicJsonDocument` or a `StaticJsonDocument`.
Alternatively, you can create the `JsonDocument` out of this function and pass a reference.

Best regards,
Benoit

sblantipodi, 2020-04-25:
you rock man! thanks and congrats.

closed 2020-04-25 by sblantipodi

bblanchon, 2020-04-26:
You're welcome @sblantipodi.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

sblantipodi, 2020-04-26:
done it :)