bblanchon/ArduinoJson #1248: Get stuck with serialisation

haeleng, 2020-04-27:
Hi, Benoit

I'm not a experienced programmer and do this for my hobby. I bought the book but get stuck with serialisation  of the example below, how to write C++ code for arduino.
I need this JSON message to get the systeminfo of my video camera via arduino with ethernetshield by means of HTTP.
The JSON message is
{
   "Request": {
       "Command":"GetSystemInfo",
       "SessionID":"10ffa8dcd1bb1cf5f9c252b7b8a20738"
    }
}
I do not know how to implement the Request. Is it just a String?, but what about the : { is it created automatically. As far as I understand is the Command the key and GetSystemInfo the value of an object. Same for SessionID and SessionID string
Could you please help and write this code, so I have a startoff.
Thanks in advance. You may send your answer to info@henklevels.nl

bblanchon, 2020-04-27:
Hi @haeleng,

Thank you very much for purchasing the book.

From your message, it's not clear whether you want to send or receive a JSON document, so I'll answer both.

If you read the [Chapter 4: Serializing with ArduinoJson](https://arduinojson.org/v6/doc/serialization/) to the end, you'll see that it shows how to establish a connection with an HTTP server and send a JSON document in a POST request.

Now, if you want to perform an HTTP GET request, simply look at [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/).

Best regards,
Benoit

haeleng, 2020-04-28:
Hi Benoit

Thanks. I red the book up to the end of chapter 4.
 But even after reading I get stuck.
As I said I'm a beginner. So please help.
I want to send (serialize) a JSON message. 
This is the Pan Tilt command I like to send to the camera
{
  "Request" : 
  { 
    "Command" : "SetPTCtrl" ,
    "SessionID" : "7425fbcb58ee4d15dd4c1f5ace4299a4" ,
    "Params" : 
    {
      "PanDirection" : "Left" ,
      "PanPosition" : 32768 ,
      "PanSpeed" : 10 ,
      "TiltDirection" : "Up" ,
      "TiltPositon" : 32768 ,
      "TiltSpeed" : 10 
    }
  } 
}

The code I made for a Arduino Uno was
	JsonObject objControl = docHL.createNestedObject("Request");
	objControl["Command"] = "SetPTCtrl";
	objControl["SessionID"] = "ThisIsTheSessionIDandThisIs32bit";
	JsonObject objParams = docHL.createNestedObject("Params");
	objParams["PanDirection"] = "Left";
	objParams["PanPosition"] = 0;
	objParams["PanSpeed"] = 0;
	objParams["TiltDirection"] = "Stop";
	objParams["TiltPosition"] = 0;
	objParams["TiltSpeed"] = 0;

	Serial.println(F("HL Request"));
	serializeJsonPretty(docHL, Serial);

However my code gives
{
  "Request": {
    "Command": "SetPTCtrl",
    "SessionID": "ThisIsTheSessionIDandThisIs32bit"
  },
  "Params": {
    "PanDirection": "Left",
    "PanPosition": 0,
    "PanSpeed": 0,
    "TiltDirection": "Stop",
    "TiltPosition": 0,
    "TiltSpeed": 0
  }
}

I want a nested params object within the Request object, when I understand it correctly.
How do I get Params within the Request object??	
Could you please help, how to generate the PanTilt command, because I'm stuck even after hours.

bblanchon, 2020-04-29:
Hi @bblanchon,

To nest "Params" in "Request," you must call `createNestedObject()` on the "Request" object, not on the root:

```c++
JsonObject objControl = docHL.createNestedObject("Request");
JsonObject objParams = objControl.createNestedObject("Params");
```

Also, remember that the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) writes the serialization code for you. Simply paste the expected JSON document in the box and scroll to the bottom of the page.

Best regards,
Benoit

haeleng, 2020-04-29:
Thanks very much for given me the answer Benoit

However the ArduinoJson Assistant does NOT write the serialization code for my nested code.
It just gives a general example, see below
When I fill out my JSON document it gives this result at the bottum of the page. Which is not a code for nested within nested.

const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3);
DynamicJsonDocument doc(capacity);

doc["sensor"] = "gps";
doc["time"] = 1351824120;

JsonArray data = doc.createNestedArray("data");
data.add(48.75608);
data.add(2.302038);

serializeJson(doc, Serial);

bblanchon, 2020-04-29:
Here is what I get when I paste the input from your previous message:

```c++
const size_t capacity = JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(6);
DynamicJsonDocument doc(capacity);

JsonObject Request = doc.createNestedObject("Request");
Request["Command"] = "SetPTCtrl";
Request["SessionID"] = "7425fbcb58ee4d15dd4c1f5ace4299a4";

JsonObject Request_Params = Request.createNestedObject("Params");
Request_Params["PanDirection"] = "Left";
Request_Params["PanPosition"] = 32768;
Request_Params["PanSpeed"] = 10;
Request_Params["TiltDirection"] = "Up";
Request_Params["TiltPositon"] = 32768;
Request_Params["TiltSpeed"] = 10;

serializeJson(doc, Serial);
```

As you can see, the object "Params" is created inside the "Request" object.

haeleng, 2020-04-29:
Thats really great!!

closed 2020-05-17 by bblanchon