bblanchon/ArduinoJson #1250: C2766 explicit specialization; 'specialization' has already been defined when compiling on Visual Studio with /Zc:__cplusplus flag

archieLa, 2020-04-27:
Since Visual Studio 2017 when /Zc:__cplusplus flag is used then the __cplusplus macro is evaluated correctly based on C++ standard used. When that happens the C2766 occurs since both ARDUINOJSON_HAS_LONG_LONG and ARUDINO_HAS_INT_64 are evaluated to be true and cause duplicate template specialization
https://github.com/bblanchon/ArduinoJson/blob/daa279d57b20736ba3bf39de935fa919661d8ecf/src/ArduinoJson/Configuration.hpp#L8

bblanchon, 2020-04-28:
Hi @archieLa,

Thank you very much for reporting this error.
It's fixed on branch "6.x".

Best regards,
Benoit

closed 2020-04-28 by bblanchon

archieLa, 2020-04-28:
No problem glad I could help👍

bblanchon, 2020-05-15:
This fix is available in [ArduinoJson 6.15.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.2)

commit 959b1d9e (2020-04-28): Fixed error C2766 with flag `/Zc:__cplusplus` on VS2017+ (fixes #1250)