bblanchon/ArduinoJson #1251: Sending empty objects to MongoDB

dcota, 2020-04-28:
Hi!
I am trying to POST json data from my Arduino Ethernet to a MongoDB via nodejs server. In the Arduino I get some readings from a sensor and then it tries to connect do the node server to post the json with serializeJson. The problem is that I am receiving empty objects in the database. I am sendian you help me please. Bellow is my code Thanks!
```
StaticJsonDocument<500> doc;
void cookJson(){
  doc["project"] = "project_name";
  doc["project_number"] = 6073;
  JsonArray temperature = doc.createNestedArray("temp");
  JsonArray pressure = doc.createNestedArray("press");
  while(count<=5){
    unsigned long currentMillis = millis();
    if (currentMillis - previousMillis >= interval){
      previousMillis=currentMillis;
      Serial.println(count);
      temp=bmp.readTemperature();
      temperature.add(temp);
      pre=bmp.readPressure()/100;
      pressure.add(pre);
      count++;
    }
  }
}
void setup() {
  //put your init code here
  Serial.begin(9600);
  bmp.begin();
  //init ethernet connection
  Ethernet.begin(mac, ip);
  cookJson();
  serializeJson(doc, Serial);
  Serial.println();
  serializeJsonPretty(doc, Serial);
  Serial.println();
  Serial.println("connecting...");
  if(client.connect(server, 5000))
    Serial.println("Connected to server");
  else 
    Serial.println("Connection failed");
  
  Serial.println("Begin POST Request");
  client.println("POST /api/cursos HTTP/1.1");
  client.println("Host: 192.168.1.88:5000");
  //client.println("User-Agent: Arduino/1.0");
  client.println("Content-Type: application/json");
  //client.println("Connection: close");
  client.print("Content-Length: ");
  client.println(measureJson(doc));
  Serial.println(measureJson(doc));
  client.println();
  serializeJson(doc, client);
  Serial.println("POST Request completed");

}

void loop() {
  //not used
}
```
This is what I get in my terminal:

[image]

And this is what I get in the server:
[image]

And this is my Mongoose schema:
[image]

Thank you for your help!
Duarte

bblanchon, 2020-04-29:
Hi,

Your program looks right and the output on the serial port confirms that the JSON document is correct.
However, I'm not familiar with MongoDB, so it might be a problem with the request.
Do you manage to perform the same operation with curl?

Best regards,
Benoit

dcota, 2020-04-29:
Hi, thank you.
After posting my question I tried to POST the same json generated in Arduino (copy/paste from the serial monitor) using Insomnia and I was getting the same empty stuff. So, I realized that the problem was from the Arduino. After analysing once more the server code I saw the problem. It was in a function that routes the request to the MongoDB. Now it's working fine! Sorry to bother you and thank you very much.
Best regards,
Duarte

bblanchon, 2020-04-29:
You're welcome, Duarte.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2020-04-29 by bblanchon