bblanchon/ArduinoJson #1252: Parse JSON object failed using WiFiClientSecure

v11, 2020-04-28:
Hi Benoît

Thank you for providing ArduinoJson and have this wonderful documentation.
I am using an ESP866 to get some data to my e-ink display. 

At the moment i am testing the HTTPS Request to a Test-Server. I am using WiFiClientSecure. I can connect and i can receive the JSON as String (with client.readString();).

But i have no idea, how to give that information over for parsing. In an example i have seen that you can give the client directly as an argument (deserializeJson(doc, client)). But it is not working.

Do you have an idea, how to solve that problem?

Cheers, Lukas

Code
```
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

// Include JSON Library https://arduinojson.org
#include <ArduinoJson.h>

#ifndef STASSID
#define STASSID "Lukas WiFi"
#define STAPSK "*********"
#endif

const char *ssid = STASSID;
const char *password = STAPSK;

const char *host = "jsonplaceholder.typicode.com";

// HTTPS Port
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char fingerprint[] PROGMEM = "06 68 CF CF 1D 2C E8 30 5F DF 33 C9 33 C3 10 01 22 13 15 1E";

void setup()
{
    // Wifi
    Serial.begin(115200);
    Serial.println();
    Serial.print("connecting to ");
    Serial.println(ssid);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    // Use WiFiClientSecure class to create TLS connection
    WiFiClientSecure client;
    Serial.print("connecting to ");
    Serial.println(host);

    //Serial.printf("Using fingerprint '%s'\n", fingerprint);
    client.setFingerprint(fingerprint);

    if (!client.connect(host, httpsPort))
    {
        Serial.println("connection failed");
        return;
    }

    String url = "/users";
    Serial.print("requesting URL: ");
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: close\r\n\r\n");

    Serial.println("request sent");
    while (client.connected())
    {
        String line = client.readStringUntil('\n');
        if (line == "\r")
        {
            Serial.println("headers received");
            break;
        }
    }

    // String line = client.readString();

    // JSON
    // Allocate the JSON document
    const size_t capacity = JSON_ARRAY_SIZE(10) + 10 * JSON_OBJECT_SIZE(2) + 10 * JSON_OBJECT_SIZE(3) + 10 * JSON_OBJECT_SIZE(5) + 10 * JSON_OBJECT_SIZE(8) + 3730;
    DynamicJsonDocument doc(capacity);

    // Parse JSON object
    DeserializationError error = deserializeJson(doc, client);

    if (error)
    {
        Serial.print(F("deserializeJson() failed: "));
        Serial.println(error.c_str());
        return;
    }

    JsonObject root_0 = doc[0];
    Serial.println("JSON Docss");
    Serial.println(root_0);

    // Get the Name:
    const char *root_0_name = root_0["name"];

    // if (line.startsWith("{\"state\":\"success\""))
    // {
    //     Serial.println("esp8266/Arduino CI successfull!");
    // }
    // else
    // {
    //     Serial.println("esp8266/Arduino CI has failed");
    // }
    Serial.println("Name was");
    Serial.println("==========");
    Serial.println(root_0_name);
    Serial.println("==========");
    Serial.println("closing connection");
}
```

Output
```
WiFi connected
IP address: 
192.168.0.158
connecting to jsonplaceholder.typicode.com
requesting URL: /users
request sent
headers received
deserializeJson() failed: InvalidInput
```

bblanchon, 2020-04-29:
Hi @v11,

As suggested in the documentation for [DeserializationError::InvalidInput](https://arduinojson.org/v6/api/misc/deserializationerror/#invalidinput), this error is caused by the [chunk transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).
Passing `HTTP/1.0` instead of `HTTP/1.1` in the request is enough to fix the problem.

BTW, to diagnose this issue, I used `ReadLoggingStream`, as suggested in [How to debug data coming from a stream?](https://arduinojson.org/v6/how-to/debug-data-coming-from-a-stream/).
Thanks to this trick, I saw `160d` in the console, which is the size of the chunk in hexadecimal.
This value was already present in the `String` returned by `readString()`.

Best regards,
Benoit

v11, 2020-04-29:
Hi Benoit

Thank you very much! It is working now. 

```
client.print(String("GET ") + url + " HTTP/1.0\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: close\r\n\r\n");
```

Cheers from Switzerland,
Lukas

closed 2020-04-29 by v11

bblanchon, 2020-04-29:
You're welcome Lukas.
Good luck with your project!