bblanchon/ArduinoJson #1253: i need help to deserializeJson to array using loop

gylmd, 2020-04-29:
my json is   
[{"s1":[77,0,22,0,77,0,22,0,33,0,0],"s2":[66,0,11,0,22,0,33,0,0,0,0],"s3":[0,0,0,0,0,0,0,0,0,0,0],"s4":[0,0,0,0,0,0,0,0,0,0,0]},{"s1":[0,0,0,0,0,0,0,0,0,0,0],"s2":[0,0,0,0,0,0,0,0,0,0,0],"s3":[77,0,22,0,77,0,22,0,33,0,0],"s4":[77,0,22,0,77,0,22,0,33,0,0]}]
i want to set value into array like tike this my_array[ i ][ j ][ k ]
i=2   j=4   k=11

i get very long code from website https://arduinojson.org/v6/assistant/

bblanchon, 2020-04-29:
Hi @gylmd,

Did you try this:

```c++
doc[2]["s4"][11] = X;
```

Best regards,
Benoit

gylmd, 2020-04-29:
> Hi @gylmd,
> 
> Did you try this:
> 
> ```c++
> doc[2]["s4"][11] = X;
> ```
> 
> Best regards,
> Benoit

i want to make loop

gylmd, 2020-04-29:
thank u Benoît Blanchon it is work

closed 2020-04-29 by gylmd