bblanchon/ArduinoJson #1255: How to extract an array from json data

joelzavala, 2020-04-30:
Hi, I'm working with an esp8266 and using the ArduinoJson and ESP8266HTTPClient libraries.
I have a data buffer (raw infrared codes) of type int in my server. 
Example:
[1, 2, 3, 4, 5] -> How is it displayed on the server
That buffer is what returns me the ESP8266HTTPClient library as payload and string type.
Then I get that payload (the variable) and deserialize it with the ArduinoJson library.
Now, what I need is to put that array or buffer in an unsigned int buffer variable, how do I do it with the arduino json library?

bblanchon, 2020-05-01:
Hi @joelzavala,

The simplest way to extract an array of (unsigned) integers with ArduinoJson is to use the [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/) helper function.

Best regards,
Benoit

joelzavala, 2020-05-01:
Hello again Benoit :).
I used the function you told me about earlier and it generated the following error

no matching function for call to 'copyArray(ArduinoJson6151_0000010::DynamicJsonDocument&, unsigned int [((sizetype)<anonymous>) + 1)])

My code is as follows:
`void solicitarBuffer(String request) {
  
  String payload;

  sendPost(request, &payload);

  if (payload == "false") {
    Serial.println("Error");
    return;
  } else {

    DynamicJsonDocument doc(1000);

    DeserializationError error = deserializeJson(doc, payload);

    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      return;
    } else {

      unsigned int buffer[doc.size()];
      copyArray(doc, buffer);

      /*uint16_t buffer[doc.size()];
        for (int i = 0; i < doc.size(); i++) {
        buffer[i] = (unsigned int)doc[i];
        }*/

      irsend.sendRaw(buffer, doc.size(), 38);

    }
  }
}`

bblanchon, 2020-05-02:
Hi @joelzavala,

Indeed, that's a problem I need to fix: `copyArray()` should accept a `JsonDocument` directly.
Meanwhile, you must explicitly cast it to a `JsonArray`:

```c++
copyArray(doc.as<JsonArray(), buffer);
```

Best regards,
Benoit

bblanchon, 2020-05-15:
The fix is available in [ArduinoJson 6.15.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.2)

closed 2020-05-28 by bblanchon