bblanchon/ArduinoJson #1256: De/Serialize enum type with JSON

lloydrichards, 2020-05-03:
Greetings,

I'm trying to set up a state machine using an esp32 and its deep sleep functionality.  I want to save the current state to the SPIFFS using JSON  and then read it again after wake up.  I'm doing this already with other variables but when I try to include my SENSOR_STATUS and BATTERY_STATUS which are enum types in my library I get errors about the type.

I've been trying to figure out the JsonVariant::as<T> but to no avail.

Basically, I have an enum:

```
enum STATUS_BATTERY
{
  CRITICAL_BATTERY,
  LOW_BATTERY,
  GOOD_BATTERY,
  FULL_BATTERY,
  CHARGING
};
```
And i want to save it as a JSON so i can store and retrieve the values.  I recreated two functions
```
void MimirTesting::initConfig()
{
    if (SPIFFS.begin())
    {

        Serial.println("mounted file system");
        if (SPIFFS.exists("/config.json"))
        {
            //file exists, reading and loading
            Serial.println("reading config file");
            fs::File configFile = SPIFFS.open("/config.json", "r");
            if (configFile)
            {
                Serial.println("opened config file");
                size_t size = configFile.size();
                // Allocate a buffer to store contents of the file.
                std::unique_ptr<char[]> buf(new char[size]);

                configFile.readBytes(buf.get(), size);
                DynamicJsonDocument configJson(1024);
                DeserializationError error = deserializeJson(configJson, buf.get());
                if (error)
                {
                    Serial.println("failed to load json config");
                    return;
                }
                Serial.println("\nparsed json");
                serializeJson(configJson, Serial);

                JsonVariant _SENSOR_STATUS = configJson["SENSOR_STATUS"];
                SENSOR_STATUS = _SENSOR_STATUS.as<STATUS_ERROR>();
                JsonVariant _WIFI_STATUS = configJson["WIFI_STATUS"];
                WIFI_STATUS = _WIFI_STATUS.as<STATUS_ERROR>();
                JsonVariant _SERVER_STATUS = configJson["SERVER_STATUS"];
                SERVER_STATUS = _SERVER_STATUS.as<STATUS_ERROR>();
                JsonVariant _MICROSD_STATUS = configJson["MICROSD_STATUS"];
                MICROSD_STATUS = _MICROSD_STATUS.as<STATUS_ERROR>();
                JsonVariant _SHT31D_L_STATUS = configJson["SHT31D_L_STATUS"];
            }
            else
            {
                Serial.println("failed to load json config");
            }
        }
    }
    else
    {
        Serial.println("failed to mount FS");
    }
}
```
and
```
void MimirTesting::saveConfig()
{
    DynamicJsonDocument newConfigJson(1024);
    newConfigJson["SENSOR_STATUS"] = SENSOR_STATUS;
    newConfigJson["WIFI_STATUS"] = WIFI_STATUS;
    newConfigJson["SERVER_STATUS"] = SERVER_STATUS;
    newConfigJson["MICROSD_STATUS"] = MICROSD_STATUS;

fs::File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile)
    {
        Serial.println("failed to open config file for writing");
    }

    //serializeJson(newConfigJson, Serial);
    serializeJson(newConfigJson, configFile);
    configFile.close();
}
```
But running this I get this error:

```
lib/MimirTesting/MimirTesting.cpp:357:65:   required from here
.pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantRef.hpp:259:24: error: no matching function for call to 'variantAs(ArduinoJson6150_0000010::VariantData* const&)'
     return variantAs<T>(_data);
                        ^
In file included from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Operators/VariantOr.hpp:9:0,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Operators/VariantOperators.hpp:9,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantRef.hpp:12,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson.hpp:17,
                 from .pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson.h:9,
                 from lib/MimirTesting/MimirTesting.cpp:16:
.pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantAs.hpp:55:59: note: candidate: template<class T> typename ArduinoJson6150_0000010::enable_if<ArduinoJson6150_0000010::is_integral<T>::value, T>::type ArduinoJson6150_0000010::variantAs(const ArduinoJson6150_0000010::VariantData*)
 inline typename enable_if<is_integral<T>::value, T>::type variantAs(
                                                           ^
.pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantAs.hpp:55:59: note:   template argument deduction/substitution failed:
.pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantAs.hpp: In substitution of 'template<class T> typename ArduinoJson6150_0000010::enable_if<ArduinoJson6150_0000010::is_integral<T>::value, T>::type ArduinoJson6150_0000010::variantAs(const ArduinoJson6150_0000010::VariantData*) [with T = STATUS_ERROR]':
.pio/libdeps/esp32dev/ArduinoJson_ID64/src/ArduinoJson/Variant/VariantRef.hpp:259:24:   required from 'typename ArduinoJson6150_0000010::enable_if<(((! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::ArrayRef>::value) && (! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::ObjectRef>::value)) && (! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::VariantRef>::value)), typename ArduinoJson6150_0000010::VariantAs<T>::type>::type ArduinoJson6150_0000010::VariantRef::as() const [with T = STATUS_ERROR; typename ArduinoJson6150_0000010::enable_if<(((! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::ArrayRef>::value) && (! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::ObjectRef>::value)) && (! ArduinoJson6150_0000010::is_same<T, ArduinoJson6150_0000010::VariantRef>::value)), typename ArduinoJson6150_0000010::VariantAs<T>::type>::type = STATUS_ERROR]'
```

I'm a little lost now and not sure where to go so if anyone has any suggestions on what I'm missing, I would be very grateful!

Thanks

bblanchon, 2020-05-04:
Hi @lloydrichards,

Thank you very much for opening this issue.
To me, it should work out of the box, so I'm flagging this as a bug.

Until I fix this bug, please pass through an `int`:

```c++
STATUS_ERROR status = (STATUS_ERROR)doc["status"].as<int>();
```

See a demo here: https://wandbox.org/permlink/EdC0aVUDTdJZo9QR

Best regards,
Benoit

closed 2020-05-04 by bblanchon

lloydrichards, 2020-05-04:
Thanks so much.  The quick fix worked like a charm and now I see you've committed this change so will go ahead and test the first way.  Much appreciated for you quick, helpful response :D

bblanchon, 2020-05-04:
You're welcome @lloydrichards.
Thanks again for reporting this issue.

bblanchon, 2020-05-15:
This fix is available in [ArduinoJson 6.15.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.2)

commit 3bb0a7aa (2020-05-04): Added support for `enum`s in `as<T>()` and `is<T>()`  (fixes #1256)