bblanchon/ArduinoJson #1258: Arduinojson shows the Values as Zeros after deserialization

kota107, 2020-05-05:
I am try to connect Arduino Nano to NodeMCU using serial connection and the send the Data to a Webserver.
I used for this Arduinojson library  to serialize the data in the Code of the Nano-board und deserialize in the code of NodeMCU.
The Work of the Arduino Nano works and the data are correct on the serialmonitor
but in the Serial Monitor of NodeMCU I got  the following:

 deserializeJson() returned InvalidInput
 0.00,0.00000,0.00000,0.00,0.00000,0.00000,0.00000

and some times I got 
71683.55,1.69093,1.22043,0.00,0.00000,0.00000,0.00000  which represnt only the first 3 Vales out of 7 Values

anyone can Help me ?

the code of the Nano
`const size_t capacity = JSON_OBJECT_SIZE(50) + 100;
  DynamicJsonDocument konzentration(1024); 
  konzentration.capacity();
  konzentration["CO2"] = ppm_CO2;
  konzentration["CO"] = ppm_CO;
  konzentration["CH4"] = ppm_CH4;
  konzentration["NOx"] = ppm_NOx;
  konzentration["NH3"] = ppm_NH3;
  konzentration["H2S"] = ppm_H2S;
  konzentration["VOC"] = ppm_VOC;
  serializeJsonPretty(konzentration, Serial);`

the code of the NodeMCU
` const size_t capacity = JSON_OBJECT_SIZE(50) + 100;
   DynamicJsonDocument konzentration(1024);
   konzentration.capacity();
   DeserializationError err = deserializeJson(konzentration,Serial2020);
   if (err) {
   Serial.print(F("deserializeJson() returned "));
   Serial.println(err.c_str());
   //Serial.println();

   float CO2 = konzentration["CO2"];
   float CO  = konzentration["CO"];
   float CH4 = konzentration["CH4"];
   float NOx = konzentration["NOx"];
   float NH3 = konzentration["NH3"];
   float H2S = konzentration["H2S"];
   float VOC = konzentration["VOC"];
    
    Serial.print(CO2);  Serial.print(",");
    Serial.print(CO, 5);  Serial.print(",");
    Serial.print(CH4, 5);  Serial.print(",");
    Serial.print(NOx);  Serial.print(",");
    Serial.print(NH3, 5);  Serial.print(",");
    Serial.print(H2S, 5); Serial.print(",");
    Serial.print(VOC, 5);
    Serial.println();`

bblanchon, 2020-05-06:
Hi @kota107,

`InvalidInput` means that the JSON input was invalid.
This often comes from errors on the serial link; try to reduce the baud rate.

Best regards,
Benoit

bblanchon, 2020-05-06:
The errors often come from voltage mismatch and clock skew.
Please see:

* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

kota107, 2020-05-07:
Thanks so much I reduced the baud rate to 1200 for the "linkSerial" and also for the Serial to 19200 and it works correctly.

closed 2020-05-07 by kota107