bblanchon/ArduinoJson #1260: DynamicJsonDocument returns a null object

sblantipodi, 2020-05-06:
Hi all,
I have a class with a function like this

```
DynamicJsonDocument myClass::readConfigFile() {
    // ...
    DynamicJsonDocument doc(2048);
    deserializeJson(doc, file);

String val = doc["value"];
Serial.print(val); // IT WORKS OOK HERE
    // here doc["value"] is correctly 
    return doc;
}
```

when In another class I call:
DynamicJsonDocument doc = myclassInstance.readConfigFile();
String val = doc["value"];
Serial.print(val); // HERE VAL IS EMPTY

Why val is empty when I try to extract it from a function call?

Thanks

bblanchon, 2020-05-07:
Hi,

I tried to reproduce the issue, but it works as expected.
See online demo: https://wandbox.org/permlink/XrgdlP9UNQZEemNV

However, if `file` is of type `char*`, [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) uses the "zero-copy mode" and stores pointers instead of copies, so you must ensure that the input buffer stays in memory.
A simple workaround is to cast to `const char*` or pass the `File` instance directly to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).

Best regards,
Benoit

sblantipodi, 2020-05-07:
I am trying to write SPIFFS in the ESP8266, it seems that closing the file before returning the DynamicJsonDocument solved the problem.

I'm not using char *, I'm storing a DynamicJsonDocument inside SPIFFS,
the document I'm storing contains "String"

Still don't understand why I had the problem since no char* has been used.

Thanks for the answer, for your time and for the awesome work.
Arduino would not be that good without your lib.

bblanchon, 2020-05-08:
I don't understand either.
Please reopen this issue if you need.

closed 2020-05-08 by bblanchon