bblanchon/ArduinoJson #1261: Accessing STRAVA and request Data - RAM Problem?

v11, 2020-05-07:
Hei Benoit

I use your library intensively by now and basically everything works with my ESP8266. I have a sketch that works like this:

- Create a secure connection (WiFiClientSecure)
- Connect with Strava
- Authentication with Strava API
- Receive and parse data with ArduinoJSON

It all works and I can test it successfully.

Now I want to output the data to my E-Ink display. There I have also examples which work. Well, if I put it all together, it doesn't work anymore and I get strange characters in the log after parsing.

I don't think this has anything to do with ArduinoJson. Is it possible that I have reached the Ram or power limit of the microcontroller?

I don't expect you to look at it closely but maybe you can judge that in a few seconds. Otherwise just close it. :-)

Thanks for that,
Best regards from Switzerland.

Output
```
WiFi connected
IP address: 192.168.0.158
connecting to www.strava.com
requesting URL: www.strava.com/oauth/token?client_id=47136&client_secret=*******refresh_token=*******&grant_type=refresh_token
request sent
headers received
Access Token: *******
closing connection
requesting URL: www.strava.com/api/v3/athletes/20405053/stats?access_token=*******
request sent
headers received
Lon�s_PowerOn : 9
_Update_Full : 9
_PowerOff : 1
```

Code
```
// Wifi Setup: PreProcessor: Chech if SSID was already defined
#ifndef STASSID
#define STASSID "Lukas WiFi"
#define STAPSK "**********"
#endif

// Wifi: Library: esp8266 Wifi support. Based on WiFi.h from Arduino WiFi shield library.
#include <ESP8266WiFi.h>

// Wifi: Library: Variant of WiFiClient with Transport Layer Security (TLS) support. Deprecated predecessor: Secure Sockets Layer (SSL)
#include <WiFiClientSecure.h>

// NTP: Library: Library to get the current time. Needs internet connection.
#include <NTPClient.h>
#include <WiFiUdp.h>

// JSON: Library: Include JSON Library https://arduinojson.org
#include <ArduinoJson.h>

// Wifi: Set WiFi Credentials, Host and Port (443)
const char *ssid = STASSID;
const char *password = STAPSK;
const char *host = "www.strava.com";
const int httpsPort = 443;

// Wifi: SHA1 fingerprint of the certificate, rendered as a hexadecimal number, 40 digits long
// Wifi: Use the web browser to show the certficate and read it from there
const char fingerprint[] PROGMEM = "78 7C 22 62 BF 73 0B AC 99 3A F3 6E 1B EE 5E D9 B8 1A 68 64";

// GxEPD2: Base class GxEPD2_GFX can be used to pass references or pointers to the display instance as parameter, uses ~1.2k more code.
[... 253 lines cut ...]
        display.setCursor(30, 30 + 12);

        // Print some text
        display.print("Strava: Biggest Ride Distance: ");
        display.println(biggest_ride_distance);

        display.setCursor(30, 60 + 12);
        display.print("Recent Ride Totals: ");
        display.print(recent_ride_totals_count);

        display.setCursor(30, 90 + 12);
        display.print("All Ride Totals: ");
        display.print(all_ride_totals_count);

        // Draw Rect
        display.drawRect(10, 10, 780, 460, GxEPD_BLACK);

    } while (display.nextPage());
}

void loop()
{
}

```

bblanchon, 2020-05-08:
Hi @v11,

I think you're running out of RAM.
At the end of the compilation, there are two lines that tell how much RAM and how much Flash you use. Make sure there is a lot of free RAM left

To reduce RAM consumption, start by moving some strings to Flash, especially at the beginning, where you compose the URL, and the GET request.
Have a look at: 
* [How to format strings without the String class](https://cpp4arduino.com/2020/02/07/how-to-format-strings-without-the-string-class.html)
* [8 tips to use the String class efficiently](https://cpp4arduino.com/2018/11/21/eight-tips-to-use-the-string-class-efficiently.html), number 8 in particular.

I don't know how fonts are implemented, but I bet they use some global data.
Get rid of all unused font headers and reduce the number of fonts you actually use.

Regarding ArduinoJson, it seems that everything is fine 👍 

Best regards,
Benoit

v11, 2020-05-08:
Hi Benoit

Thank you very much for this Tipps. I will check this out. Hopefully i can solve this. :-)
Have a nice day. You can close this issue!

Lukas

bblanchon, 2020-05-08:
You're welcome.
Good luck with your project.

closed 2020-05-08 by bblanchon

v11, 2020-05-11:
I could solve it. It was not the RAM. I just had to init the Display (Waveshare E-Ink) in the Setup before everything else. Have no Idea why it is working, but it works! :-)