bblanchon/ArduinoJson #1262: creating auto iterate nested object in ArduinoIDE

ajakmyslisz, 2020-05-08:
hello, how can i do an auto iterate nested object?
 i make a device to scan near beacons and i want to send it to server but i dont know number of beacons (this value can be changed at time max of 64) 
 i have JSON like this:
`{
  "code": "string",
  "cubes": [
    {
      "uuid": "string",
      "major": 0,
      "minor": 0,
      "range": 0,
      "battery_level": "string"
    },
{
      "uuid": "string",
      "major": 0,
      "minor": 0,
      "range": 0,
      "battery_level": "string"
    }
]
}
`
after this code i generate a serialization:
`doc["code"] = "string";

JsonArray cubes = doc.createNestedArray("cubes");

JsonObject cubes_0 = cubes.createNestedObject();
cubes_0["uuid"] = "string";
cubes_0["major"] = 0;
cubes_0["minor"] = 0;
cubes_0["range"] = 0;
cubes_0["battery_level"] = "string";

JsonObject cubes_1 = cubes.createNestedObject();
cubes_1["uuid"] = "string";
cubes_1["major"] = 0;
cubes_1["minor"] = 0;
cubes_1["range"] = 0;
cubes_1["battery_level"] = "string";

....
`

but i need to add object like cubes_X (for 0 to number of beacon) in the loop 

how can i do this ?
this is even possible in Arduino IDE to generate object name (cube_X) in a loop like for()  and after reach  a number of beacon serialize to String and send to server? 

I work on ESP32 with 8Mb of memmory so i dont care about size of string and number of beacon (in the design assumptions i establish mux number of beacons at level of 64)

bblanchon, 2020-05-08:
Hi @ajakmyslisz,
I'm sorry

ajakmyslisz, 2020-05-08:
i want auto create object with arduinoIDE and JsonObject cubes_1 = cubes.createNestedObject();
how can i make cubes_1 cubes_2 etc. without hardcode this?

can i use something like for loop to do this?
for(int i=0;i<NumberOfObject;i++){
JsonObject cubes_[from 0 to NumberOfObject] = cubes.createNestedObject();
}

Name of cubes_X must be unique and i want to generate this via ArduinoIDE but is it posible to meke something like array of names aray[64]={cube_1,cube_2... etc};
and after this use this array to auto increment names of nested object? 
or you can find better solution to generate large structures of nestedobject

bblanchon, 2020-05-09:
I don't think it's possible to do that.
Why do you need the name of the variable to be unique?

ajakmyslisz, 2020-05-09:
> I don't think it's possible to do that.
> Why do you need the name of the variable to be unique?

this val will be sent to server  like other object and that is a reason to  do auto increment object name, Object name must be predefinied in Liblary(hard coded) but wat in sytuation when i scan about  30 beacon and in the code i definied 64  other will be empty and i dont want to send epmpy record to sever

bblanchon, 2020-05-10:
If I understand correctly, your question is 

> is it possible to generate the name of a C++ variable dynamically at run-time?

In that case, the answer is clearly **no**.

Now, if you want to share a bunch of values between two compilation units, simply use an array.

Anyway, this question is not about ArduinoJson.

ajakmyslisz, 2020-05-10:
> If I understand correctly, your question is
>
>
> In that case, the answer is clearly **no**.
>
> Now, if you want to share a bunch of values between two compilation units, simply use an array.
>
> Anyway, [...]

ok soo if i want to send to server about 64 record of beacon using ArduinoJson i need to hard-code (predefinied ) all of 
`JsonObject cubes_X = cubes.createNestedObject();
cubes_X["uuid"] = "string";
cubes_X["major"] = 0;
cubes_X["minor"] = 0;
cubes_X["range"] = 0;
cubes_X["battery_level"] = "string"`
and after this use for() loop to fill this structure?

bblanchon, 2020-05-10:
You can call `createNestedObject()` in a loop, and reuse the name `cube_X` inside the loop scope.

ajakmyslisz, 2020-05-11:
ok, i do this like
`const size_t capacity = JSON_ARRAY_SIZE(64) + JSON_OBJECT_SIZE(2) + 64 * JSON_OBJECT_SIZE(5);
  DynamicJsonDocument doc(capacity);
  doc["code"] = numerId;

  JsonArray cubes = doc.createNestedArray("cubes");`

and in the switch case  definition:
`switch (BeaconNumber) {
        case 0: {
            JsonObject cubes_0 = cubes.createNestedObject();
            cubes_0["uuid"] = BeaconUUID;
            cubes_0["major"] = BeaconMajor;
            cubes_0["minor"] = BeaconMinor;
            cubes_0["range"] = wynik3;
            cubes_0["battery_level"] = "65%";
            break;
          }`

this solution generate a lot of code but it will finally work for me 
so thanks a loot of your help  bblanchon

issue to close.

closed 2020-05-11 by ajakmyslisz