bblanchon/ArduinoJson #1263: Build fails with esp-idf

sebasje, 2020-05-08:
Hi Benoit!

I'm trying to build ArduinoJson within esp-idf. After adjusting the CmakeLists.txt, it tries to compile but fails with an error that I don't really understand. (g++ xtensa-esp32-elf-g++ (crosstool-NG esp-2020r1) 8.2.0). This is the error I get:

In file included from ../components/ArduinoJson/src/ArduinoJson/Operators/VariantOr.hpp:9,
                 from ../components/ArduinoJson/src/ArduinoJson/Operators/VariantOperators.hpp:8,
                 from ../components/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:12,
                 from ../components/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from ../components/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from ../components/ArduinoJson/src/ArduinoJson.hpp:17,
                 from ../components/ArduinoJson/src/ArduinoJson.h:9,
                 from ../main/effectconfigs.h:6,
                 from ../main/effectconfigs.cpp:1:
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'T ArduinoJson6151_1100010::variantAs(ArduinoJson6151_1100010::VariantData*, ArduinoJson6151_1100010::MemoryPool*) [with T = String]':
../components/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:256:50:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::VariantRef::as() const [with T = String; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'
../components/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:75:52:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::MemberProxy<TParent, TStringRef>::as() const [with TValue = String; TObject = ArduinoJson6151_1100010::ObjectRef; TStringRef = const char*; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'
../main/effectconfigs.cpp:70:38:   required from here
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:92:22: error: no matching function for call to 'variantAs<String>(ArduinoJson6151_1100010::VariantData*&)'
   return variantAs<T>(data);
          ~~~~~~~~~~~~^~~~~~
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:56:59: note: candidate: 'template<class T> typename ArduinoJson6151_1100010::enable_if<ArduinoJson6151_1100010::is_integral<T>::value, T>::type ArduinoJson6151_1100010::variantAs(const ArduinoJson6151_1100010::VariantData*)'
 inline typename enable_if<is_integral<T>::value, T>::type variantAs(
                                                           ^~~~~~~~~
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:56:59: note:   template argument deduction/substitution failed:
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp: In substitution of 'template<class T> typename ArduinoJson6151_1100010::enable_if<ArduinoJson6151_1100010::is_integral<T>::value, T>::type ArduinoJson6151_1100010::variantAs(const ArduinoJson6151_1100010::VariantData*) [with T = String]':
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:92:22:   required from 'T ArduinoJson6151_1100010::variantAs(ArduinoJson6151_1100010::VariantData*, ArduinoJson6151_1100010::MemoryPool*) [with T = String]'
../components/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:256:50:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::VariantRef::as() const [with T = String; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'
../components/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:75:52:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::MemberProxy<TParent, TStringRef>::as() const [with TValue = String; TObject = ArduinoJson6151_1100010::ObjectRef; TStringRef = const char*; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'
../main/effectconfigs.cpp:70:38:   required from here
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:56:59: error: no type named 'type' in 'struct ArduinoJson6151_1100010::enable_if<false, String>'
../components/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'T ArduinoJson6151_1100010::variantAs(ArduinoJson6151_1100010::VariantData*, ArduinoJson6151_1100010::MemoryPool*) [with T = String]':
../components/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:256:50:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::VariantRef::as() const [with T = String; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'
../components/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:75:52:   required from 'typename ArduinoJson6151_1100010::VariantAs<T>::type ArduinoJson6151_1100010::MemberProxy<TParent, TStringRef>::as() const [with TValue = String; TObject = ArduinoJson6151_1100010::ObjectRef; TStringRef = const char*; typename ArduinoJson6151_1100010::VariantAs<T>::type = String]'

(full log: https://pastebin.com/UwP9WGA4 )

I built the 6.x branch and also tried the 6.5.1 tag, both expose the same error.

I had to adjust the CMakeLists.txt, since esp-idf doesn't understand all the scriptables (it basically allows to set some vars), but I understand that the above error doesn't stem from compiler options? (If so, which? I can try to set that...)

Love your project by the way, the book is a nice cherry on top of a well-designed library!

bblanchon, 2020-05-08:
Hi,

ArduinoJson assumes you can use `String` only if `ARDUINO` is defined.
You can override this behavior by manually setting [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/) to 1.

Best regards,
Benoit

sebasje, 2020-05-08:
Wow, that was superquick and helpful! It builds now.

Thanks a lot!

closed 2020-05-08 by sebasje

bblanchon, 2020-05-08:
You're welcome @sebasje.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: