bblanchon/ArduinoJson #1264: nesting an array inside object

peppeg85, 2020-05-08:
hello, this is not an issue, is a question, i have to send this json:
````
{
 u:"username",
p:"password",
id:2,
d:[{
        tscd: 0,
        tbme: 0,
        hscd: 0,
        hbme: 0,
        cscd: 0,
        M1: 1.89,
        M2: 2.22,
        M4: 3.33,
        M10: 44.44,
        P1: 16.51,
        P2: 0,
        P4: 0,
        P10: 0,
        PSize:0,
        no2: 0,
        out: 0,
        pressure: 0,
        h: 0, 
        tstamp: 0
     }]
}
````
so i wrote this code:

````
  const int arrayCapacity = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(19);
        StaticJsonDocument<arrayCapacity> docReadings;
        JsonObject readings = docReadings.createNestedObject();
        readings["tscd"]= 0;
        readings["tbme"]= 0;
        readings["hscd"]= 0;
        readings["hbme"]= 0;
        readings["cscd"]= 0;
        readings["M1"]= 1.89;
        readings["M2"]= 2.22;
        readings["M4"]= 3.33;
        readings["M10"]= 44.44;
        readings["P1"]= 16.51;
        readings["P2"]= 0;
        readings["P4"]= 0;
        readings["P10"]= 0;
        readings["PSize"]=0;
        readings["no2"]= 0;
        readings["out"]= 0;
        readings["pressure"]= 0;
        readings["h"]= 0; 
        readings["tstamp"]= 0;
        const int JSONcapacity = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(19);
        StaticJsonDocument<JSONcapacity> doc;
        doc["u"]="giuseppe";
        doc["p"]="12345678";
        doc["id"]=2;
        doc["data"]=readings;

        char output[1024];
        serializeJson(doc, output);
 ````

but server-side arrives this object:

{ u: '', p: '', id: '', data: '' }

now, my questions are:

is the document capacity enugh? with ArduinoJson Assistant is should be 494 for esp32
can i assign the nested array as i did or i'm wrong?
thank you

peppeg85, 2020-05-08:
so, reading your book at page 125 i wrote this code:
````
  const int JSONcapacity = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(19);
  StaticJsonDocument<JSONcapacity> doc;
  doc["u"]="username";
  doc["p"]="0000000";
  doc["id"]=2;
  
  JsonObject readings = doc.createNestedObject("data");
  readings["tscd"]= 0;
  readings["tbme"]= 0;
  readings["hscd"]= 0;
  readings["hbme"]= 0;
  readings["cscd"]= 0;
  readings["M1"]= 1.89;
  readings["M2"]= 2.22;
  readings["M4"]= 3.33;
  readings["M10"]= 44.44;
  readings["P1"]= 16.51;
  readings["P2"]= 0;
  readings["P4"]= 0;
  readings["P10"]= 0;
  readings["PSize"]=0;
  readings["no2"]= 0;
  readings["out"]= 0;
  readings["pressure"]= 0;
  readings["h"]= 0; 
  readings["tstamp"]= 0;
  
  char output[1024];
  serializeJson(doc, output);
````
but still doesn't work

bblanchon, 2020-05-08:
Hi @peppeg85,

The code above works as expected.
See a demo here: https://wandbox.org/permlink/ayuMkltXxFm3T3pP

Best regards,
Benoit

peppeg85, 2020-05-08:
i think with httpclient it can't send the json, maybe it's too big, if I serial print output i have blank row

````
   HTTPClient http;
     
     http.begin(server+"/iot/sendData", test_certificate);
      
     http.addHeader("Content-Type", "application/json");
     int httpResponseCode = http.POST(output);     
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
      http.end();
````

i think the output is too big

peppeg85, 2020-05-08:
is there a way to delete the object once used? at the server arrive some old data

bblanchon, 2020-05-09:
You could discard the JsonDocument after serialization:

```c++
char output[1024];
{
  DynamicJsonDocument doc(JSONcapacity);
  // ...insert values...
  serializeJson(doc, output);
}
// ... HTTP stuffs...
```

Of course, you can get a cleaner code by extracting this block in a function.

peppeg85, 2020-05-10:
ok, thank you! i will try, the code yet is inside a function

closed 2020-05-28 by bblanchon