bblanchon/ArduinoJson #1265: filter doesn't work correctly on nested element with list

monolied, 2020-05-10:
Fine library, and nice to work with!
I just tried filters, cause it seems so useful. However, I have an issue I don't get around.
The example code is this:
```
void testfilter() {
  #define NEST_JSON "{\"response\":{\"list\":[{\"wanted\":\"this\",\"wanted_too\":9,\"unwanted\":\"naah\"},{\"wanted\":\"that\",\"wanted_too\":7,\"unwanted\":\"yuck\"}]}}"

  // create JSON filter
  StaticJsonDocument<64> jfilter;
  jfilter["response"]["list"][0]["wanted"] = true;
  jfilter["response"]["list"][0]["wanted_too"] = true;

  DynamicJsonDocument doc(1024);

  // Parse JSON object
  DeserializationError error = deserializeJson(doc, NEST_JSON, DeserializationOption::Filter(jfilter));
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  JsonArray arr = doc["response"]["list"].as<JsonArray>();
  // Extract values
  for(JsonVariant v : arr) {
    Serial.print("Wanted 1: ");
    Serial.print(v["wanted"].as<char*>());
    Serial.print(" Wanted 2: ");
    Serial.print(v["wanted_too"].as<int>());
    Serial.print(" Unwanted: ");
    Serial.println(v["unwanted"].as<char*>());
  }
}
```
Now when I run it, I don't get values for wanted_too.
So I tried adding `jfilter["response"]["list"] = true;` to the filter.
But of course I also get "unwanted" now.
So I added an explicit `jfilter["response"]["list"][0]["unwanted"] = false;` to the filter.
No dice though. I still get unwanted.

So I came to believe either there is a bug in that filter or I'm not understanding something. Which does happen.
I'd also like to add that I tried sth along the line of your weather sketch, just filtering at the list element level, like so: ` jfilter["list"][0]["wanted"] = true;` but I didn't get anything. I tried a hell of a lot more, but it only gets worse.

Platform is an Adafruit M0 Feather (ATSAMD21G18A). I'm using Arduino 1.8.12. Compiler should be arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++. Of course I can add the compiler output if of interest, but I thought it might just distract for now. It's late in the night, so I might just do something extraordinarily stupid, but I sure can't find a semantic difference to the examples I found...

bblanchon, 2020-05-10:
Hi @monolied,

The `StaticJsonDocument` for the filter is too small, so `"wanted_too"` is missing.
The [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) says you need 80 bytes on SAMD21.

See an online demo: https://wandbox.org/permlink/KKc8gjsesx6yru7V

Best regards,
Benoit

monolied, 2020-05-10:
I totally wasn't aware that the <64> was a size argument. The syntax is so different from the, say, `DynamicJsonDocument doc(1024)` declaration or some [n] size argument. I thought it was something like a type specifier for a filter... then again, I was totally sure it could be something very simple I overlooked. Thanks a lot for your time and insight!

closed 2020-05-10 by monolied

bblanchon, 2020-05-10:
You're welcome!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: