bblanchon/ArduinoJson #1266: ArduinoJson Assistant - Variable Assignment

Lifesigns37, 2020-05-10:
Hi,

New to using ArduinoJson on the ESP8266 and the script is extracting the data from the JSON which arrives like this from the site:

> `{"TotalCases":4120535,"TotalCasesToday":22247,"TotalDeaths":280867,"TotalDeathsToday":643,"TotalRecovered":1450884,"TotalTerritories":214,"CurrentUpdatedMS":1589108474597,"CurrentChangedMS":1589108012486}`

Using the assistant it provides all variables correct except CurrentUpdatedMS and CurrentChangedMS.

```
const size_t capacity = JSON_OBJECT_SIZE(8) + 140;
DynamicJsonDocument doc(capacity);

const char* json = "{\"TotalCases\":4120535,\"TotalCasesToday\":22247,\"TotalDeaths\":280867,\"TotalDeathsToday\":643,\"TotalRecovered\":1450884,\"TotalTerritories\":214,\"CurrentUpdatedMS\":1589108220258,\"CurrentChangedMS\":1589108012486}";

deserializeJson(doc, json);

long TotalCases = doc["TotalCases"]; // 4120535
int TotalCasesToday = doc["TotalCasesToday"]; // 22247
long TotalDeaths = doc["TotalDeaths"]; // 280867
int TotalDeathsToday = doc["TotalDeathsToday"]; // 643
long TotalRecovered = doc["TotalRecovered"]; // 1450884
int TotalTerritories = doc["TotalTerritories"]; // 214
long CurrentUpdatedMS = doc["CurrentUpdatedMS"]; // 1589108220258
long CurrentChangedMS = doc["CurrentChangedMS"]; // 1589108012486
```
I don't think they can be long as they are too big and I get no values. 

Reading up in the documentation I tried: 

#define ARDUINOJSON_JSON_LONG_LONG 1

This returns a value which I am unable to print to serial, as it is in milliseconds and a UNIX TIME I then divide by a 1000 using TimeLib.h as below:

```
time_t CurrentUpdateMS_t = (CurrentUpdatedMS)/1000;

char buffUpdateMS[32];
  sprintf(buffUpdateMS, "%02d/%02d/%02d %02d:%02d:%02d", day(CurrentUpdateMS_t), month(CurrentUpdateMS_t), year(CurrentUpdateMS_t), hour(CurrentUpdateMS_t) + 1, minute(CurrentUpdateMS_t), second(CurrentUpdateMS_t));
  Serial.print("Buff U: ");

  Serial.println(buffUpdateMS);
```

I should be getting the below values though after multiple iterations I'm not getting those times at all, also tried time.h and that isn't working either.

(updated: 2020-05-10 12:01:14)
(changed: 2020-05-10 11:53:32) 

Using Arduino IDE for Windows 1.8.12  and ArduinoJson 6.15.1

bblanchon, 2020-05-10:
Hi @Lifesigns37,

Yes, to support 64-bit integers in ArduinoJson, you must set [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/) to `1` (although you wrote `ARDUINOJSON_JSON_LONG_LONG` in your question).

This feature is well tested, and I have currently no reason to believe there is a bug there.
If you believe there is a bug in ArduinoJson, please isolate the problem, and provide a code example with no external dependency.

Best regards,
Benoit

bblanchon, 2020-05-10:
I just realized that, maybe, your message was:

> ArduinoJson uses `long` when it should use `long long`

I'm going to fix this issue ASAP.

Lifesigns37, 2020-05-10:
Thank you for your super quick response and yes the second was my question. 

Granted you were right about my define statement and happy to see it working now. 

Hope you and loved ones are save in these times.

bblanchon, 2020-05-10:
I updated the Assistant: it uses `long long` in the parsing program and shows an alert with a link to the documentation.

closed 2020-05-10 by bblanchon

Lifesigns37, 2020-05-10:
Excellent Thanks.