bblanchon/ArduinoJson #1268: Problem with Filters: Serializing large Data-Set

v11, 2020-05-12:
Hello Benoit

Another Week, another question. Thank you for your time! 

I have a dataset which i receive from Strava (See below). I just need to have the distance field. I tried to setup a filter like:

```
 StaticJsonDocument<64> filter;
    filter[0]["distance"] = true;
```

The response in my opinion should look like this
```
[
    {
        "distance": 29161.5
    },
    {
        "distance": 64210.4
    },
    {
        "distance": 126456.0
    }
]
```
With your assistant i calculated the size for that small document:

```
    const size_t capacity = JSON_ARRAY_SIZE(3) + 3 * JSON_OBJECT_SIZE(1) + 30;
    DynamicJsonDocument doc(capacity);

    DeserializationError error = deserializeJson(doc, client, DeserializationOption::Filter(filter));
```

The problem is: I still get "deserializeJson() failed: NoMemory" in the console.
What could be wrong here?

Unfiltered Strava Data, 3 Bike Rides
```
[
    {
        "resource_state": 2,
        "athlete": {
            "id": 20405053,
            "resource_state": 1
        },
        "name": "Mittagsradfahrt",
        "distance": 29161.5,
        "moving_time": 6828,
        "elapsed_time": 7605,
        "total_elevation_gain": 674.0,
        "type": "Ride",
        "workout_type": null,
        "id": 3412803927,
        "external_id": "garmin_push_4895123715",
        "upload_id": 3645690202,
        "start_date": "2020-05-07T10:11:29Z",
        "start_date_local": "2020-05-07T12:11:29Z",
        "timezone": "(GMT+01:00) Europe/Zurich",
        "utc_offset": 7200.0,
        "start_latlng": [
            47.490625,
            8.187544
        ],
        "end_latlng": [
            47.491331,
            8.187432
        ],
        "location_city": null,
        "location_state": null,
[... 142 lines cut ...]
        },
        "trainer": false,
        "commute": false,
        "manual": false,
        "private": false,
        "visibility": "everyone",
        "flagged": false,
        "gear_id": "b3854903",
        "from_accepted_tag": false,
        "upload_id_str": "3663203820",
        "average_speed": 7.26,
        "max_speed": 18.7,
        "average_temp": 18,
        "device_watts": false,
        "has_heartrate": false,
        "heartrate_opt_out": false,
        "display_hide_heartrate_option": false,
        "elev_high": 1941.2,
        "elev_low": 537.8,
        "pr_count": 18,
        "total_photo_count": 4,
        "has_kudoed": false
    }
]
```

bblanchon, 2020-05-13:
Hi @v11,

Thanks for reporting this problem.

Indeed, I forgot to mention it in the documentation: when filtering, `deserializeJson()` needs some extra room in the `JsonDocument` to store object keys temporarily.

In your case, it seems that le longest string is `"display_hide_heartrate_option"`, so you need to add 30+ bytes to the capacity of the `JsonDocument`.

Best regards,
Benoit

v11, 2020-05-13:
Thank you for your answer. I did i t like that and it is working:

```
const size_t capacity = JSON_ARRAY_SIZE(3) + 3 * JSON_OBJECT_SIZE(1) + 30 + 30;
    DynamicJsonDocument doc(capacity);
```

I have a follow up question:
I don't know how many rides i will do per week. In the JSON at the moment are 3 rides. Next week there will be possible 5 rides. Could the solution be, just to increase the JSON_ARRAY_SIZE to max rides i probably can do per Week? 

` JSON_ARRAY_SIZE(7)`

Or is there a chance to know the count of the array in the first place and calculate the capacity after?

bblanchon, 2020-05-14:
You could use a constant, like so:

```c++
const size_t maxRidesPerWeek = 7;

const size_t capacity = JSON_ARRAY_SIZE(maxRidesPerWeek ) + maxRidesPerWeek  * JSON_OBJECT_SIZE(1) + 30 + 30;
DynamicJsonDocument doc(capacity);
```

Or, you could use the `ESP.getMaxFreeBlockSize()` or `ESP.getMaxAllocHeap()`, as shown in [How to determine the capacity of the JsonDocument](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/).

v11, 2020-05-14:
I did it with the constant for now. Thank you! You can close this question. 👍🏻

closed 2020-05-14 by bblanchon