bblanchon/ArduinoJson #1269: Special character in JSON string gives "Deserialization failed: NotSupported"

ArnieO, 2020-05-14:
Arduino IDE
ArduinoJson ver 6.15.1
ESP32
**************
I have a JSON string where one field contains a 'special character' (in this case the Norwegian character 'ø' used in a device name).
This causes the deserialization to fail with this message: _Deserialization failed: NotSupported_

By replacing the special character with 'o' the deserialization works as expected.

Is this a bug, or is there a list of characters that must be avoided in the JSON string?

This is the JSON string that fails (MQTT output from Domoticz):
{
  "Battery" : 255,
  "RSSI" : 12,
  "description" : "",
  "dtype" : "General",
  "hwid" : "22",
  "id" : "00087433",
  "idx" : 5433,
  "name" : "Strømpris",
  "nvalue" : 0,
  "stype" : "Custom Sensor",
  "svalue1" : "16.83",
  "unit" : 1
}

bblanchon, 2020-05-14:
Hi @ArnieO,

See:

* [Why does ArduinoJson return NotSupported?](https://arduinojson.org/v6/faq/why-does-arduinojson-return-notsupported/)

Best regards,
Benoit

ArnieO, 2020-05-14:
Yepp, that works!
Impeccable - merci beaucoup!

closed 2020-05-14 by ArnieO

bblanchon, 2020-05-14:
You're welcome!
Thank you for using ArduinoJson.