bblanchon/ArduinoJson #1272: Not able to parse String variable to JsonDocument

zaindroid, 2020-05-17:
Hi, I am trying to create a Json parameter for my HTTP post request on ESP32 using arduino IDE(1.8.12) but when I assign String variable to parser , it shows NULL in the output. Can you please see what am I doing wrong. I am also attaching my code.

String mac= "24:6F:28:96:F9:F4";
const size_t capacity = JSON_OBJECT_SIZE(3);
DynamicJsonDocument doc(capacity);
doc["grant_type"].set("authorization_code");
doc["client_id"].set(mac);
doc["authorization_code"].set("1758f542225a40aece0f7ba3e86379eeca9f9e09406ac0ad3c49060ae7fcd7dccd156434534534");
serializeJsonPretty(doc, Serial);
//doc.prettyPrintTo(Serial);
//serializeJsonPretty(doc, myData);

//============ OUTPUT==================

{
  "grant_type": "authorization_code",
  "client_id": null,
  "authorization_code": "1758f542225a40aece0f7ba3e86379eeca9f9e09406ac0ad3c49060ae7fcd7dccd1564045575144"
}

//====================================

"client_id" key shows null value.

It works fine if I assign the value directly , instead of using String type variable.

doc["client_id"].set("24:6F:28:96:F9:F4");

with following output:

//================output-2=====================

{
  "grant_type": "authorization_code",
  "client_id": "24:6F:28:96:F9:F4",
  "authorization_code": "1758f542225a40aece0f7ba3e86379eeca9f9e09406ac0ad3c49060ae7fcd7dccd1564045575144"
}
//================================================

bblanchon, 2020-05-18:
Hi @zaindroid,

When you use a `String`, [JsonVariant::set()](https://arduinojson.org/v6/api/jsonvariant/set/) makes a copy in the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/), so you must increase the capacity accordingly.

See also the [Serialization tutorial](https://arduinojson.org/v6/doc/serialization/), section 4.6 "Duplication of strings".

By the way, why are you calling `.set()` instead of using the assignment operator as shown in the examples?

Best regards,
Benoit

closed 2020-05-28 by bblanchon

zaindroid, 2020-05-28:
thanks alot, its really helped. I was using assignment operator earlier and it was having same issue thats why I used .set method. But yeah,  duplication of strings was my issue. thanks